linuxboot/heads #731: Build fails with 'nvramcui' as second payload

Siproqu, 2020-05-30:
It fails with:

```
** Default configuration is based on 'configs/defconfig'
#
# configuration written to /home/user/work/heads/build/coreboot-4.8.1/payloads/nvramcui/.config
#
#
# configuration written to /home/user/work/heads/build/coreboot-4.8.1/payloads/nvramcui/.config
#
    CC         libpci/libpci.libpci.o
cc1: error: -mpreferred-stack-boundary=2 is not between 3 and 12
make[3]: *** [Makefile:274: /home/user/work/heads/build/coreboot-4.8.1/payloads/nvramcui/build/libpci/libpci.libpci.o] Error 1
make[2]: *** [Makefile:12: /home/user/work/heads/build/coreboot-4.8.1/payloads/nvramcui/libpayload] Error 2
make[1]: *** [payloads/Makefile.inc:43: payloads/nvramcui/nvramcui.elf] Error 2
make[1]: Leaving directory '/home/user/work/heads/build/coreboot-4.8.1'
make: *** [Makefile:382: /home/user/work/heads/build/coreboot-4.8.1/t430s/.build] Error 1

```

After changing mpreferred-stack-boundary to 3 in
`heads/build/coreboot-4.8.1/payloads/libpayload/arch/x86/Makefile.inc`

it fails with:
```
    CC         libpci/libpci.libpci.o
<command-line>: error: "_LP64" redefined [-Werror]
<built-in>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[3]: *** [Makefile:274: /home/user/work/heads/build/coreboot-4.8.1/payloads/nvramcui/build/libpci/libpci.libpci.o] Error 1
make[2]: *** [Makefile:12: /home/user/work/heads/build/coreboot-4.8.1/payloads/nvramcui/libpayload] Error 2
make[1]: *** [payloads/Makefile.inc:43: payloads/nvramcui/nvramcui.elf] Error 2
make[1]: Leaving directory '/home/user/work/heads/build/coreboot-4.8.1'
make: *** [Makefile:382: /home/user/work/heads/build/coreboot-4.8.1/t430s/.build] Error 1
```

Unfortunately, I can't find the source of this error.
Any suggestions?

MrChromebox, 2020-06-01:
what's the point of building secondary payloads since Heads can't launch them?

tlaurion, 2020-06-02:
> what's the point of building secondary payloads since Heads can't launch them?

@MrChromebox those are ELF binaries. Why couldn't we launch them?

Siproqu, 2020-06-03:
I have built the nvramcui in a standalone coreboot build (i386), but couldn't run it. The kernel doesn't accept it and the elf file is interpreted as a shell script.
Heads is built as an i386 kernel, isn't it?

tlaurion, 2020-06-07:
@Siproqu coreboot only is I386

Siproqu, 2020-07-16:
As I do not have the time to test it again and the settings applied by nvramcui won't change frequently anyways, here is a [workaround](https://github.com/osresearch/heads/issues/711#issuecomment-636397708).
Obviously the linked patch file of the workaround has to be adjusted for the user's needs.

closed 2020-07-16 by Siproqu