linuxboot/heads #733: certificate of 'mirrors.kernel.org' is not trusted from debian:bullseye

tlaurion, 2020-06-03:
```
2020-06-03 16:17:25+00:00 WGET https://mirrors.kernel.org/sourceware/lvm2/LVM2.2.02.168.tgz

if ! wget -O "/root/project/packages/LVM2.2.02.168.tgz.tmp" https://mirrors.kernel.org/sourceware/lvm2/LVM2.2.02.168.tgz ; then exit 1 ; fi ; mv "/root/project/packages/LVM2.2.02.168.tgz.tmp" "/root/project/packages/LVM2.2.02.168.tgz" 

--2020-06-03 16:17:25--  https://mirrors.kernel.org/sourceware/lvm2/LVM2.2.02.168.tgz

Resolving mirrors.kernel.org (mirrors.kernel.org)... 198.145.21.9, 2001:19d0:306:6:0:1994:3:14

Connecting to mirrors.kernel.org (mirrors.kernel.org)|198.145.21.9|:443... connected.

ERROR: The certificate of 'mirrors.kernel.org' is not trusted.

ERROR: The certificate of 'mirrors.kernel.org' has expired.

make: *** [Makefile:382: /root/project/packages/LVM2.2.02.168.tgz] Error 1

```

tlaurion, 2020-06-03:
Ticker opened https://github.com/docker/for-linux/issues/1032

lsafd, 2020-06-03:
Having the same problem on Fedora 30.

tlaurion, 2020-06-03:
> Works now for me.
> 
> ```
> $ openssl s_client -connect mirrors.kernel.org:443
> CONNECTED(00000003)
> depth=2 C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
> verify return:1
> depth=1 C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2
> verify return:1
> depth=0 OU = Domain Control Validated, OU = PositiveSSL Multi-Domain, CN = kernel.org
> verify return:1
> ---
> Certificate chain
>  0 s:OU = Domain Control Validated, OU = PositiveSSL Multi-Domain, CN = kernel.org
>    i:C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2
>  1 s:C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2
>    i:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
>  2 s:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
>    i:C = SE, O = AddTrust AB, OU = AddTrust External TTP Network, CN = AddTrust External CA Root
> […]
> […]
> Verification: OK
> […]
> ```

@paulmenzel from where?
Under QubesOS debian-10, it works. I'm testing docker debian:buster on another PR.

tlaurion, 2020-06-03:
Well... let's wait they update their certs package? That would be upstream to all distro?

tlaurion, 2020-06-03:
@paulmenzel 
Under debian-10:
```
user@x230-privacybeast-staging:~/heads$ make BOARD=x230-privacybeast lvm2
2020-06-03 13:48:49-04:00 WGET https://mirrors.kernel.org/sourceware/lvm2/LVM2.2.02.168.tgz
--2020-06-03 13:48:49--  https://mirrors.kernel.org/sourceware/lvm2/LVM2.2.02.168.tgz
Resolving mirrors.kernel.org (mirrors.kernel.org)... 149.20.37.36
Connecting to mirrors.kernel.org (mirrors.kernel.org)|149.20.37.36|:443... connected.
ERROR: The certificate of ‘mirrors.kernel.org’ is not trusted.
ERROR: The certificate of ‘mirrors.kernel.org’ has expired.
make: *** [Makefile:382: /home/user/heads/packages/LVM2.2.02.168.tgz] Error 1

```

openssl s_client -connect mirrors.kernel.org:443
```
CONNECTED(00000003)
depth=2 C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
verify return:1
depth=1 C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2
verify return:1
depth=0 OU = Domain Control Validated, OU = PositiveSSL Multi-Domain, CN = kernel.org
verify return:1
---
Certificate chain
 0 s:OU = Domain Control Validated, OU = PositiveSSL Multi-Domain, CN = kernel.org
   i:C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2
 1 s:C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2
   i:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
 2 s:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust RSA Certification Authority
   i:C = SE, O = AddTrust AB, OU = AddTrust External TTP Network, CN = AddTrust External CA Root
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIGszCCBZugAwIBAgIQRF7gFMlJ3UO909a39zv1mzANBgkqhkiG9w0BAQsFADBf
MQswCQYDVQQGEwJGUjEOMAwGA1UECBMFUGFyaXMxDjAMBgNVBAcTBVBhcmlzMQ4w
DAYDVQQKEwVHYW5kaTEgMB4GA1UEAxMXR2FuZGkgU3RhbmRhcmQgU1NMIENBIDIw
HhcNMTkwOTI3MDAwMDAwWhcNMjAwOTI3MjM1OTU5WjBbMSEwHwYDVQQLExhEb21h
aW4gQ29udHJvbCBWYWxpZGF0ZWQxITAfBgNVBAsTGFBvc2l0aXZlU1NMIE11bHRp
LURvbWFpbjETMBEGA1UEAxMKa2VybmVsLm9yZzCCAaIwDQYJKoZIhvcNAQEBBQAD
ggGPADCCAYoCggGBAOD0/Tk0EeH6/ahZQAiBtoMMY8Bxmql2kxJ+smBIP9Yq+MtJ
utc/CeUbnTTnpLDf1nTjqJ6AGyCE+pzw8sPSXKJrY6he1jjCafjsx193KMvqCUty
SZgDdsV7AKr4KjbbQ9CE3tTR1cBKYcCvro4elAXcLbG53qWe/UXwcIPmvwj8n2WW
irMHTr4b+x1Pr7B2Vhc2IHFdnzb43krTXiXuuWCo84281hxO7EIlD3Enjm7rICpU
coldqOaNS3LRkeiR8RrbQfyiqI8XncSykjzVbOZVSVRCvLzRL0MsBKU1F/WMoBYc
ahV92wnYnpGD1s7Wi1eP8ne5+5SPqwS43G4AXxH0hdU7gkHS4i0n7nGmmfRIxD+I
57dvXdnxgSyT21IHp+lMFashblRg8+ZZD5Oy1ouTqBe604FXsjryeQqRUFePTPRB
vYxlg31qne/UUPpo1GcDAjsTv6YSUyUjXoINBQsvpDUgHYkOkHXEyMVRynpBsbJA
p0elmqf4aMr89tn72QIDAQABo4IC7TCCAukwHwYDVR0jBBgwFoAUs5Cn2MmvTs1h
PJ98rV1/Qf1pMOowHQYDVR0OBBYEFLr+wGAi2JEEPCMYIvyjFmMlmLK/MA4GA1Ud
DwEB/wQEAwIFoDAMBgNVHRMBAf8EAjAAMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggr
BgEFBQcDAjBLBgNVHSAERDBCMDYGCysGAQQBsjEBAgIaMCcwJQYIKwYBBQUHAgEW
GWh0dHBzOi8vY3BzLnVzZXJ0cnVzdC5jb20wCAYGZ4EMAQIBMEEGA1UdHwQ6MDgw
NqA0oDKGMGh0dHA6Ly9jcmwudXNlcnRydXN0LmNvbS9HYW5kaVN0YW5kYXJkU1NM
Q0EyLmNybDBzBggrBgEFBQcBAQRnMGUwPAYIKwYBBQUHMAKGMGh0dHA6Ly9jcnQu
dXNlcnRydXN0LmNvbS9HYW5kaVN0YW5kYXJkU1NMQ0EyLmNydDAlBggrBgEFBQcw
AYYZaHR0cDovL29jc3AudXNlcnRydXN0LmNvbTBdBgNVHREEVjBUggprZXJuZWwu
b3JnghJhcmNoaXZlLmtlcm5lbC5vcmeCDmdpdC5rZXJuZWwub3JnghJtaXJyb3Jz
Lmtlcm5lbC5vcmeCDnd3dy5rZXJuZWwub3JnMIIBBAYKKwYBBAHWeQIEAgSB9QSB
8gDwAHYAsh4FzIuizYogTodm+Su5iiUgZ2va+nDnsklTLe+LkF4AAAFtcsuYjgAA
BAMARzBFAiAgbvK3x78RoQXt035wmJqTm/wMFWgfla2ctSyBXLzepAIhAKApjbW3
8p0jlaBkrKQA8eEieenCN5F+PZtm8JSHhAD2AHYAXqdz+d9WwOe1Nkh90EngMnqR
mgyEoRIShBh1loFxRVgAAAFtcsuYsgAABAMARzBFAiEA4JnyL5WAsHHd4WbhqoG6
/C3KFYmZGg04YUFxqAzwbpMCIBjHZMvO/LKSdVcKC45c5FJwc75O/2+7vbkfFAn1
/WV4MA0GCSqGSIb3DQEBCwUAA4IBAQASeh5QNteAKqY+sr7uBTHq56v1MJbbdMO7
QJaCSQd4P2OSQDA83oGfdkj458+d9gMTvBu+pNi1/l0aIz1IMEsuAJNXhN5jLCB/
n1CHaTK5b9Oda96+MejWAiiTNZo1UBLQ5ixNvGp1MHDklELm/supbatSCP65eEpp
E7OI5lLxCLrvsiwUaSKIIO2tEIgwiMkwopdMgwJa7RqljUP7YlYKnAxizOi+yTrA
nXA0OqLtrl5pwnN3Uj/F91X6c6tOvHWkNZ1qPad6r7ZHCP8mq3RFiMeSixiJ6LR2
gtFRKmvMIUIESh60F91+2AeUfCa/tBfOM+PDpsNNaZ+iHHaicAIw
-----END CERTIFICATE-----
subject=OU = Domain Control Validated, OU = PositiveSSL Multi-Domain, CN = kernel.org

issuer=C = FR, ST = Paris, L = Paris, O = Gandi, CN = Gandi Standard SSL CA 2

---
No client certificate CA names sent
Peer signing digest: SHA512
Peer signature type: RSA
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 5284 bytes and written 436 bytes
Verification: OK
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 3072 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: 841FE877433C13C35DD2AAA6F15F21E43DD55871A8D8F228E514E17430C6E393
    Session-ID-ctx: 
    Master-Key: 337268E8870A264A451FD3C3C6289B57560CCFFF4850727CE3C2DE79B91B936D3B95BB89C589566F9C11B62291D64CBB
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    Start Time: 1591206556
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
---

```

tlaurion, 2020-06-03:
Wrote to webmaster@kernel.org

tlaurion, 2020-06-03:
```
sudo apt-get install ca-certificates
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ca-certificates is already the newest version (20190110).

```

tlaurion, 2020-06-03:
Was fixed by webmaster@kernel.org

closed 2020-06-03 by tlaurion

paulmenzel, 2020-06-04:
I tested under Debian Sid/unstable.

> Was fixed by webmaster@kernel.org

Did they say what the problem was? Your `openssl s_client` output also shows `Verification: OK`, so I do not know what’s wrong.

GNU Wget seems to be linked against GnuTLS though.

    $ ldd /usr/bin/wget | grep tls
    	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007f154beaa000)

tlaurion, 2020-06-04:
@paulmenzel nope. They just did their magic and asked me to test again and wget worked again.

aflyhorse, 2021-09-30:
> GNU Wget seems to be linked against GnuTLS though.

Out of topic. My wget in WSL Debian also linked against GnuTLS, but after ```apt install ca-certificates``` it did the trick.