linuxboot/heads #734: Kernel is not reproducible

tlaurion, 2020-06-03:
@osresearch @BlackMaria @MrChromebox @flammit :

First step accomplished:

- [x] Each commit now triggers two CIs. For [example](https://github.com/osresearch/heads/commit/488d6529e463a9732f653f1d6bb89080f4768237): Click on the green check, and it will show you CircleCI and GitlabCI clickable builds. 

- [ ] Make vmlinuz reproducible again.

  - [795b307945343c5c7eb0b55a9285bb73f96832d295da9be5e11887db804ebe02  /root/project/build/x230-hotp-verification/bzImage](https://app.circleci.com/pipelines/github/osresearch/heads/162/workflows/2d04cd6b-4d3e-457f-97f5-47a3f188e3a4/jobs/170/parallel-runs/0/steps/0-114)

  - [69961890c7ee425855f8620562b5627584f7b2c5f52132291836445087cc90ef  /builds/tlaurion/heads-osresearch/build/x230-hotp-verification/bzImage](https://gitlab.com/tlaurion/heads-osresearch/-/jobs/565440344)

- [x] Have heads.cpio reproducible
  - shared hash: 980a1e671df217b2fa68896dfc6920695a65ed02c4f7e8135c5f093f50731ed7  /builds/tlaurion/heads-osresearch/build/x230-hotp-verification/heads.cpio

- [x] Have modules.cpio reproducible 
  - shared hash: a2fe6993f933f9a99a8d2b999f400ae57f834bc6443679c45db879c1dcdf3222  /root/project/build/x230-hotp-verification/modules.cpio

- [ ] Have tools.cpio reproducible

- [ ] Have initrd.cpio.xz reproducible

- [ ] have reproducible roms

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/571#issuecomment-633295172_

tlaurion, 2020-06-07:
@daym?

daym, 2020-06-07:
Still at it!

tlaurion, 2020-07-22:
So for commit 9719510f395bcb54efc6b01d309184ba637e6f29 (previous commit)

GitlabCI artifacts (all): https://gitlab.com/tlaurion/heads-osresearch/-/jobs/634898242/artifacts/download
Specifics:
https://gitlab.com/tlaurion/heads-osresearch/-/jobs/634898242/artifacts/raw/build/qemu-coreboot/bzImage
build/qemu-coreboot/coreboot.rom
https://tlaurion.gitlab.io/-/heads-osresearch/-/jobs/634898242/artifacts/build/qemu-coreboot/hashes.txt
build/qemu-coreboot/logs.tar.gz

CircleCI all artifacts:https://app.circleci.com/pipelines/github/osresearch/heads/178/workflows/1ffc4b88-5001-460c-ad5b-aaa2cde9c904/jobs/189/artifacts
Specifics:
https://gitlab.com/tlaurion/heads-osresearch/-/jobs/634898242/artifacts/browse/build/qemu-coreboot/
https://gitlab.com/tlaurion/heads-osresearch/-/jobs/634898242/artifacts/file/build/qemu-coreboot/bzImage
https://gitlab.com/tlaurion/heads-osresearch/-/jobs/634898242/artifacts/file/build/qemu-coreboot/coreboot.rom
https://gitlab.com/tlaurion/heads-osresearch/-/jobs/634898242/artifacts/file/build/qemu-coreboot/logs.tar.gz

daym, 2020-08-02:
Converting a bzImage to a ELF image is possible via https://github.com/marin-m/vmlinux-to-elf worked, diffoscope takes almost an hour on the result--but it worked. Seems that the main difference is the start of the program headers--everything else seems okay enough.

krystian-hebel, 2020-09-21:
> ```
> ├── readelf --wide --decompress --hex-dump=.init.data {}
> │ @@ -31872,21 +31872,23 @@
> │    0xffffffff81ede7d0 1569e481 ffffffff 7880e481 ffffffff .i......x.......
> │    0xffffffff81ede7e0 e7abe481 ffffffff 75b6e481 ffffffff ........u.......
> │    0xffffffff81ede7f0 deb9e481 ffffffff fb03e581 ffffffff ................
> │    0xffffffff81ede800 4525e581 ffffffff bec83381 ffffffff E%........3.....
> │    0xffffffff81ede810 b766e581 ffffffff 507be581 ffffffff .f......P{......
> │    0xffffffff81ede820 bf92e581 ffffffff f0cce581 ffffffff ................
> │    0xffffffff81ede830 6d782c81 ffffffff 2a0be581 ffffffff mx,.....*.......
> │ -  0xffffffff81ede840 aa13e581 ffffffff 1f8b0800 00000000 ................
> ... snip...
> │ +  0xffffffff81ede840 aa13e581 ffffffff fd377a58 5a000001 .........7zXZ...
> ... snip...
> ```

Those last bytes are two magic numbers for compressed files, the first one is for GZIP, and the second for XZ (see [Wikipedia](https://en.wikipedia.org/wiki/List_of_file_signatures)). It is a buffer used by function [populate_rootfs](https://github.com/torvalds/linux/blob/bebc6082da0a9f5d47a1ea2edc099bf671058bd4/init/initramfs.c#L613), and as far as I'm aware it is built from files in [/usr](https://github.com/torvalds/linux/tree/bebc6082da0a9f5d47a1ea2edc099bf671058bd4/usr). The format of rootfs file should be controllable with [CONFIG_INITRAMFS_COMPRESSION](https://github.com/torvalds/linux/blob/bebc6082da0a9f5d47a1ea2edc099bf671058bd4/usr/Makefile#L9), which `depends on INITRAMFS_SOURCE!=""` [here](https://github.com/torvalds/linux/blob/bebc6082da0a9f5d47a1ea2edc099bf671058bd4/usr/Kconfig#L220).

I do not know how it chooses the compression method if the option itself is disabled (I assume .config is identical in both cases, but I might be wrong about that), but then again I looked primarily on the newer sources and this part of configuration was changed couple of times. It seems that back in 4.14 it was impossible to use uncompressed rootfs.

tlaurion, 2020-09-24:
> Those last bytes are two magic numbers for compressed files, the first one is for GZIP, and the second for XZ (see [Wikipedia](https://en.wikipedia.org/wiki/List_of_file_signatures)). It is a buffer [...]

@krystian-hebel Thank you so muh for your analysis.
The linux-qemu config on which this build was from is here: https://github.com/osresearch/heads/blob/9719510f395bcb54efc6b01d309184ba637e6f29/config/linux-qemu.config and was generated from make gendefconfig, which is why the config file only contains differences from defconfig.

If one does, for the same of commit id https://github.com/osresearch/heads/commit/9719510f395bcb54efc6b01d309184ba637e6f29 for which rom was given in previous comment:
```
git checkout 9719510f395bcb54efc6b01d309184ba637e6f29
git reset --hard
(already build here)
cp config/linux-qemu.config build/linux-4.14.62/.config
cd build/linux-4.14.62
make meuconfig
(save)
(exit)

```
Then:
```
user@x230-master:~/heads/build/linux-4.14.62$ grep INITRAMFS_COMPRESSION .config
CONFIG_INITRAMFS_COMPRESSION=".xz"
user@x230-master:~/heads/build/linux-4.14.62$ 
```

Both roms were produce with the same [linux config file](https://github.com/osresearch/heads/blob/9719510f395bcb54efc6b01d309184ba637e6f29/config/linux-qemu.config)

tlaurion, 2020-09-24:
@krystian-hebel 
Actually, you might have been right. Why selected behavior was different depending of host build system is not explained in my mind yet, but here is the facts:

[image]
Now ticked.

[image]
Save.

Then:
```
user@x230-master:~/heads/build/linux-4.14.62$ make savedefconfig
scripts/kconfig/conf  --savedefconfig=defconfig Kconfig
user@x230-master:~/heads/build/linux-4.14.62$ cp defconfig ../../config/linux-qemu.config 
user@x230-master:~/heads/build/linux-4.14.62$ git diff
diff --git a/config/linux-qemu.config b/config/linux-qemu.config
index bb49391..31bb1f6 100644
--- a/config/linux-qemu.config
+++ b/config/linux-qemu.config
@@ -25,6 +25,7 @@ CONFIG_INITRAMFS_SOURCE="../../../blobs/dev.cpio"
 # CONFIG_RD_LZMA is not set
 # CONFIG_RD_LZO is not set
 # CONFIG_RD_LZ4 is not set
+CONFIG_INITRAMFS_COMPRESSION_XZ=y
 CONFIG_CC_OPTIMIZE_FOR_SIZE=y
 # CONFIG_SGETMASK_SYSCALL is not set
 # CONFIG_SYSFS_SYSCALL is not set
@@ -248,9 +249,7 @@ CONFIG_GENERIC_PHY=y
 CONFIG_DMI_SYSFS=y
 CONFIG_GOOGLE_FIRMWARE=y
 CONFIG_GOOGLE_MEMCONSOLE_X86_LEGACY=y
-# CONFIG_EXT2_FS is not set
 CONFIG_EXT4_FS=y
-CONFIG_EXT4_USE_FOR_EXT2=y
 CONFIG_XFS_FS=y
 # CONFIG_DNOTIFY is not set
 # CONFIG_INOTIFY_USER is not set

```
Where `CONFIG_INITRAMFS_COMPRESSION_XZ=y` is now explicited.

tlaurion, 2020-09-24:
Same applies to all other linux configs for other boards it seems (tested for linux-x230.config also)

tlaurion, 2020-10-02:
builds to be compared for kernel with commit https://github.com/tlaurion/heads/commit/b03fc6ad2487000555295fd352767b15b9a16f31

https://circleci.com/gh/tlaurion/heads/487?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link
https://gitlab.com/tlaurion/heads/-/jobs/770639041

tlaurion, 2020-10-03:
> Those last bytes are two magic numbers for compressed files, the first one is for GZIP, and the second for XZ (see [Wikipedia](https://en.wikipedia.org/wiki/List_of_file_signatures)). It is a buffer [...]

@krystian-hebel other ideas? Implementing the above changes still produce non-reproducible kernels as shown under https://github.com/osresearch/heads/pull/850#issuecomment-703175011

krystian-hebel, 2020-10-04:
Issue remains the same, i.e. GZIP vs XZ.
Maybe do the same with `RD_XZ`? It is a long shot, but it is used [here](https://github.com/torvalds/linux/blob/bebc6082da0a9f5d47a1ea2edc099bf671058bd4/Makefile#L915). However, unless I misunderstood this, it would mean that in some cases initramfs may be compressed twice, using two different methods (plus the compression of the kernel itself).

krystian-hebel, 2020-10-04:
By manually dd-ing that initrd out I can confirm that CPIOs are identical in both cases:

```
$ xzcat test_ci.xz | cpio -t
dev
dev/console
lib/ld-
lib/ld-musl-x86_64.so.1
2 bloki

$ gzip -c -d test_gl.gz | cpio -t
dev
dev/console
lib/ld-
lib/ld-musl-x86_64.so.1
2 bloki

$ xzcat test.xz | sha256sum 
00e62b12c5519ff505e1ea37e7395a56c94e4ca9fc4ba1e04a04b73e50d2526e  -

$ gzip -c -d test_gl.gz | sha256sum 
00e62b12c5519ff505e1ea37e7395a56c94e4ca9fc4ba1e04a04b73e50d2526e  -
```

All differences in code seem to be just in memory addresses of data and called functions. `test_ci.xz` is 0x20 (32) bytes bigger than `test_gl.gz`, but the offsets in the code differ by 0x30 and 0xf0 bytes, depending whether it accesses data before or after kernel decompression. These offsets seem strange, but it may be caused by alignment of individual sections - the main XZ-compressed part of the kernel is not aligned, and because of compression its size does not depend on the size of uncompressed data (not directly anyway).

tlaurion, 2020-10-04:
> Issue remains the same, i.e. GZIP vs XZ.
> Maybe do the same with `RD_XZ`? It is a long shot, but it is used [...]

@krystian-hebel As said previously, Heads store config as `make savedefconfig result`. I expended that in latest commit so we have a clear view on from what linux config the bzImage is built from to try to understand better from where the variation comes from. 

So https://github.com/tlaurion/heads/commit/653582e1576f439541ee0ce90bca6dff61536377 builds only for qemu-coreboot now. We know cpio is reproducible, and from past tests, we know that only the kernel is not reproducible as of now. The question is still why.

On that matter, you can see https://github.com/tlaurion/heads/commit/653582e1576f439541ee0ce90bca6dff61536377#diff-883dee0e42d0c32ec557a7bd837e6b67 what is "supposed" to be part of the `make savedefconfig`. 

To produce such config, what was done was:

- Expend defconfig to explicit linux config:
```
cp config/linux-qemu.config build/linux-4.14.62/.config
cd build/linux-4.14.62/
make menuconfig
(save)
cp .config ../../config/linux-qemu.config
cd ~/heads
git add config/linux-qemu.config
```
Nothing else on that matter.

- I removed all other boards of that PR from GitlabCI and CIrcleCI.

tlaurion, 2020-10-04:
As of  [CircleCI build artifact's hashes](https://app.circleci.com/pipelines/github/tlaurion/heads/458/workflows/06c94ae9-66d4-42a1-b927-337a890eb754/jobs/493/parallel-runs/0/steps/0-108) which [matches even previous CI builds](https://github.com/osresearch/heads/issues/734#issuecomment-662572884) (normal)
`13f03e823822fd854cbed6ddaadc9408d6626f5d32a1870fa7522242fea4d567  /root/project/build/qemu-coreboot/bzImage`

- CircleCI [downloadeable artifacts](https://app.circleci.com/pipelines/github/tlaurion/heads/458/workflows/06c94ae9-66d4-42a1-b927-337a890eb754/jobs/493/artifacts)
- CircleCI direct download of bzImage: https://493-103208611-gh.circle-artifacts.com/0/build/qemu-coreboot/bzImage

- GitlabCI (To be edited when builds succeeds in 2 hours) 
(Note that GitlabCI cannot build with cache since https://github.com/osresearch/heads/issues/768#issuecomment-703268815 is not yet figured out...)

krystian-hebel, 2020-10-04:
For GitlabCI it also matches the previous build. Even with full, explicit config it seems to either a) choose a different path when parsing Makefiles, or b) re-parse configuration from Kconfig files on final `make`, after which .config is modified/"sanitized" in a different manner on those two CIs. This kind of narrows it down, I think the next logical step would be to see if the build system uses environment variables for compression stuff.

PR #1630 (2024-04-26): make openssl libtss2 tpm2 reproducible
fixes #1616 directly.

---

Past work related to fixating pkg-config config correctly and limiting host system to bleed in builds fixed some previous issues as well, where this PR fixed CircleCI to use working dir of heads (as opposed to project in the past):
closes #734 
closes #892 

----
TODO:

- [ ] Test extensively for tpm2 toolstack regression. Under qemu-tpm2 boards, I could see quickly warnings about public key counterpart not being valid, but didn't see any regression for selaing/unsealing ops up to DUK sealing/unsealing. https://github.com/linux [... cut]
closed 2024-04-26 by JonathonHall-Purism