linuxboot/heads #735: libpng not downloadable from sourceforge

tlaurion, 2020-06-03:
```
2020-06-03 20:26:27+00:00 WGET https://download.sourceforge.net/libpng/libpng-1.6.34.tar.gz

if ! wget -O "/root/project/packages/libpng-1.6.34.tar.gz.tmp" https://download.sourceforge.net/libpng/libpng-1.6.34.tar.gz ; then exit 1 ; fi ; mv "/root/project/packages/libpng-1.6.34.tar.gz.tmp" "/root/project/packages/libpng-1.6.34.tar.gz" 

--2020-06-03 20:26:27--  https://download.sourceforge.net/libpng/libpng-1.6.34.tar.gz

Resolving download.sourceforge.net (download.sourceforge.net)... 216.105.38.13

Connecting to download.sourceforge.net (download.sourceforge.net)|216.105.38.13|:443... 
connected.

HTTP request sent, awaiting response... 301 Moved Permanently

Location: https://downloads.sourceforge.net/project/libpng/libpng16/1.6.34/libpng-1.6.34.tar.gz [following]

--2020-06-03 20:26:27--  https://downloads.sourceforge.net/project/libpng/libpng16/1.6.34/libpng-1.6.34.tar.gz

Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 216.105.38.13

Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|216.105.38.13|:443... connected.

HTTP request sent, awaiting response... 
302 Found

Location: https://managedway.dl.sourceforge.net/project/libpng/libpng16/1.6.34/libpng-1.6.34.tar.gz [following]

--2020-06-03 20:26:27--  https://managedway.dl.sourceforge.net/project/libpng/libpng16/1.6.34/libpng-1.6.34.tar.gz

Resolving managedway.dl.sourceforge.net (managedway.dl.sourceforge.net)... 208.79.213.6

Connecting to managedway.dl.sourceforge.net (managedway.dl.sourceforge.net)|208.79.213.6|:443... 
connected.

GnuTLS: Error in the pull function.

Unable to establish SSL connection.

make: *** [Makefile:382: /root/project/packages/libpng-1.6.34.tar.gz] Error 1

```

lsafd, 2020-06-03:
I had this issue, but after trying to build a second time it resolved itself. It may just be inconsistent.

tlaurion, 2020-06-03:
[Src](https://app.circleci.com/pipelines/github/tlaurion/heads/202/workflows/e86e1585-39eb-466c-bac5-16379092e33d/jobs/219).

Fix https://github.com/tlaurion/heads/commit/18f8265e880f4d199466ab282c9261a0fa9925df

PR #736 (2020-06-03): Libpng download from GitHub
fixes #735
commit 9090f1a1 (2020-06-03): libpng : moving archive download from sourceforge to github
Fixes #735
closed 2020-06-03 by tlaurion