linuxboot/heads #737: Blank Screen After Flashing x230-hotp-verification

lsafd, 2020-06-03:
First off I would like to say that I am a relatively new user so please excuse the mistakes that I will probably make.

I am flashing to a x230 and having similar issues to #667. I am using `x230-hotp-verification` as my final payload for the internal flashing. I have previously used the `x230` as my final payload and had success, so this is not my first installation. 

Steps:
1. Flash `x230-flash.rom` image to the top chip.
2. Flash 8MB unlocked and me cleaned image to the bottom chip.
3. Boot to heads shell (So far everything is working)
4. Run "mount-usb".
5. Run `flash.sh -c /media/coreboot.rom` to flash the 12MB `coreboot.rom` image (from `/build/x230-hotp-verification/`).
6. Run `reboot` to reboot. (I have also tried rebooting using the power button.)

I end up with a blank screen, but the green light for the power button turns on and off as expected.

Attached is the `coreboot.rom` file. I have done a clean build and ended up with a `coreboot.rom` file that has the exact same sha256 hash.

Any help would be much appreciated.

[rom_file.zip](https://github.com/osresearch/heads/files/4709026/rom_file.zip)

tlaurion, 2020-06-04:
Will retest tomorrow

tlaurion, 2020-06-07:
@lsafd still not sure

lsafd, 2020-06-08:
> Reconsidering what to do from here for the x230-hotp-verification platform, which doesn't compile if we go back to CBFS_REGION of 70000, not having enough space:

@tlaurion Isn't this the issue (From #667)? Do you just mean that you don't yet now how to solve this problem? Is there an older version or commit that I should clone from that is guaranteed to be working for x230-hotp-verification (or something similar)? 

Thanks a ton for looking into this I realize you're pretty busy!

tlaurion, 2020-06-19:
@lsafd : I will retest, but x230-htop-verification applied on top of https://github.com/osresearch/heads-wiki/blob/master/Clean-the-ME-firmware.md is supposed to work.

tlaurion, 2020-06-19:
Can you confirm the behavior is still current with latest downloaded rom from [circleci](https://app.circleci.com/pipelines/github/osresearch/heads) ?

Or a clean build 

- `make BOARD=x230-hotp-verification real.clean && rm build/* crossgcc`
- `make BOARD=x230-hotp-verification`

AUX234, 2020-06-19:
> Can you confirm the behavior is still current with latest downloaded rom from [circleci](https://app.circleci.com/pipelines/github/osresearch/heads) ?
> 
> Or a clean build
> 
> * `make BOARD=x230-hotp-verification real.clean && rm build/* crossgcc`
> * `make BOARD=x230-hotp-verification`

@tlaurion I am having the same issue as @lsafd with a clean build and the latest downloaded rom from circleci

tlaurion, 2020-06-19:
So it means t430 and x230 are different....

AUX234, 2020-06-20:
@tlaurion @lsafd I was able to successfully install heads by building coreboot.rom using #748

tlaurion, 2020-06-20:
Their CBFS region for x230-hotp-verification is `CONFIG_CBFS_SIZE=0x710000` instread of `CONFIG_CBFS_SIZE=0x800000` (https://github.com/osresearch/heads/pull/748/files#diff-540d604c350e8fd52c25559b642795fbR5)

lsafd, 2020-06-23:
With the changed in #756 I was able to get things working. The only caveat is that about half of the time when I try to boot the screen stays black and my computer beeps loudly at me. However, the main issue has been solved, so I will close this issue.

Thanks a ton!

closed 2020-06-23 by lsafd

tlaurion, 2020-06-24:
@lsafd that is not normal?

lsafd, 2020-06-24:
I never experienced this problem when using the coreboot.rom file from x230 (as opposed to x230-hotp-verification), so my assumption was that that this is unusual. Also, I may make another issue for some problems related to luks tpm stuff I'm having.

tlaurion, 2020-10-19:
Fixed under https://github.com/osresearch/heads/commit/30852aa2739e71ed8c2c889cae91ef3995cfe75a

commit 30852aa2 (2020-10-19): xx30 boards: suspend/resume fix (#838)

* x230-hotp-verification: revert to coreboot "CONFIG_CBFS_SIZE=0x700000" by disabling board "CONFIG_DROPBEAR=y" and "CONFIG_LINUX_E1000E" to save space per @alex-nitrokey test under #770 to fix #608
* x230-htop-verification board: add clarifications on x230 board differences. Fixes #737 #770 #608
* X230 board & coreboot config: Fix x230 board, removing dropbear and e1000e driver.
* t430: board and coreboot config par with x230 to circumvent CBFS linked suspend/resume issues.