linuxboot/heads #739: slang cannot be downloaded

tlaurion, 2020-06-06:
```
--2020-06-06 18:38:00--  https://www.jedsoft.org/releases/slang/slang-2.3.1a.tar.bz2

Resolving www.jedsoft.org (www.jedsoft.org)... 199.180.249.46

Connecting to www.jedsoft.org (www.jedsoft.org)|199.180.249.46|:443... failed: Connection refused.

make: *** [Makefile:382: /root/project/packages/slang-2.3.1a.tar.bz2] Error 1

```

tlaurion, 2020-06-06:
email sent at webmaster@jedsoft.org. 
Didn't found the lib anywhere else.

paulmenzel, 2020-06-07:
The administrators seems to have fixed it.

```
$ curl -I https://www.jedsoft.org/releases/slang/old/slang-2.3.1a.tar.bz2
HTTP/1.1 200 OK
Content-Type: application/x-gtar-compressed
Accept-Ranges: bytes
ETag: "971726120"
Last-Modified: Fri, 25 Nov 2016 02:36:45 GMT
Strict-Transport-Security: max-age=15768000; includeSubdomains
Content-Length: 1577749
Date: Sun, 07 Jun 2020 05:22:43 GMT
Server: lighttpd/1.4.53

```

By the way, distributions normally also offer the source archives. For different versions, Debian’s snapshot service comes in handy: https://snapshot.debian.org/package/slang2/.

PS: [Version 2.3.2 seems to be current.](https://www.jedsoft.org/releases/slang/)

tlaurion, 2020-06-07:
@paulmenzel : an effort to upgrade toolstack is going to happen once #590 has given results or another path is chosen (LTO seems to be the candidate.)

closed 2020-06-07 by tlaurion