linuxboot/heads #740: Measured boot for KGPE-D16 not working

Tonux599, 2020-06-06:
Setting 'CONFIG_MEASURED_BOOT=y' in this boards config causes it to fail to build with the error below. Currently with a compatible TPM module only PCRs 4 and 7 are populated meaning that the boot block, ROM stage, RAM stage and the kernel/initrd are not measured and as such, the TPMTOTP will not reflect any changes.

```
tail /home/user/tonux599-heads/build/log/coreboot.log
-----
    CC         ramstage/southbridge/amd/sb700/pci.o
    CC         ramstage/southbridge/amd/sb700/ramtop.o
    CC         ramstage/southbridge/amd/sb700/reset.o
    CC         ramstage/southbridge/amd/sb700/sata.o
    CC         ramstage/southbridge/amd/sb700/sb700.o
    CC         ramstage/southbridge/amd/sb700/sm.o
    CC         ramstage/southbridge/amd/sb700/smbus.o
    CC         ramstage/southbridge/amd/sb700/spi.o
    CC         ramstage/southbridge/amd/sb700/usb.o
    CC         ramstage/security/tpm/tss/tcg-2.0/tss.o
src/security/tpm/tss/tcg-2.0/tss.c:309:10: error: conflicting types for 'tlcl_define_space'
 uint32_t tlcl_define_space(uint32_t space_index, size_t space_size)
          ^~~~~~~~~~~~~~~~~
In file included from src/security/tpm/tss/tcg-2.0/tss_structures.h:18,
                 from src/security/tpm/tss/tcg-2.0/tss.c:15:
src/security/tpm/tss.h:65:10: note: previous declaration of 'tlcl_define_space' was here
 uint32_t tlcl_define_space(uint32_t index, uint32_t perm, uint32_t size);
          ^~~~~~~~~~~~~~~~~
make[1]: *** [Makefile:356: kgpe-d16-workstation-nouveau/ramstage/security/tpm/tss/tcg-2.0/tss.o] Error 1
```

Tonux599, 2020-06-06:
Looking in `ramstage/security/tpm/tss/` shows folders `tcg-1.2` and `tcg-2.0` which I'm assuming relates to TPM1.2 & TPM2.0. The build is failing when dealing with the TPM2.0 folder so just need to stop it from doing that as only TPM1.2 is supported anyway. @flawedworld I noticed in #442 looks very similar to this. Any tips?

Tonux599, 2020-06-06:
Commenting out `select TPM2` in the boards Kconfig gives a successful build. Will test and report back.

flawedworld, 2020-06-07:
@Tonux599 The solution for T430 was to backport the TPM - see https://github.com/osresearch/heads/commit/5a033fa80dc8d6134fd31f6a8077e1b10b370328

tlaurion, 2020-06-07:
> Commenting out `select TPM2` in the boards Kconfig gives a successful build. Will test and report back.
> Edit: ROM works but PCRs 0-3 not populated. Measured boot still not functional.

@Tonux599 gave the most love I could to #472 . Would love to hear your findings for TPM 1.2 support. Buying a module myself now.

Tonux599, 2020-06-08:
CBMEM log reveals a bit, TPM is initialised okay. Its when PCRs are extended that it fails. See below.
```
TPM initialization.
TPM: Init
Found TPM SLB9635 TT 1.2 by Infineon
TPM: Open
TPM: Startup
TPM: command 0x99 returned 0x0
TPM: OK.
```
```
TPM: pcr 3 measure ff02b338 @ 3524: a1615718de98d6d619812c6b9fd3c3f5177bc137
src/drivers/pc80/tpm/tis.c:515 unexpected TPM status 0xff
src/drivers/pc80/tpm/tis.c:711 failed sending data to TPM
TPM: command 0x14 send/receive failed: 0x10000001
```
@tlaurion this is going out my expertise a bit so if you have any input or know anyone please tag :)

tlaurion, 2020-06-09:
@Tonux599 please link your branch so I can review.

Tonux599, 2020-06-09:
> @Tonux599 please link your branch so I can review. Intuition is that coreboot config being at fault.

[This](https://github.com/Tonux599/heads/tree/kgpe-d16-workstation-tpm-testing) branch outlines my current efforts. Thanks @tlaurion

tlaurion, 2020-06-09:
> @tlaurion is TPM on stock? will try to search again it

@0rb677 https://www.ebay.ca/itm/Asus-TPM-Module-20-pin-1-02G-ver-1-2-compatible/114247958589

tlaurion, 2020-06-09:
@Tonux599 https://github.com/osresearch/heads/issues/712#issuecomment-641424861

Tonux599, 2020-06-09:
@tlaurion just a heads up the module I use is 1.02H, not sure if there is any difference.
Bit more info in cbmem from setting `CONFIG_DEBUG_TPM=y`, see below. Thought: can PCRs be different sizes on different modules? I'm wondering if the sha1 sum is too big to fit in the PCR.
```
TPM initialization.
TPM: Init
lpc_tpm: Read reg 0xf00 returns 0xb15d1
Found TPM SLB9635 TT 1.2 by Infineon
TPM: Open
lpc_tpm: Read reg 0x0 returns 0x81
lpc_tpm: Write reg 0x0 with 0x2
lpc_tpm: Read reg 0x0 returns 0xa1
lpc_tpm: Write reg 0x18 with 0x40
lpc_tpm: Read reg 0x18 returns 0x40
TPM: Startup
lpc_tpm: Read reg 0x18 returns 0x40
lpc_tpm: Read reg 0x19 returns 0x8
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0xc1
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0xc
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Read reg 0x18 returns 0x8
lpc_tpm: Read reg 0x18 returns 0x88
lpc_tpm: Read reg 0x18 returns 0x88
lpc_tpm: Read reg 0x19 returns 0x8
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x99
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Read reg 0x18 returns 0x8
lpc_tpm: Read reg 0x18 returns 0x88
lpc_tpm: Read reg 0x18 returns 0x88
lpc_tpm: Read reg 0x19 returns 0x8
lpc_tpm: Write reg 0x18 with 0x1
lpc_tpm: Read reg 0x18 returns 0x8
lpc_tpm: Read reg 0x18 returns 0x80
lpc_tpm: Read reg 0x18 returns 0x80
lpc_tpm: Write reg 0x18 with 0x20
lpc_tpm: Read reg 0x18 returns 0x80
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x90
lpc_tpm: Read reg 0x19 returns 0x8
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x24 returns 0xc4
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x24 returns 0xa
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x90
lpc_tpm: Read reg 0x18 returns 0x90
lpc_tpm: Read reg 0x19 returns 0x2
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x24 returns 0x0
lpc_tpm: Read reg 0x18 returns 0x80
lpc_tpm: Read reg 0x18 returns 0x80
lpc_tpm: Write reg 0x18 with 0x40
lpc_tpm: Read reg 0x18 returns 0x80
lpc_tpm: Write reg 0x18 with 0x40
lpc_tpm: Read reg 0x18 returns 0x40
TPM: command 0x99 returned 0x0
lpc_tpm: Read reg 0x0 returns 0xa1
lpc_tpm: Write reg 0x0 with 0x20
lpc_tpm: Read reg 0x0 returns 0x81
TPM: OK.
```
```
TPM: pcr 3 measure ff02b578 @ 3524: a1615718de98d6d619812c6b9fd3c3f5177bc137
lpc_tpm: Read reg 0x18 returns 0xff
lpc_tpm: Read reg 0x19 returns 0xffff
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0xc1
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x22
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x14
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x0
lpc_tpm: Write reg 0x18 with 0x3
lpc_tpm: Write reg 0x18 with 0xa1
lpc_tpm: Write reg 0x18 with 0x61
lpc_tpm: Write reg 0x18 with 0x57
lpc_tpm: Write reg 0x18 with 0x18
lpc_tpm: Write reg 0x18 with 0xde
lpc_tpm: Write reg 0x18 with 0x98
lpc_tpm: Write reg 0x18 with 0xd6
lpc_tpm: Write reg 0x18 with 0xd6
lpc_tpm: Write reg 0x18 with 0x19
lpc_tpm: Write reg 0x18 with 0x81
lpc_tpm: Write reg 0x18 with 0x2c
lpc_tpm: Write reg 0x18 with 0x6b
lpc_tpm: Write reg 0x18 with 0x9f
lpc_tpm: Write reg 0x18 with 0xd3
lpc_tpm: Write reg 0x18 with 0xc3
lpc_tpm: Write reg 0x18 with 0xf5
lpc_tpm: Write reg 0x18 with 0x17
lpc_tpm: Write reg 0x18 with 0x7b
lpc_tpm: Write reg 0x18 with 0xc1
lpc_tpm: Read reg 0x18 returns 0xff
lpc_tpm: Read reg 0x18 returns 0xff
lpc_tpm: Read reg 0x19 returns 0xffff
lpc_tpm: Write reg 0x18 with 0x37
lpc_tpm: Read reg 0x18 returns 0xff
lpc_tpm: Read reg 0x18 returns 0xff
lpc_tpm: Read reg 0x18 returns 0xff
src/drivers/pc80/tpm/tis.c:515 unexpected TPM status 0xff
src/drivers/pc80/tpm/tis.c:711 failed sending data to TPM
TPM: command 0x14 send/receive failed: 0x10000001
```

Tonux599, 2020-06-09:
From #712 
> Ordered the TPM 1.2 module off of ebay and should receive it prior of end of June. Will get back at #740 once received, but haven't seen any coreboot changes in the config there for infineon support? Maybe I missed it.

Will investigate however from casual looking it looks like there shouldn't be any additional options to support infineon TPM as I assume that all comply with the same standard.

0rb677, 2020-06-11:
> > @tlaurion is TPM on stock? will try to search again it
> 
> @0rb677 https://www.ebay.ca/itm/Asus-TPM-Module-20-pin-1-02G-ver-1-2-compatible/114247958589 is what I bought, delivered before end of June.

Bought it too.
```
Estimated delivery:  Sun, Jun 28–Wed, Jul 15
```

tlaurion, 2020-08-06:
@Tonux599 https://github.com/osresearch/heads/pull/472#issuecomment-670179016

tlaurion, 2020-08-07:
Seems like it would be possible to exclude mrc from measurements in coreboot config linked to a board.

Tonux599, 2020-12-03:
Fixed with #867 which has now been merged.

closed 2020-12-03 by Tonux599