linuxboot/heads #745: OEM Factory Reset - operation cannot be done while there is no TPM on the motherboard.

0rb677, 2020-06-09:
@tlaurion

[image]

https://github.com/osresearch/heads/blame/master/initrd/bin/oem-factory-reset#L125-L140
https://github.com/osresearch/heads/blob/master/initrd/bin/config-gui.sh#L166-L168

tlaurion, 2020-06-09:
@0rb677 Congrats, you found a bug where[ CONFIG_TPM is not validated](https://github.com/osresearch/heads/blame/master/initrd/bin/oem-factory-reset#L125-L140) as everywhere [else in the code base prior to operation](https://github.com/osresearch/heads/blob/master/initrd/bin/config-gui.sh#L166-L168)

tlaurion, 2020-06-09:
@MrChromebox ?

MrChromebox, 2020-06-09:
@tlaurion I'd fixed that locally and forgot to push a PR, I'll do that shortly

MrChromebox, 2020-06-09:
addressed by #747

tlaurion, 2020-06-09:
@0rb677 : merged into https://github.com/osresearch/heads/pull/472 Please report success/fail!

0rb677, 2020-06-11:
@tlaurion  OEM FACTORY RESET can't reset gpg-key, on x220 it works.

0rb677, 2020-06-11:
Clear process on x220

[image]

tlaurion, 2020-06-11:
> @tlaurion OEM FACTORY RESET can't reset gpg-key, on x220 it works.

@0rb677 
The x220 has CONFIG_TPM=y.
kgpe-d16 has CONFIG_TPM=n

https://github.com/osresearch/heads/blob/cbad9b663724c95508f0eec027841d93c723c006/initrd/bin/oem-factory-reset#L311

Implemented this validation. Screenshots provided, coming from x220, is not testing previous PR.
Which is why it was advised to test new artifact on kgpe-d16 from #472 [artifacts](https://circleci.com/gh/tlaurion/heads/263?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link)

0rb677, 2020-06-11:
@tlaurion ok re-building.

0rb677, 2020-06-11:
@tlaurion 

I dont understand it works with couple of bugs.
Now i attached gpg-card  and external USB to PC when heads initialized. Then quit to shell and lauch oem-reset script.

1. It access to gpg-card
2. It generates oem-key
3. It changes Admin/Pin
4. Exporting key to USB
5. Reading Firmware
6. Adding gerenerate key to current firmware and re-flashing
7. Signed boot files and generate checksums

So it works?

0rb677, 2020-06-11:
@tlaurion congrats it done.

0rb677, 2020-06-11:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

tlaurion, 2020-06-11:
I'm confused with things unrelated to this ticket and not sure of your state with contradictory information,

If I recall well, your use case made me create:
boards/kgpe-d16_workstation-usb_keyboard/kgpe-d16_workstation-usb_keyboard.conf

so make `BOARD=kgpe-d16_workstation-usb_keyboard` which on a clear checkout should be at latest commit:
```
git fetch tlaution
git checkout  kgpe-d16_current_working
git reset --hard
git log
```
should show
```
Merge: a3de462 cbad9b6
Author: Thierry Laurion <insurgo@riseup.net>
Date:   Tue Jun 9 13:59:38 2020 -0400

    Merge remote-tracking branch 'mrchromebox/factory_reset_no_tpm' into kgpe-d16_current_working

commit cbad9b663724c95508f0eec027841d93c723c006 (mrchromebox/factory_reset_no_tpm)
```

0rb677, 2020-06-11:
@tlaurion can you enjoy to slack?

tlaurion, 2020-06-11:
> @tlaurion i cant boot from `Default boot Menu` and /boot is signed by oem-key, also rom is signed too.
> 
> [image]

If you flashed a new public key, you changed your rom, measurements are now invalid.
You have to:

- Resign your configs
- Set a new boot default

Screenshots are always helpful.

0rb677, 2020-06-11:
@tlaurion can you join to channel? It too hard to answer you in so many threads.

tlaurion, 2020-06-11:
so @0rb677 is having different behaviors when calling the script from command line vs from the GUI. The command line calls are successful, while doing it from the GUI results in this:
https://github.com/osresearch/heads/blob/master/initrd/bin/oem-factory-reset#L66-L68

with `/tmp/gpg_card_edit_output` being empty and uninformative.

@MrChromebox : any idea why GUI calling /bin/oem-factory-reset is different then from console calling it, which is functional?

MrChromebox, 2020-06-11:
@tlaurion nothing jumps out at me, will have to take a look

tlaurion, 2020-12-02:
@0rb677 is that still relevant?

tlaurion, 2020-12-02:
@0rb677 the kgpe-d16 board is now TPM enabled. So if you want a TPM-less board, you will have to disable TPM inside of the board configs on master.

tlaurion, 2020-12-02:
Please close this issue or tag me back if still relevant.

MrChromebox, 2020-12-02:
@tlaurion are you still opposed to setting CONFIG_TPM dynamically upon boot as I do in PR #836 ?

tlaurion, 2020-12-03:
@MrChromebox Im against rubber ducky USB dongle ressembling nitokey/librem key able to sed everything terminal in a glimpse of an eye and make Heads lie about its real state as proposed in that PR, yes.

For TPM being dynamic, I have nothing against that specific part of that PR following your point there.

MrChromebox, 2020-12-03:
@tlaurion we have in fact many older Librem 13v2/v3 models which do not have a TPM, as well as ones that do. Having a separate board definition for those is a logistical nightmare, because many times the users aren't even aware. I'll resubmit the part of the patch setting CONFIG_TPM dynamically at boot as a separate PR if you agree, otherwise I'll just keep the patch in Purism's tree for now

closed 2020-12-04 by 0rb677