linuxboot/heads #746: Upgrade nitrokey-hotp-verification module

tlaurion, 2020-06-09:
@szszszsz upgrade module/nitrokey-hotp-verification with https://github.com/Nitrokey/nitrokey-hotp-verification/commit/1cf1e1f3601b79692b81d409597136b8ff7c723a

tlaurion, 2020-06-09:
@szszszsz : I would recommend renaming the module to nitrokey-hotp-verification and Makefile accordingly so that users know who is the clear maintainer of that module.

tlaurion, 2020-06-09:
I will do the housekeeping for other boards.

szszszsz, 2020-06-09:
Indeed. Maybe the more generic name could be used, like `security-key-hotp-verification`, but that's a detail.

cc: @kylerankin @jans23 @Nitrokey

kylerankin, 2020-06-09:
I like the idea of renaming it to be a generic name like
'security-key-hotp-verification' since now both Librem Keys and Nitrokeys
work, and hopefully one day even more keys could work!

tlaurion, 2020-06-09:
The token words that were accepted were: 
- "USB security dongle" to talk about GPG supporting cards, as opposed to keys which are known to be the numeric result of them per unmerged PR of Heads documentation.
- "HOTP USB security dongle" to talk about supported HOTP devices under heads.

alex-nitrokey, 2020-06-10:
I would like to try to add a PR these days that makes sure that 

* HOTP USB Security dongle is default (with a `CONFIG_HOTPKEY` or alike)
* there is a possibility to use a branding (either Nitrokey or Libremkey) via `CONFIG_HOTPKEY_BRANDING` or alike

while it is totally annoying to handle multiple branding in code (it really is!) I think it has a major advantage if users know which device to put in (for both, users of Librem Notebook and Libremkey and users that use another combination).

What do you think about this approach?

tlaurion, 2020-06-11:
> I would like to try to add a PR these days that makes sure that
> 
>     * HOTP USB Security dongle is default (with a `CONFIG_HOTPKEY` or alike)

CONFIG_HOTP_USB_SECURITY_DONGLE seems fit.
> 
>     * there is a possibility to use a branding (either Nitrokey or Libremkey) via `CONFIG_HOTPKEY_BRANDING` or alike

As previously discussed in Nitrokey attempt to detect the keys per hardware ID, there was this idea of having the branding stored inside of /boot prior of signing with GPG.

The approach of binding this in boards configs forces maintainers to split boards once more, having _LIBREMKEY and _NITROKEY flavors for each HOTP supporting boards. A better solution would be required. Isn't there any other way to distinguish dongles from firmware fingerprints?
> 
> 
> while it is totally annoying to handle multiple branding in code (it really is!) I think it has a major advantage if users know which device to put in (for both, users of Librem Notebook and Libremkey and users that use another combination).
> 
> What do you think about this approach?

PR #748 (2020-07-23): Update hotp-verification (fixes #746)
- [x] use newer version of hotp-verification module to avoid patch
- [x] testing new version on board

What do you think, shall I make two separate PRs instead?

(will fix #746)
closed 2020-07-23 by tlaurion