linuxboot/heads #754: Define target `clean` in Makefile

techge, 2020-06-18:
As mentioned in #752 it would be nice to have a `make clean` to prevent pitfalls while trying to build heads and maybe even for CI later on.

tlaurion, 2020-06-18:
@techge https://github.com/osresearch/heads-wiki/blame/master/Building.md#L152-L155

Not enough?

techge, 2020-06-18:
I see. Not sure though, would this clean `install/include` as well?

techge, 2020-06-18:
Okay, nevermind. Now I am only wondering if `make clean` would be more suitable for dummys :(

closed 2020-06-18 by techge

tlaurion, 2020-06-18:
> I see. Not sure though, would this clean `install/include` as well?

```
real.clean:
        for dir in \
                $(module_dirs) \
                $(kernel_headers) \
        ; do \
                if [ ! -z "$$dir" ]; then \
                        rm -rf "build/$$dir"; \
                fi; \
        done
        cd install && rm -rf -- *

```
```

modules.clean:
        for dir in $(module_dirs) \
        ; do \
                $(MAKE) -C "build/$$dir" clean ; \
                rm "build/$$dir/.configured" ; \
        done

```

Per https://github.com/osresearch/heads-wiki/blob/master/Building.md#helpful-targets-and-options