linuxboot/heads #757: Upgrade Kernel to 5.7.4

ankit-singh3, 2020-06-22:
Hi,

Looking to enable WIFI for "Intel® Wi-Fi 6 AX201 160MHz" as per https://www.intel.in/content/www/in/en/support/articles/000005511/network-and-i-o/wireless-networking.html.
But it's supported only for Kernel 5.2+ and hence want to upgrade this to 5.7.4 (current kernel version here is 4.14.62)
Have updated the below items:-
1.) Added linux-5.7.4.tar.xz tolinuxboot/packages (downloaded from http://cdn.kernel.org/pub/linux/kernel/v5.x/)
2.) Edited linuxboot/modules/linux file with new kernel version and Hash value
3.) Added patch file to linuxboot/patches (again downloaded from http://cdn.kernel.org/pub/linux/kernel/v5.x/) 

After all these changes its giving an compilation error as below:-
    --> Makefile:382: recipe for target '/home/ankit_singh3/code/EMBEDDED_OS1/linuxboot/build/linux-5.7.4/.canary' failed
    --> make: *** [/home/ankit_singh3/code/EMBEDDED_OS1/linuxboot/build/linux-5.7.4/.canary] Error 1
    --> make: Leaving directory '/home/ankit_singh3/code/EMBEDDED_OS1/linuxboot'

**Please let me know if i have missed anything as am bit new to this.**

Regards,
Ankit Singh**

Tonux599, 2020-06-26:
@ankit-singh3 why do you wish to enable WiFi in heads?

ankit-singh3, 2020-06-26:
> 
> 
> @ankit-singh3 why do you wish to enable WiFi in heads?

Hi @Tonux599 ,
Need to download some files over HTTPS on laptop which might not be on wired network all the time.
Am able to enable WIFI on some old cards but "Intel® Wi-Fi 6 AX201 160MHz" is a new card and required new version for kernel as per Intel Specification.

Tonux599, 2020-07-02:
I assume you wish to do that from your Linux distribution as opposed to the heads runtime itself? In which case heads can kexec into any kernel version.

tlaurion, 2020-07-22:
@ankit-singh3 : I do not understand why you would want wifi under Heads' kernel either.

There is a need to upgrade the kernel (sdcard reader not supported under current kernel for X230 on some models, KGPE-D16 Aspeed integrated card support way better supported on newer kernel), while updating kernel requires a lot of testing and size reduction.... 

The next kernel upgrade would ideally integrate [LTO patchset](https://github.com/osresearch/heads/pull/730).

 @ankit-singh3 : Please close the issue if what you intended was to get your wifi card supported in OS and not under Heads. If wanted under Heads, you have to be conscious of the requirement for wpa_supplicant and other requirements, which may never land under Heads for #590.

tlaurion, 2020-07-22:
Actually, please reopen if you consider pertinent with proper justifications.

closed 2020-07-22 by tlaurion