linuxboot/heads #764: oem-factory-reset failing during gpg factory-reset

alex-nitrokey, 2020-06-30:
The oem-factory-reset is failing for me if started directly after starting the device (e.g. after flashing), but is working fine if you call `oem-factory-reset` out of the recovery shell.

Did anybody experience this as well? I couldn't see the reason yet. It seems to fail at [this position](https://github.com/osresearch/heads/blob/8dc5b7616aec2586c4a3bd59ac8cfca1cfa01e91/initrd/bin/oem-factory-reset#L71).

MrChromebox, 2020-06-30:
it's failing for me regardless if started from the boot prompt or manually from the menu; I believe this started when #646 was merged

alex-nitrokey, 2020-06-30:
> I believe this started when #646 was merged

I was thinking the same - yet, I know it did work :dagger: 

> it's failing for me regardless if started from the boot prompt or manually from the menu;

please try going into recovery shell and type `oem-factory-reset` from there. This should work. Though the GnuPG commands are output which was not the case in the past. This is why I believe that updating gpg2 could be the cause here.

MrChromebox, 2020-06-30:
> please try going into recovery shell and type oem-factory-reset from there. This should work.

yes, see my edit :)

> This is why I believe that updating gpg2 could be the cause here.

will have to try reverting that to isolate, then can diagnose

tlaurion, 2020-07-02:
@alex-nitrokey : I would prefer having mount-usb from preceded with a prompt prior of exporting the keys and keep factory-reset outputing normally to /media if not already mounted.

I will merge my changes to that script in subsequent PR for the reownership wizard to work with upstreamed code, and to me it doesn't really make sense to have a factory reset without exporting the public key.

techge, 2020-07-03:
> @alex-nitrokey : I would prefer having mount-usb from preceded with a prompt prior of exporting the keys and keep factory-reset outputing normally to /media if not already mounted.
I will merge my changes to that script in subsequent PR for the reownership wizard to work with upstreamed code, and to me it doesn't really make sense to have a factory reset without exporting the public key.

This was intended for #766 , wasn't it?

It is just a suggestion. In our experience many people do not care about the GnuPG keys and just want the boot measurement, thus having a pubkey is not always needed or even confusing and the factory-reset aborts if you do not put in an usb drive... But I totally see that this might be controversial. I just wanted to suggest the change at least as we have implemented for NitroPad.

PS: do not be confused, I am logged in with another account (thus, not alex-nitrokey).

MrChromebox, 2020-07-07:
confirmed, reverting 972c25de7df9cf797fd211bc795fe057462a6432 [upgrade gpg toolstack to latest versions] fixes the issue

tlaurion, 2020-07-08:
Which means wthe following line is at fault since gpg modified something:
https://github.com/osresearch/heads/blob/8dc5b7616aec2586c4a3bd59ac8cfca1cfa01e91/initrd/bin/oem-factory-reset#L67

techge, 2020-07-09:
> Which means the following line is at fault since gpg modified something:

Yes, it appears to me that this is right.

tlaurion, 2020-07-09:
So this might be linked with TTYs being foo-bar since moved to musl-cross-make. ( #665 ) and the fix having tty hardcoded

tlaurion, 2020-07-09:
> but is working fine if you call oem-factory-reset out of the recovery shell.

@alex-nitrokey I don't get that part though

techge, 2020-07-10:
As far as I remember, all the output of GnuPG was suppressed back then when it still worked. If you open the script via recovery shell, you see all the output of GnuPG while the process is still automated.

Therefore, I guess it is something in the way we are handling the GnuPG output, but I could not see what is going wrong there yet.

tlaurion, 2020-08-01:
will be resolved by #777

closed 2020-08-01 by tlaurion