linuxboot/heads #770: x230-hotp-verification: not waking from suspend

stmc-droid, 2020-07-03:
After suspending a x230 when trying to wake it does a hard restart instead. I am using the latest x230-hotp-verification build. Does anyone else have the same issue? I tested with Qubes and Tails.

lsafd, 2020-07-06:
Also having the same issue. 

I would recommend reading through #608 if you haven't already. These suspend issues may be related to the version of Coreboot used by Heads (This is just secondhand from that issue).

tlaurion, 2020-07-06:
Which is funny since I do not have that issue in my old artifact, deployed with the [PrivacyBeast as per QubesOS certified firmware](https://gitlab.com/tlaurion/heads/-/jobs/312958349/artifacts/raw/build/x230-libremkey/coreboot.rom?inline=false?job=build). Coreboot version is the same as used here, so not the cause.

I have tested latest build and can confirm the issue.
@MrChromebox I have seen that you have fixe similar issues in the past. 

You have a troubleshooting path to recommend?

MrChromebox, 2020-07-07:
@tlaurion the issue you linked was due to a change in Chrome EC event handling on the coreboot side which broke things on certain ChromeOS devices

tlaurion, 2020-08-12:
@MrChromebox @jans23 @osresearch @PatrickRudolph : see comment https://github.com/osresearch/heads/pull/721#issuecomment-673115540

Coreboot 4.12 doesn't fix this issue

jans23, 2020-08-13:
@szszszsz Any ideas?

PatrickRudolph, 2020-08-13:
Please provide more details.
Which resume doesn't work? S3? S4?
Please provide a log of a regular boot and a failing S3/S4 wake up boot.

The problem can be a invalid MRC cache, changed e820 memory map during regular boot vs wake up boot, touched hardware that shouldn't be touched (Intel IGD), ...

tlaurion, 2020-08-13:
S3.

Not sure how I could provide a log for failing since the machine reboots as if the power button was a reset button, where the half moon led showing the laptop is under sleep is still on.

@PatrickRudolph you can flash the referred ROMs for testing.

alex-nitrokey, 2020-09-14:
@tlaurion are you sure this issue started when changing to musl-cross-make? I am surprised that this did not caught our attention earlier then...

I am not sure how to debug this either. But I was wondering if this might be related to cbfs size as well, as it is happening on hotp-verification build and not the `x230`, right?

When adapting the cbfs size after failed boots ( #737 ), we just assumed a size that is okay. So maybe most stuff *seemed* to work, but not everything is okay - like this issue, maybe even #765 .

Thus, what I was trying is the following:
* setting `CONFIG_CBFS_SIZE=0x700000` in `config/coreboot-x230-hotp-verification.config`
* disabling `CONFIG_DROPBEAR` and `CONFIG_LINUX_E1000E` to save some space

This build resulted in a "suspendable" version of `x230-hotp-verification`. Would be glad for a confirmation of my observation.

tlaurion, 2020-09-17:
Will look into this, thanks for the report.

tlaurion, 2020-09-19:
Successfull build per https://github.com/tlaurion/heads/commit/b1886f3bf2be0f2f8ed8dab19da5fec459a34d41
Direct link to ROM to reproduce and comment: https://430-103208611-gh.circle-artifacts.com/0/build/x230-hotp-verification/coreboot.rom

If it works, it means that the maximum space available in CBFS for BIOS is really 0x700000 (I remember I did the calculations somewhere under Heads....)

I'll try to test in next test batch @alex-nitrokey

alex-nitrokey, 2020-09-23:
> Successfull build per tlaurion@b1886f3

I just tested the build. Suspend works fine here!

> EDIT: explanation is here

Thanks! I vaguely remembered this post. Good work!

tlaurion, 2020-10-17:
Will be fixed by #838

tlaurion, 2020-10-19:
fixed under https://github.com/osresearch/heads/commit/30852aa2739e71ed8c2c889cae91ef3995cfe75a

closed 2020-10-19 by tlaurion