linuxboot/heads #771: GPG keypair generation from Heads, then copied to USB Security dongle

tlaurion, 2020-07-05:
Current customer concerns:
- What do I do if I loose the USB Security dongle
- What do I do if I lock myself out of the USB Security dongle

Solutions:
- [ ] Have the keypair generated inside of Heads then [copied to USB Security dongle](https://docs.puri.sm/Librem_Key/Getting_Started/User_Manual.html#generate-gpg-keys-on-your-computer)
- [ ] Have the factory reset code also generate a Reset Code (that permits to unlock Admin PIN)

Quick fix:
- [ ]  Admin PIN size should be 20 characters maximum

tlaurion, 2020-07-05:
@alex-nitrokey @kylerankin @MrChromebox : Thoughts?

alex-nitrokey, 2020-07-06:
As far as I can see, the function of the dongle is to detect unauthorized alternation of the firmware/everything in /boot. So, losing/breaking the dongle does solely mean that the user needs a new device to be able to detect such change on the system again. The problem arises if the dongle gets lost during a situation in which the pc was unattended and exposed as well.

One could implement a feature to prepare two dongles at once during oem factory reset to have a backup device.

Imho it is not useful to let the key be exported if you like to make Heads secure and easy to use for the innocent. If you really like a full backup of your GnuPG keys you already can prepare your dongle first and then provide the public key to Heads - though this is for the more experienced users.

I do not see how rising the PIN size should help here?

MrChromebox, 2020-07-07:
a second dongle isn't going to be a drop-in replacement since the counters will be out of sync, and you'd still have to generate a new secret with the backup dongle.

techge, 2020-07-07:
> a second dongle isn't going to be a drop-in replacement since the counters will be out of sync

A right, forgot that.

> and you'd still have to generate a new secret with the backup dongle.

This would have been solvable by copying the secret to both dongles, I guess.

tlaurion, 2020-07-09:
The goal here is really to be loss-proof from encrypted mails perspective. I think it is ok to accept the user to have to validate with Disk Unlock Key code and or TOTP and not update dom0 before having secrets sealed through HOTP on new device with options provided below.

Menu options:
- [ ] Heads keypair generation + key copied to smartcard + PINs generation + Reseal HOTP secret
- [ ] Copy keys to smartcard + USB Security dongle PIN generation + Reseal HOTP secret

Thoughts?

alex-nitrokey, 2020-07-15:
I would adapt this to 

* [ ] keypair generation in GnuPG + keytocard command in GnuPG + (maybe export +) shred keys + PINs generation + Reseal HOTP secret
* [ ] Generate keys on smartcard + PIN generation + Reseal HOTP secret

That's maybe nitpicking, but I want to make sure the process is described as I see it.

Please note that [keypair generation and copying to (multiple) devices](https://www.nitrokey.com/documentation/openpgp-create-backup) would be a process that includes a lot of screenscraping in GnuPG which might be error prone. Honestly, I really think "power users" shall just do this themselves and copy a public key to the BIOS like they can today (and I did it). John and Jane Doe may just use the key on the one and only USB Security dongle and thus do not need to think how to store a exported GnuPG key securely.

tlaurion, 2020-09-23:
@alex-nitrokey will attack that soon.

- [ ]  Generate GPG keypair under Heads + keytocard command in GnuPG + export to secure storage + shred keys + PINs generation + Reseal HOTP secret (Fast + Lost proof)
- [ ] Generate GPG keypair on smartcard + Export to secure storage + shred keys + PIN generation + Reseal HOTP secret (Slow + Lost proof)
- [ ] Generate GPG keypair on smartcard + PIN generation + Reseal HOTP secret (Risky: No private key backup)

> Please note that keypair generation and copying to (multiple) devices would be a process that includes a lot of screenscraping in GnuPG which might be error prone. 

Screecaping should not be a problem unless we change versions of GPG, which should come with regression testing if we learn from our errors this time.

> Honestly, I really think "power users" shall just do this themselves and copy a public key to the BIOS like they can today (and I did it). 

The problem, and a  lot of request (1 per week at least) is the user don't understand the risks of loosing their USB Security dongle. They actually believe wrongly that Nitrokey [Pro/Sotorage]/Librem Key permit private key backup, and do not understand that once their key is lost, they lose decryption of all their past encrypted mails.... Until they loose their dongle.

> John and Jane Doe may just use the key on the one and only USB Security dongle and thus do not need to think how to store a exported GnuPG key securely.

I do not understand your point considering real risks of loosing the physical dongle here. Please clarify.

jans23, 2020-09-24:
I agree that such process should not require a 2nd computer.

For clarification: IMO "export to secure storage" in the first two paragraphs refer to two different things. In the first paragraph it means exporting **private** key and in second paragraph it means exporting **public** key. Right?

If the purpose of the LUKS-encrypted drive is only to store the private PGP key, I suggest to use GPG encryption instead. It should be easier to implement and easier to use for the user (she can copy the encrypted file to whatever other backup media she prefers).

alex-nitrokey, 2020-09-24:
I totally see your point in providing a all-in-one-failsafe solution for the innocent user. I guess my problem is just that I am not sure if things are really getting better. In the end it is a matter of taste. That is to say, I do not want to argue *against* you proposal in any way (it should be doable), it is just not the path I would use.

I personally think that the whole process is getting just more complicated for users without technical knowledge. I opt to losing data/keys instead of letting users do bad stuff. They already need to remember/know the difference of User PIN, Admin PIN, (TPM password, theoretically), passphrase for the LUKS disk on the machine - and now the passphrase for the key backup. They all should be different. I see more potential data loss in losing these or in willingly typing these in inappropriate situations than in losing/breaking the smartcard.

> I do not understand your point considering real risks of loosing the physical dongle here but if you consider a USB Security dongle to be useable only for HOTP, which most users would use for other means if they knew prior of the fact. Please clarify.

I guess I just meant the fact that they do not need to think about how to create a secure usb drive on which the backup key is stored. As far as I can see the process you described above would somehow solve this issue already.

Generally, I think the process you describe is legitimate. If I would go for a solution to create a backup I would go either in a similar way you did or I would propose configuring *two smart cards* at once - thus having a smart card-protected backup. In any case a user should be informed in detail what the security implications of a key-on-encrypted-disk vs. key-on-smartcard are.

jans23, 2020-09-24:
> If I would go for a solution to create a backup I would go either in a similar way you did or I would propose configuring two smart cards at once - thus having a smart card-protected backup.

I support that. This has been requested by some of our users already.

tlaurion, 2020-09-24:
> I agree that such process should not require a 2nd computer.
> 
> For clarification: IMO "export to secure storage" in the first two paragraphs refer to two different things. In the first paragraph it means exporting **private** key and in second paragraph it means exporting **public** key. Right?

Yes. Sorry for the confusion. Saving the private key requires secured storage, while public key can be saved, and should be saved, on public partition.

> 
> If the purpose of the LUKS-encrypted drive is only to store the private PGP key, I suggest to use GPG encryption instead. It should be easier to implement and easier to use for the user (she can copy the encrypted file to whatever other backup media she prefers).

This solution, while seeming more logic, doesn't fit the use case of being able to restore the private key to be restored. The Heads GUI, resposible to restore the private key, would expect a LUKS partition to find the private key under it, from where it would restore it. Having an encrypted gpg file would not resolve the issue when key is lost to be able to provision a new USB Security dongle.

tlaurion, 2020-09-24:
> I agree that such process should not require a 2nd computer.

Awesome
> 
> For clarification: IMO "export to secure storage" in the first two paragraphs refer to two different things. In the first paragraph it means exporting **private** key and in second paragraph it means exporting **public** key. Right?

Yes. Sorry for the confusion. Saving the private key requires secured storage (non linked to secure storage linked to private that is to be restored by copytocard later on), while public key can be saved, and should be saved, on public partition.

> 
> If the purpose of the LUKS-encrypted drive is only to store the private PGP key, I suggest to use GPG encryption instead. It should be easier to implement and easier to use for the user (she can copy the encrypted file to whatever other backup media she prefers).

This solution, while seeming more logic, doesn't fit the use case of being able to restore the private key to be restored to do a copytocard. LUKS is supported everywhere, even on windows, and onve again, the goal here is for the user to be guided in a programmatic way to have its private key restored on a new USB Security dongle in case the prior one is lost (from Heads, again)

The Heads GUI, responsible to restore the private key, would expect a LUKS partition to find the private key under it once unlocked and mounted, from where it would be able to restore it on a new USB Security dongle ordered and received. Having an encrypted gpg file would not resolve the issue (current need) when private key is lost to be able to provision a new USB Security dongle.

The risks we really try to mitigate here are:
- Do not store the private key on untrusted media
- USB Security dongle being lost resulting in encrypted disks and emails staing in encrypted form forever
- Loss of signing identity

Between the moment of the USB Security dongle being lost and the moment the new ordered one is received, the user can:
- Not update dom0 and rely on Disk Unlock Key passphrase to attest firmware integrity/boot integrity
- Update dom0 but boot insecurely his laptop
- Not be able to decrypt new received encrypted mail

Until he goes into the new heads menu and select the `Restore private key under new USB Security dongle + change PINs + reseal + resign` option or something similar.

tlaurion, 2020-09-24:
@alex-nitrokey 
> Generally, I think the process you describe is legitimate. If I would go for a solution to create a backup I would go either in a similar way you did or I would propose configuring _two smart cards_ at once - thus having a smart card-protected backup. 

Asking the user to buy two Nitrokey Storage i nthe idea of loosing the key is something I would think psychologically difficult for the user, after the fact. And if a user thinks of buying 2 keys because he might be disorganized, then the possibility of loosing 2 keys (instead of buying a lineyard and have a USB drive securely stored somewhere...) is problematic idea per se. Now he looses the second key and he still doesn't have a way to restore it, the private key not having been exported when it was actually accessible.

This path would once again use Heads to generate the keys and store it with copytocard, while not having a physical backup.

> In any case a user should be informed in detail what the security implications of a key-on-encrypted-disk vs. key-on-smartcard are.

I'm sorry I don't understand the point here since there is no difference for me or langage used is not on point.
If you are talking about smartcard based keypair generation vs Heads based keypair generation, you might have a point, where the normal risks here are again on prime number generation and entropy generation. As long as we do not rely on the vulnerable hardware extensions to do the work (blink blink Intel, not under the x230, not used by currently built GPG AFAIK) then there would be a difference in the kind of trust the user is required to have into the random number generator, let it be inside of the USB Security dongle/ Heads. 

Where there is no added risk in having the key also stored onto a secured storage.

The main points, for me, to move the keypair generation inside of Heads instead of having it inside of the smartcard are two folded:
- Lower keypair generation time of RSA 4096 bits (5 minutes each in card: 15 minutes)
- Have a backup of the private key inside a secured storage.

> I personally think that the whole process is getting just more complicated for users without technical knowledge. I opt to losing data/keys instead of letting users do bad stuff. They already need [...]

Well, personally, I would have deduplicated once more the Disk Recovery Key passphrase here (in reownership, which would take advantage of this codepath as well). So we would have:
- TPM passphrase (named PIN.... disagree) = GPG Admin passphrase (without augmenting risk since GPG Admin PIN is rate limited and kicks out user after 3 attempts)
- Disk Recovery Key passphrase = Disk Recovery Key passphrase for USB secured drive partition (User should never type that passphrase but in secured space, and boots his machine with Disk Unlock Key passphrase as of right now, or with GPG User PIN in Librem setups if I got this well)

No added risk here. The reownership wizard requires provisioning secrets to be inside of a LUKS container already. A user not coming from the reownership wizard would be asked to insert a blank USB thumb drive to have it prepared for that use case. He refuses? then warning: no backup of your private key will be possible. Please provide a USB drive to store your public key. User refuses, he can be warned with the possibility of exporting it in a later time.

tlaurion, 2020-09-24:
If we have agreement, I will start coding this, starting with #777

alex-nitrokey, 2020-09-28:
> If we have agreement, I will start coding this, starting with #777

> No added risk here. The reownership wizard requires provisioning secrets to be inside of a LUKS container already. A user not coming from the reownership wizard would be asked to insert a blank USB thumb drive to have it prepared for that use case. He refuses? then warning: no backup of your private key will be possible. Please provide a USB drive to store your public key. User refuses, he can be warned with the possibility of exporting it in a later time, which is the consequence of current code base.

I have no objections in having this workflow in heads. I might disagree with some design decisions, but I neither think they are dangerous or wrong nor do I feel like they would make the situation worse, on the contrary, I think it is generally a good idea that tackles an important issue. It probably all depends on the use case/threat model/user in mind. I prefer a simpler setup for innocent users, in which there is no backup and no additional usb drive as well.

So I am totally fine and glad with having this feature in heads.

PS:
> I'm sorry I don't understand the point here since there is no difference for me or langage used is not on point.
If you are talking about smartcard based keypair generation vs Heads based keypair generation, you might have a point, where the normal risks here are again on prime number generation and entropy generation. As long as we do not rely on the vulnerable hardware extensions to do the work (blink blink Intel, not under the x230, not used by currently built GPG AFAIK) then there would be a difference in the kind of trust the user is required to have into the random number generator, let it be inside of the USB Security dongle/ Heads.

Yes, I indeed referred to the fact that having no backup key outside of a smartcard is considered generally the safest as you can not do something wrong here - you just can not export it in any way. So having keys generated on card is less error prone imho. Having said that, you can of course handle private keys with care outside of a smartcard, I am doing this myself, of course. Would I recommend Jane Doe? Not really. Again, this is my personal feeling about it, I totally see the use for your workflow.

tlaurion, 2020-12-08:
@jans23 @kylerankin @techge @alex-nitrokey @MrChromebox 
Ok, with #919 being in the air, here is my next steps where I need your disagreements/comments/threats :)

- [ ] change mount-usb to integrate/adapt mount-sdcard code and go more general into mounting external-drives move away to mount-external-storage ([able to deal with mmc controller if present and loop on new devices popping up including those new ones](https://gitlab.com/tlaurion/heads/-/blob/PrivacyBeast_X230_QubesOS_Certified/initrd/bin/mount-sdcard?__cf_chl_captcha_tk__=ae45f0540cbe9588e41912e19763fde16d5c091e-1607449209-0-Ab1vEEwYKbZs0ubX4jKTQYdzrDoq43iT9BknMRHu7PsK_vUiAyOsW9oku-5sHZcBOtuNp17UJcOskECwkzGF0503h_JvtLMDHqIcvG9w2jvcpXgkJBnm7637yTn7adQVVybp3Q2z-o39eK4bEhAuWpjDoQbA2T-5NCgzqjLIcIc_JgZa0cEZhdUe_a8zEi8Pg6Ir9XQBDndLAehP7PnncZz_aZxj8raIM_vcbjBbs3MaiR9pof6F03QZTgXXEeSGE2fATcjRW2kYt7yhiE_mVXrxcniRckDS4kYUHTogJHNzAZM4VDIN55b24G7QpHn1o76CGA1pc9HzT72hFrycm2how_55dxdY0yyvTgkw-g-aBw82tn2OqDvJo_3NXP4tB8UJA77BrJEOpfGrxavpkafla2Q4CiR1g32DaMvybha8weXjuf1ejdavv5Igf9DDbpc1H8UzWI1INyMH8w25nAWxk8UMFasSc-q2rvflP0zq3EyHvBh7HrsKS8e5Op1DRGWHd7SJfpdUP33VWD3btqr7K4-A_RYvETwYI8oWImT7BMULHeRcf2-UDBXbZhiU524HY04uRKXRzTbn9agAHberqrL2BGq2bog_E8WyCNba5ZM-1V6O22CuDPk3wyywtWL4pRFxuOvI7cGNdI0CEMQ#L60-95)). Will require adding up mmc modules, pushing newer kernel depending on current old release to the last LTS release (since the x230 has two possible controllers: Ricoh and Realtek controllers)
- [ ] change mount-external-storage to be able to [deal with encrypted storage ](https://gitlab.com/tlaurion/heads/-/blob/PrivacyBeast_X230_QubesOS_Certified/initrd/bin/mount-sdcard?__cf_chl_captcha_tk__=37b7c4d2331879bee08a211338187679bebb18bf-1607453120-0-AadkzjPgvA0ZUilj93YhMeP8TWqNtGromj-33DWBtRoGxDOohwkf0wYTk67GBgdex6qR_1rE63C-7gF4lRdVrAgxgPaUY7WE2jjA3o4PKbtJoobCUxGV0QpvbrQmrtA-6cHkUVOuecNbv8BIOwjTnE-5WC-qjNQ25zilaRp1vdbMLsXa9HUaxADi12ZqnkGpHbRe4nvPB2fZibQhsAGH-2PazevdnCYJkyTJNEWRmm4Qb6ompfzIQA1dUeZkO86hFuVZnF1kkb50kZd2_Mxcw_MuWG4icQEnLpuheqJgCEHUG5TPfU0tHXvADifEzyh-eMvwItqHKB3vi3CLQvfR6OtOZJGMARggNz__sAEJ4opMD9yHdWOwhfRwoT-vELIuUn0_idGg-EAAswRijXjrQRINzVXrt8ilLH2QBBZm0v9R7Vi3bDw0jN38equEOVr0mog1kC1USFHsyQKeHKM98u1R2-aIpaXicGWBNOkiYCsiFRbG0fakPrqp8iMtAsUZUPg6O1f4nrfemOUxN4HARX_hLLNkA4JK3bxfJrjGS6fMZMfA920b3gryWCpOliAg9jol0TuqFqB-bMI7vJcHNr9QQktbi5YesGVo3DKAOBQmDBmz1IqAiHDU4DmhkXs1ZR3AstB8ondOiughk3Aky08#L237-243)
- [ ] Implement NKSTORECLI mount-external-storage logic
- [ ] Implement oem factory reset keygen under Heads with external storage of private key over encrypted storage
- [ ] Implement RESET PIN into oem factory reset/gpg factory reset
- [ ] Implement Restore of private keys to USB Security dongle (keytocard from encrypted storage)

saper, 2023-02-28:
I have a funny idea how to solve this. Can we use something else than gpg?

jans23, 2023-02-28:
The only other software I know and which is capable of replacing GnuPG is [Sequoia](https://sequoia-pgp.org/). AFAIK HEADS doesn't has Rust in the tool chain yet.

saper, 2023-02-28:
What I mean I'd like to use raw APDU interface to the smartcard (and PKCS#11) but not `scdaemon` interface.

tlaurion, 2023-03-01:
@jans23 last time i checked, sequoia was not supporting smartcard?

tlaurion, 2023-03-01:
@saper : I'm not sure I understand why this would help the present ticket?

tlaurion, 2023-03-01:
I will start attacking this issue in the next weeks, providing testing scripts first then iterate from them with integration.

Reminder: goal is to generate keypair in memory, then copy those keypairs into luks encrypted container and on smartcard after reset of smartcard.

Then to offer possibility to use only Luis container to use keypair in case no USB dongle is avail.

And to restore keypairs to USB security dongle.

saper, 2023-03-03:
I understand that this is one of the possible solutions for the customer concern.

I have two slightly different/broader ideas to solve the "I flushed the token down the toilet" problem:

1) Provide a mechanism to securely install the key onto multiple tokens. 

Less ambitious:

2) Primary keypair is printed to the screen and/or provided in an easy to memorize form, then it is used to generate actual keys stored on the hardware token(s). If the token is lost, another keypair can be generated. Next level could be revocation mechanism of the lost keypair.

We can surely discuss it outside of this bug (if you prefer to focus on the "generate the key outside of the smartcard" solution.

tlaurion, 2023-03-03:
@saper 

If:
- master key, signing, auth and encryption subkeys are generated in ram and stored on encrypted container (which decryption key passphrase permits usage)

Then
- subkeys can be used in case of flushed down the toiley/lost in the sea
- subkeys can be used to provision one/multiple keys

The idea here is to propose to the user in factory reset/reownership automatic creation of 
- master key
- subkeys
- user/admin pin (passphrase)
- use admin passphrase as luks key passphrase
- partitioning and saving keys to encrypted thumb drive
- copy subkeys to smartcard

Then
- having gpg submenus under heads to use backup to provision X number of keys

But most importantly, implement authentication for
- accessing recovery shell when a backup is known to exist
- boot from USB
- flashing new firmware

This way, heads with authentication could:
- lock the platform prior of kexec (limiting internal firmware upgrades to Heads only)
- mitigate DoS attacks (wiping disk from recovery shell, flashing invalid firmware, booting whatever from USB)

----

Accessing smartcard content without scdaemon seems nice but would deserve a separate issue, explaining goals, actual limitations of Heads (and gpg for that goal) and a short description of what is proposed/desired to attain that goal.

jans23, 2023-03-06:
> @jans23 last time i checked, sequoia was not supporting smartcard?

These days Sequoia does [support OpenPGP Card](https://gitlab.com/sequoia-pgp/pks-openpgp-card).

hko-s, 2023-03-15:
> @jans23 last time i checked, sequoia was not supporting smartcard?

I've written a set of OpenPGP card client libraries in Rust (https://gitlab.com/openpgp-card/openpgp-card), which will be more integrated into the Sequoia ecosystem.

In particular, the openpgp-card libraries will be used in the https://gitlab.com/sequoia-pgp/sequoia-keystore project as the backend for OpenPGP card devices. However, the Sequoia keystore project is still in a pre-alpha stage.
(pks-openpgp-card is an experimental project by @wiktor-k, which is not intended for wider use).

Depending on what you want to do, the https://crates.io/crates/openpgp-card-tools CLI tool might be of interest. It can be used to set up OpenPGP cards, and also to perform cryptographic operations on them. The tool has an optional JSON mode (for scripted setup - or checking the state of - OpenPGP cards).

tlaurion, 2023-03-23:
@saper 

From [gnupg 2.3 news](https://dev.gnupg.org/source/gnupg/browse/master/NEWS)
> * New tool gpg-card as a flexible frontend for all types of supported smartcards.

That would permit to get rid of a lot of screenscaping if not all, yet again this needs to be tested.

On slack/matrix, you said that you adding keys from newer version of gnupg on OS made weird behaviors happen under Heads (which is using really old 2.2.21), this was meant to happen. If I look at the same news from gnupg website linked above, and the bugs you reported trying to deal with curves algo and some reported fixes and your experience with dealing with Tails newer gnupg toolchain, it is my hope (we do not want to change things too much here for current use case of Heads which is basically to detach sign /boot content and have those verified at each boot) while supporting as well what users use.

As said in Q&A of your talk at DUG#1, best case here would be to confirm/infirm that newer gpg toolstack fits the need and doesn't bug. Once code is made to securely create usb thumb drive encrypted container and apply current not so hard RSA 4096 subkeys current use case from factory reset, I will see if PoC code permits to deal correctly with keygen for other key material. I was going to use bookworm for this testbed... but this is still 2.2.40 :/

tlaurion, 2023-11-28:
Integrated in #1515

closed 2023-12-23 by tlaurion