linuxboot/heads #775: Build Error "Could not add ../../build/x230/bzImage, too big?

ArtificialAmateur, 2020-07-09:
Attempting to build x230 `coreboot.rom` 12MB image.

Build log:

```
tail ../build/log/coreboot.log

printf "    CBFS       fallback/payload\n"
    CBFS       fallback/payload
x230/util/cbfstool/cbfstool x230/coreboot.pre.tmp add-payload -f ../../build/x230/bzImage -n fallback/payload  -c none  -r COREBOOT   -C "intel_iommu=igfx_off quiet" -I "../../build/x230/initrd.cpio.xz"
E: Could not add [../../build/x230/bzImage, 7001028 bytes (6836 KB)@0x0]; too big?
E: Failed to add '../../build/x230/bzImage' into ROM image.
E: Failed while operating on 'COREBOOT' region!
E: The image will be left unmodified.
make[1]: *** [Makefile.inc:920: x230/coreboot.pre] Error 1
rm x230/util/cbfstool/fmd_parser.c x230/util/cbfstool/fmd_scanner.c
make[1]: Leaving directory '/home/artificial/Downloads/heads/build/coreboot-4.8.1'
```

tlaurion, 2020-07-10:
Build system's distribution? We are at the limits of available SPI space here without Neutering/Cleaning ME and using that freed space being defined in ifd....

tlaurion, 2020-07-10:
[Here, build system debian/bullseye](https://app.circleci.com/pipelines/github/osresearch/heads/177/workflows/8cb4c231-c737-41b3-b33e-a95e4af96342/jobs/188), under [x230](https://app.circleci.com/pipelines/github/osresearch/heads/177/workflows/8cb4c231-c737-41b3-b33e-a95e4af96342/jobs/188/parallel-runs/0/steps/0-117):

closed 2020-08-01 by tlaurion