linuxboot/heads #780: "HOTP: Invalid Code" and "Error PCR mismatch from TPM_Unseal"

kcortman, 2020-07-22:
Greetings and warm salutations! 

Firstly, I want to express incredible gratitude to the Heads community for the hard work put into this project. It represents both an incredible technological accomplishment and milestone!  

I’ve recently 'given a go' at flashing my X230 after successfully building x230-hot-verification via the osresearch/heads repo.  All went well!  I also have a factory reset NitroKey Pro (reset via the NitroKey App... and have proceeded to setup the pair as "fresh"). 

The issue I’m having (albeit a very important one) is my inability to get HOTP to show anything other than “HOTP: invalid code”.  

Let me first say that I’ve scoured the forums (to date: Heads, Nitrokey, Purism) for what I’m witnessing and can’t seem to figure out what is wrong (and am starting to think hardware or something way 'left field'). ANY suggestions would be incredible : ) !

Without making this post unduly long, my ailment is that after TPM reset and/or TOTP/HOTP reset, I always get "HOTP:invalid code." 

**Procedure:** 
-Succesfful Heads flash and boot:
[image]

-Begin "OEM Factory Reset":
[image]

-Follow prompts:
[image]

-OEM Factor Reset Complete
[image]

-Reboot 

-Generate new HOTP/TOTP secret:
[image]

-Follow prompts
[image]

*At this point, I want to note; that up until now, the NitroKey Pro is blinking "red," but after this initialization (and before I "Press Enter to continue") it begins to blink "green"!  Which seems like a good thing... but then, next I know, it's blinking "red" again at the next screen showing "HOTP: invalid code": 
[image]

**Expected behavior:**
"HOTP: Success"

Please also note, that my hwclock is 'spot on'.  both 'date' and 'hwclock' are right on UTC time so I fairly certain it's not do to drift or anything like that.  Plus, to this point, HOTP is always reading "invalid code" and not just betwixt restarts or somethign like that.  HOTP never, ever, shows "success".  

**Analysis:**
Digging a bit more... I've been playing in the recovery shell, and I've noticed if I use "unseal-totp" or "unseal-hotp," it returns: "Error PCR mismatch from TPM_Unseal" and associated "Unable to unseal TOTP secret" (and 'HOTP secret', respectively). 

Command "seal-totp" always spits out a QRCode.
Command "seal-libremkey" always gives me the the "Error PCR mismatch from TPM_Unseal" 

**Conclusion:** 
TPM isn't sealing.  But, as I am a newbie, I'm lost for explanation.  I've not seen this type of behavior posted in the forum which makes me believe I'm not following instructions properly (always a possibility ; ), or something is  physically broken?  

I want to thank any and all suggestions or explanation that may come from this discussion.  

Any help would be very appreciated.

MrChromebox, 2020-07-23:
this is expected when building from master, as the libremkey-hotp-verification module needs to be bumped to fix an error. Pull in #761 or #774 to fix and will work as expected

tlaurion, 2020-07-23:
#761 specifies 
`hotp-verification_version := 5fb260e631b237a298b6dcca47bbd728f2c5ac3a`

#774 specifies version to commit id:
`03a198c418a60c54ef3ec67ea8a9a2d29b675b9b`

kcortman, 2020-07-23:
Well, I'll be....  Thank you both, MrChromebox and tlaurion. 

**HOTP: Success**
[image]

--closed---

closed 2020-07-23 by kcortman