linuxboot/heads #782: HOTP Branding PR (#761) broke HTOP verification

MrChromebox, 2020-07-28:
PR #761 broke compilation with CONFIG_HOTPKEY=y:

in modules/hotp-verification

> hotp-verification_output := \
>	hotp_verification \
>	hotp_initialize

hotp_verification is not a product of the module Makefile (it still builds libremkey_hotp_initialize and nitrokey_hotp_initialize)
hotp_initialize script doesn't exist (libremkey_hotp_initialize and nitrokey_hotp_initialize do though)

we need to patch nitrokey-hotp-verification until these are fixed upstream. I'll submit PRs for both.

MrChromebox, 2020-07-28:
Pushed PR #783 here and upstream: https://github.com/Nitrokey/nitrokey-hotp-verification/pull/18

the former is only needed until the latter is accepted and hotp-verification module updated

alex-nitrokey, 2020-07-30:
Please see https://github.com/osresearch/heads/pull/783#issuecomment-666512867

closed 2020-08-01 by tlaurion