linuxboot/heads #785: Can't boot Fedora ISOs : CHECKSUM files are not detached signed.

tlaurion, 2020-07-30:
From [Fedora](https://getfedora.org/en/security/):
> curl https://getfedora.org/static/fedora.gpg | gpg --import

- [ ] The public key inside of Heads is still valid

>gpg --verify-files *-CHECKSUM

```
gpg : Good signature from "Fedora (32) .... "
gpg: not a detached signature; file blahblah.iso was NOT verified
```
>sha256sum -c *-CHECKSUM

`sha256sum: WARNING: 21 of 21 computed checksums did NOT match` while doing a sha256sum manually of the iso and comparing with the SHA256SUM file line shows the same hash obtained by validating manually fro command line...

What do we do with that?
- [kexec-iso-init](https://github.com/osresearch/heads/blob/master/initrd/bin/kexec-iso-init#L20) depends on gpgv wrapper to call its [gpg vefification](https://github.com/osresearch/heads/blob/master/initrd/bin/gpgv#L3)
- `gpg --verify-files` doesn't work, probably from gpg changes since reverted gpg 2.2.10
- `sha256sum -v` doesn't work since we depend on busybox, not linux directly build binaries.

@MrChromebox, any thoughts? 
Not directly linked to #784 but this prohibited me to test directly your PR, since ISO boot was supposed to work. (Fedora, QubesOS and Tails can boot from iso if detached signed file is provided side by side with iso. No idea exactly when Fedora changed but even their archive repos have `-CHECKSUM` now and no `.iso.asc` nor `.iso.sig`.

MrChromebox, 2020-07-30:
I'm thinking we need to revisit the gpg upgrade and figure out why it broke the oem factory reset when run from gui-init vs recovery shell (TTY-related?), assuming that updating gpg resolves this issue.

tlaurion, 2020-07-30:
@MrChromebox Fedora changed their ways. Integrity validation relies on sha256sum and not on gpg anymore...

```
user@Insurgo:/media$ curl https://getfedora.org/static/fedora.gpg | gpg --import
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 10955  100 10955    0     0   4045      0  0:00:02  0:00:02 --:--:--  4045
gpg: key EF3C111FCFC659B9: public key "Fedora (30) <fedora-30-primary@fedoraproject.org>" imported
gpg: key 50CB390B3C3359C4: public key "Fedora (31) <fedora-31-primary@fedoraproject.org>" imported
gpg: key 6C13026D12C944D0: public key "Fedora (32) <fedora-32-primary@fedoraproject.org>" imported
gpg: key 3B49DF2A0608B895: public key "EPEL (6) <epel@fedoraproject.org>" imported
gpg: key 6A2FAEA2352C64E5: public key "Fedora EPEL (7) <epel@fedoraproject.org>" imported
gpg: key 21EA45AB2F86D6A1: public key "Fedora EPEL (8) <epel@fedoraproject.org>" imported
gpg: key 7BB90722DBBDCF7C: public key "Fedora (iot 2019) <fedora-iot-2019@fedoraproject.org>" imported
gpg: Total number processed: 7
gpg:               imported: 7
```
```

user@Insurgo:/media$ gpg --verify-files *-CHECKSUM
gpg: Signature made Fri 24 Apr 2020 12:58:35 PM EDT
gpg:                using RSA key 6C13026D12C944D0
gpg: Good signature from "Fedora (32) <fedora-32-primary@fedoraproject.org>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: 97A1 AE57 C3A2 372C CA3A  4ABA 6C13 026D 12C9 44D0
```

```
user@Insurgo:/media$ sha256sum -c *-CHECKSUM
Fedora-Workstation-Live-x86_64-32-1.6.iso: OK
sha256sum: WARNING: 19 lines are improperly formatted
```

tlaurion, 2020-07-30:
> I'm thinking we need to revisit the gpg upgrade and figure out why it broke the oem factory reset when run from gui-init vs recovery shell (TTY-related?), assuming that updating gpg resolves this issue.

But yeah, #777 .

tlaurion, 2022-05-17:
Note: fedora key should be removed, having it in distrib keys is misleading.

PR #1163 (2022-05-17): Remove fedora public key. They don't detach sign ISOs since before 2020.
Fixes #785
closed 2022-05-17 by tlaurion