linuxboot/heads #789: QubesOS 4.1 installer doesn't boot

tlaurion, 2020-08-01:
Booting 
https://openqa.qubes-os.org/tests/10816/asset/iso/Qubes-4.1-20200726-x86_64.iso results in this:
[image]

@marmarek any idea?

_Originally posted by @tlaurion in https://github.com/osresearch/heads/pull/788#issuecomment-667547990_

tlaurion, 2020-08-03:
Trying to find relevant information and will edit this comment (@marmarek)
- [no-real-mode disappeared](https://github.com/QubesOS/qubes-issues/issues/5529#issuecomment-568888958)?

tlaurion, 2020-08-05:
@marmarek: `no-real-mode` [gone](https://github.com/QubesOS/qubes-issues/issues/5529#issuecomment-568889327) under xen?

osresearch, 2020-08-06:
I just downloaded the Qubes 4.1 ISO and it appears to be using Xen 4.13.  Grepping through the 4.13 source it appears that `no-real-mode` should still be valid in the early command line parser, although the main parser doesn't seem to have it:

```
static bool skip_realmode(const char *cmdline)
{
    return find_opt(cmdline, "no-real-mode", false) || find_opt(cmdline, "tboot=", true);
}
```

which is used here:

```
void __stdcall cmdline_parse_early(const char *cmdline, early_boot_opts_t *ebo)
{
    if ( !cmdline )
        return;

    ebo->skip_realmode = skip_realmode(cmdline);
    ebo->opt_edd = edd_parse(cmdline);
    ebo->opt_edid = edid_parse(cmdline);

#ifdef CONFIG_VIDEO
    vga_parse(cmdline, ebo);
#endif
}
```

This parser has moved from `arch/x86/boot/cmdline.S` in Xen 4.8 to `cmdline.c`, although it is invoked in a funny way.

The Xen code in 4.13 that does the trampoline setup has changed, so I wonder if we can retry with the ancient patch that prevents the EBDA from being consulted.  From my 2016 xen-devel post:

```
diff --recursive -u /home/hudson/build/clean/xen-4.6.3/xen/arch/x86/boot/head.S 
./xen/arch/x86/boot/head.S
--- /home/hudson/build/clean/xen-4.6.3/xen/arch/x86/boot/head.S 2016-06-20 
08:08:22.000000000 -0400
+++ ./xen/arch/x86/boot/head.S  2016-08-03 17:56:37.511121443 -0400
@@ -86,6 +86,8 @@
         cmp     $MULTIBOOT_BOOTLOADER_MAGIC,%eax
         jne     not_multiboot
 
+#if 0
+
         /* Set up trampoline segment 64k below EBDA */
         movzwl  0x40e,%eax          /* EBDA segment */
         cmp     $0xa000,%eax        /* sanity check (high) */
@@ -108,6 +110,12 @@
         shl     $10-4,%edx
         cmp     %eax,%edx           /* compare with BDA value */
         cmovb   %edx,%eax           /* and use the smaller */
+#else
+       // coreboot does not provide an Extended BIOS Data Area pointer
+       // just stash things the Multiboot structure, adjusted to bytes
+        mov     MB_mem_lower(%ebx),%eax
+        shl     $10-4,%eax
+#endif
 
 2:      /* Reserve 64kb for the trampoline */
         sub     $0x1000,%eax
```

The other possibility is that something in the VGA has changed, which was the other place that I had to hack back in the distant past... I'm attempting to get Heads running in qemu again so that I can test it against the Qubes ISO.

Asiderr, 2020-08-10:
@osresearch @tlaurion I've built the Xen hypervisor with the patch. After the hypervisor replacement, the problem still arises. I tried to run the installation media with an older version of the hypervisor and I’ve found out that regression has appeared between 4.11 and 4.12 version.

osresearch, 2020-08-12:
[image]

Does it only fail on real hardware?  I'm running under qemu and the installer is able to boot into the graphical installer after a while, although the Xen console does not appear on the vga.

`qemu` command:
```
qemu-system-x86_64 \
  -cpu Haswell-noTSX-IBRS,vmx=on  \
  -M q35,accel=kvm \
  -m 4G \
  --serial /dev/tty \
  --bios build/qemu-coreboot/coreboot.rom \
  -cdrom /tmp/media/hudson/build/qubes/Qubes-4.1-20200726-x86_64.iso
```

Once Heads is booted, these commands to `kexec` into the Qubes installer:
```
insmod /lib/modules/libata.ko 
insmod /lib/modules/libahci.ko 
insmod /lib/modules/ahci.ko
mount -o ro /dev/sr0 /boot
kexec -l /boot/images/pxeboot/xen.gz \
  --command-line "no-real-mode console=vga vga=current" \
  --module /boot/images/pxeboot/vmlinuz \
  --module /boot/images/pxeboot/initrd.img
```

(You can use `console=com1` for xen and it will provide more feedback in this case, although that is probably not an option on the x230 or other systems)

Asiderr, 2020-08-24:
@osresearch @tlaurion  
> Does it only fail on real hardware?

I'm using x230.

I've bisected Xen from the 4.11 to 4.13 version and
I found out that problems start with:

https://github.com/xen-project/xen/commit/dcf4179

This is the first commit where Qubes OS doesn't boot properly. The Qubes
kernel cannot find the LUKS UUID and boot ends in the emergency mode.
The colorful artifacts appear in this commit:

https://github.com/xen-project/xen/commit/07c181c

I've started the thread in the Xen:

https://markmail.org/message/2owv3q2nmhvg52zs

0rb677, 2020-08-27:
Has anyone tried to use builds from here?
https://gist.github.com/marmarek/f0e225b593650212108f1ea6974f5914

tlaurion, 2020-08-27:
@0rb677 
>https://openqa.qubes-os.org/tests/10816/asset/iso/Qubes-4.1-20200726-x86_64.iso

Those are built by openqa, qubesos QA platform. I didnt built myself but if bug is upstream in Xen and non patched by @marmarek in QubesOS, I do not see the point building from scratch. 

Should I see something different?

tlaurion, 2020-09-13:
@Asiderr 

> xen-project/xen@dcf4179

>This is the first commit where Qubes OS doesn't boot properly. The Qubes kernel cannot find the LUKS UUID and boot ends in the emergency mode. 

This in linked to #749 and not Xen not being able to boot with no-real-mode on real hardware as a result of an installation. Were you able to boot the installer?

marmarek, 2020-10-01:
I tried to boot installed 4.0 system with replaced Xen to Xen 4.14 built directly from stable-4.14 branch and it worked, so there is hope.
Next I'll try with recent installation image and compare results.

MrChromebox, 2020-10-24:
no issues booting Qubes-R4.1.0-alpha20201014pre1-x86_64.iso on Librem 13v2/4, Mini, or L1UM

tlaurion, 2020-10-24:
Downloading https://mirrors.edge.kernel.org/qubes/iso/Qubes-R4.1.0-alpha20201014-x86_64.iso

tlaurion, 2020-10-26:
With newer iso, same behavior as https://github.com/osresearch/heads/issues/789#issue-671147503 is observed.
Tried to bring back coreboot native gfx init with https://github.com/tlaurion/heads/commit/945196cfe1e622f994f1f82eecc8dd6eb81eb61e but it just added more corruption.

Trying to migrate kernel to have newer i915 support with https://github.com/tlaurion/heads/commit/bafaecb16305a9ab7f0030eb0b4cdbf6d6c5321a

0rb677, 2020-10-30:
alpha 4.1 works extremely slow with sandybridge and bunch of ram.

tlaurion, 2020-11-03:
No success on my side for the moment. @jans23?

tlaurion, 2020-12-11:
Newer Xen version might fix the issue (coming next week). Crossing fingers, Xen 4.14 :)

tlaurion, 2021-02-04:
Proven to work under KGPE-D16 by @blobless with that PoC build https://app.circleci.com/jobs/github/tlaurion/heads/770?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

tlaurion, 2021-02-04:
Requires #876 to install. Proven to work also.

Asiderr, 2021-04-19:
@tlaurion Problem still occurs on the x230. Tested with current master (`BOARD=x230`) and Qubes R4.1 - 20210117.

qubicrm, 2021-04-24:
@tlaurion @marmarek 

there is some improvement... I think. 
Using Qubes R4.1-20210424, x230, heads-maximized rom (I think this issue did not exist with plain x230, but not sure). 

NOTE: My heads rom throws a message that may be related: 
_______________________
i915 0000:00:02.0: Invalid PCI ROM header signature expecting 0xaa55, got 0x230c
[drm] Failed to find VBIOS tables (VBT)
_______________________
but it works perfectly with qubes 4.04

I made  a regular USB-device with dd as usual, because the iso in a ext4 partition method is not possible due to lack public key for verification. (I also think that is not relevant)
 

same initial screen, but after a while it continues

then it shows this error trace:
___________________________________________________
? add_wait_quewe_exclusive +0x70/0x70
ilk_crtc_enable+0x220/0x350 [i915]
intel_enable_crtc+0x53/0x60 [i915]
intel_commit_modeset_enables+0x53/0x80 [i915]
intel_atomic_commit_tail+0x1fe/0x5f0 [i915]
intel_atomic_commit+0x333/0x3b0 [i915]
drm_client_modeset_commit_atomic+0x1fc/0x240 [drm]
drm_client_modeset_commit_locked+0x53/0x80 [drm]
drm_client_modeset_commit+0x24/0x40 [drm]
drm_fb_helper_set_par+0xa5/0xd0 [drm_kms_helper]
intel_fbdev_set_par+0x16/0x60 [i915]
? con_is_visible+0x27/0x40
fbcon_init+0x2b3/0x570
visual_init+0xce/0x130
do_bind_con_driver.isra.0+0x1db/0x2e0
do_take_over_console+0x36/0x40
do_fbcon_takeover+0x5c/+0xc0
do_register_framebuffer+0x1e8/0x2f0
register_framebuffer+0x1e/0x40
__drm_fb_helper_initial_config_and_unlock+0x8e/0xc0 [drm_kms_helper]
intel_fbdev_initial_config+0x14/0x30 [i915]
async_run_entry_fn+0x39/0x160
process_one_work+0x1b6/0x350
worker_thread+0x4c/0x310
? process_one_work+0x350/0x350
kthread+0x11b/0x140
? __kthread_bind_mask+0x60/0x60
ret_from_fork+0x22/0x30
---[end trace b21f3588297b9341 ]---
[drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:45:pipe A] flip_done timed out
Console: switching to clour frame buffer device 170x84
[drm:drm_atomic_helper_wait_for_dependencies  [drm_kms_helper]] *ERROR* [CRTC:45:pipe A] flip_done timed out
[drm:drm_atomic_helper_wait_for_dependencies  [drm_kms_helper]] *ERROR* [CONNECTOR:76:LVDS-1] flip_done timed out
[drm:drm_atomic_helper_wait_for_dependencies  [drm_kms_helper]] *ERROR* [PLANE:31:primary A] flip_done timed out
[drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:45:pipe A] flip_done timed out
[***] A start job is running for udev Wait for Complete Device Initialization
__________________________________________________________

and after another (long) while

__________________________________________________________
Starting Installer, one moment... 
anaconda 32.24.5-3.fc32 for Qubes OS 20210424 (pre-release) started
(...)
[drm:drm_atomic_helper_wait_for_dependencies  [drm_kms_helper]] *ERROR* [CRTC:45:pipe A] flip_done timed out
[drm:drm_atomic_helper_wait_for_dependencies  [drm_kms_helper]] *ERROR* [CONNECTOR:76:LVDS-1] flip_done timed out
18:40:17 X startup failed, aborting installation
The instalation cannot continue and the system will be rebooted
Press ENTER to continue
[drm:drm_atomic_helper_wait_for_dependencies  [drm_kms_helper]] *ERROR* [PLANE:31:primary A] flip_done timed out
[drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:45:pipe A] flip_done timed out
__________________________________________________________________

Posting this here only to give feedback, although I am not sure this is the right place (maybe Qubes github?), but I feel this maybe usefull. 
 

If you fell this info is not relevant, feel free  to remove it. Just want to help 

Regards

tlaurion, 2021-04-25:
Seems like there is discrepancy between drm driver from heads reused in kexec for QubesOS 4.1. Probably need a kernel upgrade.

tlaurion, 2021-04-25:
Downloading https://openqa.qubes-os.org/tests/16901/asset/iso/Qubes-4.1-20210422-x86_64.iso

tlaurion, 2021-04-25:
Detach sign iso:
```
user@personal:~$ sudo mount /dev/sda1 /mnt
user@personal:~$ gpg --import /mnt/Insurgo2021.asc
user@personal:~$ gpg --card-status
user@personal:~$ gpg --detach-sign -armor ~/Qubes-4.1-20210422-x86_64.iso
user@personal:~$ ls -al Qubes-4.1-20210422-x86_64.iso*
-rw-r--r-- 1 user user 5410652160 Apr 22 09:52 Qubes-4.1-20210422-x86_64.iso
-rw-r--r-- 1 user user        833 Apr 25 11:09 Qubes-4.1-20210422-x86_64.iso.asc

user@personal:~$ sudo mv Qubes-4.1-20210422-x86_64.iso* /mnt/
```

Now we can boot iso from Heads USB boot function.

tlaurion, 2021-04-25:
There is no CircleCI build because older then 30 days.

```
user@x230-master:~/heads$ git fetch osresearch
user@x230-master:~/heads$ git checkout osresearch/master
user@x230-master:~/heads$ git reset --hard
user@x230-master:~/heads$ make BOARD=x230-hotp-maximized CPUS=2
2021-04-25 11:22:59-04:00 DONE coreboot
"/home/user/heads/build/coreboot-4.8.1/x230-hotp-maximized/cbfstool" "/home/user/heads/build/coreboot-4.8.1/x230-hotp-maximized/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           84772 none
cpu_microcode_blob.bin         0x14c40    microcode       25600 none
fallback/ramstage              0x1b0c0    stage           81106 none
config                         0x2ee00    raw               746 none
revision                       0x2f140    raw               582 none
cmos_layout.bin                0x2f3c0    cmos_layout      1804 none
fallback/dsdt.aml              0x2fb40    raw             13646 none
fallback/payload               0x33100    simple elf    7130564 none
(empty)                        0x6fff00   null          4585688 none
bootblock                      0xb5f800   bootblock        1968 none
2021-04-25 11:22:59-04:00 INSTALL   build/coreboot-4.8.1/x230-hotp-maximized/coreboot.rom => build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom
0b6af0f57644acf012ee1d848d08ba17dc632aabd8d98e34c86c7ce9d2a4c4ce  build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom
2021-04-25 11:22:59-04:00 DD 8MB build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-bottom.rom
4e29e6e8d9faca67dca4150d422d26a1e5c0479c7fc237765ce1e9652aefe46f  /home/user/heads/build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-bottom.rom
2021-04-25 11:22:59-04:00 DD 4MB build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-top.rom
2eefe417bd76ec7d96f8ef757183713836c3d0428c24242104459e9845d60ffd  /home/user/heads/build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-top.rom
4b3be2b552ab6b60df7291bfbd819d9b92f88641662472e0ebd8ca7fc27a4c63  /home/user/heads/build/gawk-4.2.1/gawk
```

Flashing build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom internally

tlaurion, 2021-04-25:
Just to make sure I haven't corrupted blobs:

```
user@x230-master:~/heads$ cd blobs/xx30/
user@x230-master:~/heads/blobs/xx30$ ./download_clean_me.sh 
user@x230-master:~/heads/blobs/xx30$ cd -
user@x230-master:~/heads$ user@x230-master:~/heads$ make BOARD=x230-hotp-maximized CPUS=2

2021-04-25 11:30:39-04:00 DONE coreboot
"/home/user/heads/build/coreboot-4.8.1/x230-hotp-maximized/cbfstool" "/home/user/heads/build/coreboot-4.8.1/x230-hotp-maximized/coreboot.rom" print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           84772 none
cpu_microcode_blob.bin         0x14c40    microcode       25600 none
fallback/ramstage              0x1b0c0    stage           81106 none
config                         0x2ee00    raw               746 none
revision                       0x2f140    raw               582 none
cmos_layout.bin                0x2f3c0    cmos_layout      1804 none
fallback/dsdt.aml              0x2fb40    raw             13646 none
fallback/payload               0x33100    simple elf    7130564 none
(empty)                        0x6fff00   null          4585688 none
bootblock                      0xb5f800   bootblock        1968 none
2021-04-25 11:30:39-04:00 INSTALL   build/coreboot-4.8.1/x230-hotp-maximized/coreboot.rom => build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom
124b2e6e49eb892e8d52e98d03b454782aaf7de25508238d8d09ccbeaf890bdd  build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom
2021-04-25 11:30:39-04:00 DD 8MB build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-bottom.rom
7497f926718968d104994235229c4c7fc6d55cb0de8e3d2b2da9778934c5ea0f  /home/user/heads/build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-bottom.rom
2021-04-25 11:30:39-04:00 DD 4MB build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-top.rom
2eefe417bd76ec7d96f8ef757183713836c3d0428c24242104459e9845d60ffd  /home/user/heads/build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6-top.rom
4b3be2b552ab6b60df7291bfbd819d9b92f88641662472e0ebd8ca7fc27a4c63  /home/user/heads/build/gawk-4.2.1/gawk
```

Prior
0b6af0f57644acf012ee1d848d08ba17dc632aabd8d98e34c86c7ce9d2a4c4ce  build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom
now:
124b2e6e49eb892e8d52e98d03b454782aaf7de25508238d8d09ccbeaf890bdd  build/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-1034-gc3b0bd6.rom

tlaurion, 2021-04-25:
After importing public key in firmware and booting from USB (Options, USB boot, choosing QubesOS Install option)
- I get screen corruption as [before](https://github.com/osresearch/heads/issues/789#issue-671147503) until QubesOS is kexec'ed
- Typing Escape key gives some console output (will dig into that)
- Waiting minutes with cursor on top left of the screen....
- Then turns into 3 dots.... Then in recovery (Installation media searches from a framebuffer that it cannot find)

tlaurion, 2021-04-25:
Same error reported by @qubicrm from i915 and drm_fb_helper

tlaurion, 2021-04-25:
https://gist.github.com/i0z0m/17b01ebe9a50db5c122edf06e892cdb6
Comments the issue in detail ,where proposed workaround is to:
`Panel Self Refresh (PSR), a power saving feature used by Intel iGPUs is known to cause flickering in some instances FS#49628 FS#49371 FS#50605. A temporary solution is to disable this feature using the kernel parameter i915.enable_psr=0.`

Testing

Tonux599, 2021-04-25:
My terminology might be off here. But couldn't we just avoid using modesetting modules? So remove i915 and have just the native graphics used. It wouldn't look as nice but we can set a close resolution with `vga=` in the kernel parameters. This way whatever graphics modules are used by targeted systems would not be interfered by heads.

tlaurion, 2021-04-25:
@Tonux599 : I can revisit this. But the reasoning behind this is the usage of FBWhiptail vs Whiptail(console based).
If we do not have i915 and DRM defined, then FB is not initialized. And if no FB is initialized, we go into Whiptail mode, not FBWhiptail.

For what interests us here (QubesOS install), not having a FB defined is the reason why QubesOS installer fails under KGPE-D16 server if not activating VNC installer, since QubesOS have not implemented terminal based installer, and require a FB there.

Tonux599, 2021-04-25:
I would have to investigate more but from past experiences booting from Grub into, say debian, with nomodeset still allows for for X (along with gnome, xfce etc.) to still display graphics. Usually at a super low resolution but that can be fixed with `vga=`.

Unless the driver (i915, nouveau etc.) is still being used to allow that functionality.

tlaurion, 2021-04-25:
@Tonux599 agreed, but then [all the previous](https://github.com/osresearch/heads/issues/789#issuecomment-826350401):

> Let's remember that Heads [always kexec the OS kernel by passing the additional following arguments to kexec when booting Xen](https://github.com/osresearch/heads/blob/0d5b3d75e7b84ca85ff252fe5fe23aea990da435/initrd/bin/kexec-boot#L73):
> ```
> 	if [ "$kexectype" = "xen" ]; then
> 			# always use xen with custom arguments
> 			kexeccmd="$kexeccmd -l $filepath"
> 			kexeccmd="$kexeccmd --command-line \"$restval no-real-mode reboot=no vga=current\""
> 		elif [ "$kexectype" = "multiboot" ]; then
> 			kexeccmd="$kexeccmd -l $filepath"
> 			kexeccmd="$kexeccmd --command-line \"$restval\""
> 		else
> 			kexeccmd="$kexeccmd -l $filepath"
> 		fi
> ```
> here:
> - Reusing vga initialized kernel driver and drm driver, created fb context, to be reinitialized by the kexec'ed kernel (vga=current)
> - no-real-mode is passed to Xen
> 
> 
> Let's also remember that coreboot [passes some arguments to Heads linux payload at boot](https://github.com/osresearch/heads/blob/0d5b3d75e7b84ca85ff252fe5fe23aea990da435/config/coreboot-x230-hotp-maximized.config#L18) and that [additional, defined kernel arguments are defined under board config, which are passed to OS kernel at boot](https://github.com/osresearch/heads/blob/0d5b3d75e7b84ca85ff252fe5fe23aea990da435/boards/x230-hotp-maximized/x230-hotp-maximized.config#L58)

Would not apply.
I tried removing `vga=current` above without success, tried adding multiple fixes suggested upstream....

tlaurion, 2021-04-26:
Closest to this issue found: https://gitlab.freedesktop.org/drm/intel/-/issues/2414

tlaurion, 2021-04-27:
Repositories https://drm.pages.freedesktop.org/maintainer-tools/repositories.html

tlaurion, 2021-04-28:
So from my understanding of what is happening here, the timeout is happening while attempting to set the fb on non-existing output devices.

tried:
- upgrae to other board used 5.x kernel. Still there.
- reviewed code change regarding drm-kms_helper and i915 in 5.11.17. Nothing there.

Will try:
- disabling LVDS-1, see if error is similar.

tlaurion, 2021-04-28:
https://bugs.freedesktop.org/show_bug.cgi?id=101269

tlaurion, 2021-04-28:
@Tonux599 : the problem here with kexec is that it is expected that the heads kernel already initialized the FB which is supposed to be reused. When there is difference between initialized FB, there is flickering. As shown [here](https://github.com/osresearch/heads/issues/789#issuecomment-826350401), when heads tries to boot xen, it passes `vga=current` to the kexec'ed kernel.

My intuition here is that the kernel drivers for drm, drm_kms_helper and or i915 are the culprits in QubesOS 4.1. There is probably a workaround that can be pusherd in KERNEL_ADD board configurations for sandy/ivy bridges.

tlaurion, 2021-08-19:
Resolved with combination of https://github.com/QubesOS/qubes-issues/issues/6792#issuecomment-890534849 and #1015

tlaurion, 2021-10-13:
> Not related to boot issue, but resume issue: [QubesOS/qubes-issues#6066 (comment)](https://github.com/QubesOS/qubes-issues/issues/6066#issuecomment-941260962)
> 
> To be tested, but supposed to be fixed in Xen 4.14.3 which came with 4.1 RC1

Resume issue fixed when running #1015 and QubesOS 4.1 RC1.

tlaurion, 2021-10-15:
@no-sauce booting qubesos 4.1 requires a ROM upgrade to #1015. Are you reporting that x230 (htop or not) from there doesn't work? You have a reprogrammer?

tlaurion, 2021-12-04:
#1015 merged

closed 2021-12-04 by tlaurion