linuxboot/heads #794: Intel leak

shamen123, 2020-08-06:
I dropped this to the CB mailing list but some interesting things in this leak. Thought id highlight here

https://twitter.com/deletescape/status/1291405688204402689

https://www.tomshardware.com/news/massive-20gb-intel-data-breach-floods-the-internet-mentions-backdoors

   - Intel ME Bringup guides + (flash) tooling + samples for various
   platforms
   - Kabylake (Purley Platform) BIOS Reference Code and Sample Code +
   Initialization code (some of it as exported git repos with full history)
   - Intel CEFDK (Consumer Electronics Firmware Development Kit (Bootloader
   stuff)) SOURCES
   - Silicon / FSP source code packages for various platforms
   - Various Intel Development and Debugging Tools
   - Simics Simulation for Rocket Lake S and potentially other platforms
   - Various roadmaps and other documents
   - Binaries for Camera drivers Intel made for SpaceX
   - Schematics, Docs, Tools + Firmware for the unreleased Tiger Lake
   platform
   - (very horrible) Kabylake FDK training videos
   - Intel Trace Hub + decoder files for various Intel ME versions
   - Elkhart Lake Silicon Reference and Platform Sample Code
   - Some Verilog stuff for various Xeon Platforms, unsure what it is
   exactly.
   - Debug BIOS/TXE builds for various Platforms
   - Bootguard SDK (encrypted zip)
   - Intel Snowridge / Snowfish Process Simulator ADK
   - Various schematics
   - Intel Marketing Material Templates (InDesign)
   - Lots of other things

shamen123, 2020-08-06:
twitter link purports to have the passwords for encrypted files, and there is a link to magnet torrent. Apparently more to come in future.

SebastianMcMillan, 2020-08-07:
The Verilog stuff is huge. Verilog is a HDL (Hardware Descriptor Language) and with that in hand, we know essentially the logic of said device.

Now if those verilogs are of Xeon CPUs.... That would be *very huge* since we know know how the CPUs work, and theoretically speaking, with that in hand we would be able to manufacture our own Xeons that are nearly identical to the real thing.

I also love how they were able to find string "backdoor" and get results. That's... Reassuring.

osresearch, 2020-08-18:
Peter Stuge's post to the coreboot-devel list summarizes my feelings on the leak quite well -- the legal situation is not certain and it is interesting for some people, although not directly useful for free software projects like coreboot and Heads.

> Either way, any leaked Intel code can obviously not be contributed to the coreboot project, except possibly by Intel.
> Only Intel can publish their code under GPL, if anyone can.
> Now that source code for some modern platforms seems to be publically available, actually by far the smartest thing that Intel could do is exactly that - to publish the their source code under GPL.
> Anyone interested in the code can get it now, so Intel would be far better off working *with* that community, instead of working against it.
> But Intel's contract situation may not allow them to do so. In that case really everyone loses, except perhaps bad guys who don't care about contracts anyway and now save lots of reverse engineering effort

closed 2020-08-18 by osresearch

shamen123, 2020-08-18:
@osresearch to add my $0.02 given i was not super clear

I agree with those on CB mailing list and have sent a reply updating my thoughts. Only Intel can publish their code under GPL, if anyone can.

Summarised; 
_"The barn door is open and the horse has bolted, it would make sense to [intel sic] provide pathways to community based projects at this point. Not doing so relies purely on the integrity of community projects that would come to benefit from the data. And not every project out there (or that may start up in future) may have such integrity. And nature abhors a vacuum. 
Just want to clarify my position given the replies - my post was meant to be informative of the situation and not in any way encouraging this project to view, distribute and/or include anything from the leak. I was just passing on the news of the situation given the closeness. I probably should have stipulated that, but I figured that as there were no links to actually get the leaked data through the article I sent, that it would take someone going out of their way to actually find the raw leak. "_