linuxboot/heads #796: Generate GBE blob from coreboot's bincfg tool

tlaurion, 2020-08-09:
@PatrickRudolph @niftygeek @pietrushnic @miczyg1
@pgeorgi @gnutoo (last two authors of coreboot's bincfg tool)

So in my novice path, to be able to generate a valid gbe.bin to be included under #703, I need to: 
- Read specs
  - http://www.intel.co.uk/content/dam/doc/application-note/i-o-controller-hub-9m-82567lf-lm-v-nvm-map-appl-note.pdf
  - https://communities.intel.com/community/wired/blog/2010/10/14/how-to-basic-eeprom-checksums 
- Understand the mappings that led to the creation of gbe-ich6.spec out of precedent spec
- redo with Intel 6 Series Chipsets [spec](https://cdrdv2.intel.com/v1/dl/getContent/613456) p.118 to create a valid  `gbe-cpt.spec` file
- Test until `user@x230-master:~/coreboot/util/bincfg$ ./bincfg -d gbe-cpt.spec heads/blobs/xx30/gbe.bin gbe-cpt.set` creates a valid gbe-cpt.set file
- Test until `./bincfg gbe-cpt.spec gbe-cpt.set heads/blobs/xx30/gbe_mod.bin` produces a valid gbe_mod.bin that can be included in a bootable rom under https://github.com/osresearch/heads/pull/703 ?

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/700#issuecomment-647003701_

Thrilleratplay, 2020-08-14:
@tlaurion  @pgeorgi @GNUtoo  I have not tested this yet, but I think I have the beginnings of bincfg set/spec for the Intel 82579LM, used in the x230.  

A few notes:
* Most set values are the Intel defaults, the LED light values are different to match those in the Lenovo firmware.  
* Some of the settings will need to be changes to disable PXE or boot security. 
* Not sure what to do with the mac address.  If passed in via `make` commandline, not sure if it can be passed into `bincfg` or if done using a patch after `flashregion_3_gbe.bin` is generated.  Suggestions?

https://github.com/Thrilleratplay/coreboot/tree/deblob_intel_82579_gbe

tlaurion, 2020-08-14:
Wow.

tlaurion, 2020-08-14:
Will review soon. Sorry overloaded but happy news!!!

Thrilleratplay, 2020-08-14:
@tlaurion  and I will try to test this today.  If you get to it before I do, verify that bit 6 of word 0x19 is set to `1`.  A footnote in the spec mentions this need to be changed from the default value to indicate it is valid but I forgot to check this is done or not by the `bincfg` checksum.  This slipped my mind and only remembered when I was half asleep last night.

tlaurion, 2020-08-14:
Thanks! How much would you ask to complete this work @Thrilleratplay ?

Thrilleratplay, 2020-08-14:
@tlaurion At this point, I couldn't give an estimate on how much longer it would take.  Even if I figure out how to append the gbe checksum when it is not the last field, there is still the issue of how to add in the MAC address without having to alter the set file.  I will dig around `bincfg` source some more.

Thrilleratplay, 2020-08-15:
@tlaurion The checksum was working but was missing a series of offsets so I  was looking in the wrong place.  This appears to be working now but still could not find a way to pass the mac address through the commandline.  I tested it on my x220, which also has the same model Ethernet, no issues with basic internet function.  When you get a chance, can you also test this and get feedback before submitting it for coreboot review?

tlaurion, 2020-08-16:
@Thrilleratplay setting the mac address to 00:00:00:00:00 or [00:DE:AD:C0:FF:EE](https://notabug.org/libreboot/libreboot/commit/70fa7f5750134b080814a9f44d1a78028aa4e4e0) would do the job. QubesOS users should implement [mac randomization](https://www.qubes-os.org/doc/anonymizing-your-mac-address/#randomize-all-ethernet-and-wifi-connections) to be applied by default, while in our #703 usecase, Heads will randomize on boot. Other tools, like clonezilla, will need to have preboot scripts randomizing the mac address prior of booting. Else, we do not have a redistributabe GBE that is reproducible. Any thoughts welcome, while my preferate would be to set it to 00:DE:AD:C0:FF:EE as a reference to prior work.

Thrilleratplay, 2020-08-16:
@tlaurion I like `00:DE:AD:C0:FF:EE` as a default.  It seems this is a fine for heads but other implementations like skulls will have issues if a static binary is shared.  You are welcome to include it in heads but before I do a PR for coreboot, I want to find a way to make setting the mac or other settings can be done with relative ease.

tlaurion, 2020-08-17:
@Thrilleratplay I will test asap

Thrilleratplay, 2020-08-17:
@tlaurion   When you have time, I already created the [PR for coreboot](https://review.coreboot.org/c/coreboot/+/44510)

tlaurion, 2020-08-18:
Building here https://app.circleci.com/pipelines/github/tlaurion/heads/312/workflows/1f7eed14-21db-4ad1-b97b-a20af2578bf2/jobs/339

tlaurion, 2020-08-20:
There was an issue with the buildsystem whcih was fixed in latest commit.
I integrated generated blob in the following PR https://github.com/osresearch/heads/compare/master...tlaurion:x230-external-flash

Which is included in following build: https://app.circleci.com/pipelines/github/tlaurion/heads/332/workflows/f628e891-58d0-4dd3-8800-b26ae677d0c9/jobs/360

tlaurion, 2020-08-20:
@Thrilleratplay unfortunately, this is what seems to happen when GBE is absent, but last time I injected an invalid GBE laptop was failing to boot, bot booting giving me this error as troubelshooted in #700 

[image]

tlaurion, 2020-08-20:
@Thrilleratplay you can see in this [commit](https://github.com/osresearch/heads/commit/8a0cda16190af80c46e323095d7a139324b1e39c) what is linked to your proposed change

Thrilleratplay, 2020-08-21:
@tlaurion I'm confused.  You are receiving `probe of 0000:00:19.0 failed with error error -3` with the generate GBE or it does not boot at all?