linuxboot/heads #797: Download, innoextract Lonovo website's downloaded ME blob and put it under blobs/xx30 from a module

tlaurion, 2020-08-09:
> @tlaurion so you're proposing to put the blobs in the main Heads repo, along with reproducibility instructions? Any reason to not put them in a submodule as was discussed previously?

@MrChromebox : Any idea how to use [innoextract to extract exe content (cab)](https://github.com/osresearch/heads/pull/703/files#diff-0493cf37597ff5795588f8343ab268b7R4) in a module?

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/307#issuecomment-612697557_

MrChromebox, 2020-08-10:
> @MrChromebox : Any idea how to use innoextract to extract exe content (cab) in a module?

unfortunately not, I've never had to use it before

tlaurion, 2020-08-23:
The version pointed to lenovo is more up to date then the one provided by extract of x23 2.76.

Might want to look at older version of the exe, or repair the ifd and reupload so it matches

tlaurion, 2020-08-23:
Goal here would be to have a heads module that would be defined under board configs and called as a precondition of coreboot module if defined. 

The blobs would then all be available for coreboot module build after that script being called.

Innoextract would need to be added in both CIs to be built.

If you look into the readme of the xx30 blobs, all platfroms supported by ME are there.

For Heads, the x230 t430 and t530 could be supported the same way and created with similar boards to the x230-external-flash board definition (or XXX-full-rom, what sounds best)

tlaurion, 2020-08-23:
Once that poc is valid, same could be done for xx20 boards like the t420 x220 and others, maximizing BIOS region and making rom fully flashable (instead of just flashing the BIOS region per FLASHROM statement under board config for internal flashing / verification)

tlaurion, 2020-08-23:
Better download and verify in script for specific and expected sha256sums

tlaurion, 2020-08-23:
@Thrilleratplay https://github.com/osresearch/heads/pull/703/files#diff-0493CF37597FF5795588F8343AB268B7

Thrilleratplay, 2020-08-23:
I assumed that was the intended result.  I have a x220 Tablet I can test on.

The confusing part is the file `app/ME8_5M_Production.bin` is 6.1M despite `5M` being in the name.  `FWUpdLcl.exe` may do something to truncate the file.

tlaurion, 2020-08-24:
@Thrilleratplay 'python ~/me_cleaner/me_cleaner.py -r -t -O ~/heads/blobs/xx30/me.bin/ME8_5M_Production.bin

Thrilleratplay, 2020-08-24:
@tlaurion Sorry, part of that reply was thinking/typing out loud.  After you pointed it out, I understand that you already have the extraction command in the  PR for #703 but thought the download links were the same.  Using [the link in the PR]( https://download.lenovo.com/pccbbs/mobiles/g1rg24ww.exe), the resulting SHA256 are the same as in the PR.  

The untouched extracted `ME8_5M_Production.bin` was 6.1M in the file I downloaded and is 7.5M the one you used.  Why are these not 5M?  I wanted test updating ME using the utility and do a binary diff between extracted file and the updated version flashed in the ME region.

tlaurion, 2020-08-24:
Will reverify. 
Reviewing librem_mini PR, mrChromebox came with a nice way of doing it while self hosting binaries somewhere else.
[That](https://github.com/osresearch/heads/pull/806/files#r475901047) may get you ideas

tlaurion, 2020-11-08:
#703 is poc

closed 2020-11-08 by tlaurion