linuxboot/heads #799: Makefile: make 4.2.1 is not respecting Heads logic verification for already extracted packages

tlaurion, 2020-08-12:
We can see the result here when a cache is restored without hacks:
https://app.circleci.com/pipelines/github/tlaurion/heads/293/workflows/e50cfc8b-c61f-474b-b2de-476e3cee845d/jobs/319/parallel-runs/0/steps/0-107

We see here that the build fails since make was already extracted there and the patch was already applied.
@osresearch ?

My hack was to delete that directory prior of building when qemu-coreboot was first built [here](url), but it would be nice if make was also respecting Heads build system logic.

tlaurion, 2020-08-12:
Applied [fix here](https://github.com/tlaurion/heads/commit/a5e50c3be2322e6d135687c225e551bdc0a9d06c) which fixes the issue until fixed under Heads buildsystem

osresearch, 2020-08-12:
Is the cache being restored with the time stamps intact or are they all new? If a normal copy is done, then depending on the copy order, the `$(build)/$(make_dir)/.patch` file might be older than the `patches/make-$(make_version).patch` file, which leads to the patch attempt.

As a quick hack, the assumption could be made that `make` will never be re-patched in normal operation: 

```
diff --git a/Makefile b/Makefile
index 084ad03..10faebf 100644
--- a/Makefile
+++ b/Makefile
@@ -602,8 +602,8 @@ $(build)/$(make_dir)/.extract: $(packages)/$(make_tar)
 	tar xf "$<" -C "$(build)"
 	touch "$@"
 
-$(build)/$(make_dir)/.patch: patches/make-$(make_version).patch $(build)/$(make_dir)/.extract
-	( cd "$(dir $@)" ; patch -p1 ) < "$<"
+$(build)/$(make_dir)/.patch: $(build)/$(make_dir)/.extract
+	( cd "$(dir $@)" ; patch -p1 ) < "patches/make-$(make_version).patch"
 	touch "$@"
 
 $(build)/$(make_dir)/.configured: $(build)/$(make_dir)/.patch
```

tlaurion, 2020-08-12:
@osresearch : my concern here is that this module is the only one acting this way.

PR #808 (2020-08-20): Fix CircleCI build problems
- Fix issue #807
- Esthetical modules/coreboot changes to facilitate future coreboot maintainership putting hashes of coreboot and coreboot-blobs together
- Fix #799 with implementation of @osresearch's recommendation.

@MrChromebox : please review
commit b14e564a (2020-08-20): Fix CircleCI build problems (#808)

* CircleCI: debian:10 docker based. Give possitility to override CACHE_VERSION through CircleCI when needed
* Makefile: fix #799 with implementation of @osresearch's recommended https://github.com/osresearch/heads/issues/799#issuecomment-673059028
* modules/coreboot : indentation fix and putting version hashes together to facilitate future maintainership.
closed 2020-08-20 by tlaurion