linuxboot/heads #801: Add Librem boards inside of public CIs

tlaurion, 2020-08-13:
@MrChromebox 
> Any reason the Librem boards (which need to stay on 4.8.1 for now) aren't being built (without blobs) as part of CI testing?

But the build will obviously fail because blobs are not provided in tree:
https://github.com/osresearch/heads/blob/master/config/coreboot-librem13v2.config#L13-L23
https://github.com/osresearch/heads/blob/master/config/coreboot-librem13v4.config#L13-L23
https://github.com/osresearch/heads/blob/master/config/coreboot-librem15v3.config#L13-L23
https://github.com/osresearch/heads/blob/master/config/coreboot-librem15v4.config#L13-L23

So I will remove those boards in next commit once that point is proven with [current board+coreboot configs](https://app.circleci.com/pipelines/github/tlaurion/heads/300/workflows/ef62c2d4-12ff-41ef-9cac-de6b867381d9/jobs/326), _which is why the x220 and t420 (which have blobs/* corresponding directories) are not built from CIs since not blob free as of now_. 
https://github.com/osresearch/heads/blob/master/config/coreboot-x220.config#L13-L18
https://github.com/osresearch/heads/blob/master/config/coreboot-t420.config#L13-L18

You can join forces #307, that ticket would feel less lonely.

_Originally posted by @tlaurion in https://github.com/osresearch/heads/pull/721#issuecomment-673517175_

tlaurion, 2020-08-13:
One idea here would be to duplicate boards similarly to the x230-flash board, leaving ME GBE and IFD and other blobs regions untouched and as the responsibility of the external reprogrammer to extract the blobs from backup (following blobs/$BOARD/instructions) for the initial external flashing, while CIs builds a fake IFD based full image (x230) matching combined SPI full size flash size for flashrom to not complain and for flashrom to only care about the BIOS region inside of it, flashed internally per $BOARD flashrom statement.

Else #307 is the way...

MrChromebox, 2020-08-13:
what are your thoughts on building blob-less for CI, simply to ensure all paths are build-tested? Would require separate configs obviously but not really an issue

tlaurion, 2020-08-13:
@MrChromebox concern is to have blobs in path and legal issues linked to hosting them in github, hence #307

MrChromebox, 2020-08-13:
I feel like we shouldn't be bothering with non-publicly available blobs for CI purposes. We've spent so much effort on trying to come up with a reproducible procedure for the Thinkpads when we could have just documented / scripted the process for users to read their own firmware, clean the ME, modify the IFD, etc.

tlaurion, 2020-08-13:
@MrChromebox : the end result being a flashable BIOS region in that case, just like the x230 is currently (with faked IFD, without GBE considering its there else #700 happens, without the capability of using all ME freed space cause IFD is not unlocked, BIOS region is not extended and ME is not reduced. Else I do not understand what would be your plan here, which debate should be under #307 

Else I do not understand what would be CI reproducible considering all roms being at full capacity, which is what #307 is all about.

The proof here being for the x230 the need to augment CBFS region to 7100000 which is the absolute limit. Addition of other tools is now impossible, where x220 and t420 are currently (untested) left behind without #307 unless #590 or https://github.com/osresearch/heads/pull/730 or better ideas.

Else what you propose is actually have, per board, maximized CBFS size under coreboot configs, while those coreboot configurations not including any blobs in coreboot configs to be CI's buildable. Agreed?

MrChromebox, 2020-08-13:
> So basically, two board configs per physical boards:

I think that's a perfect solution for boards without publicly available blobs. Wouldn't be required for Librems though since all blobs are available publicly via our purism-blobs repo

tlaurion, 2020-08-13:
@MrChromebox 
> I think that's a perfect solution for boards without publicly available blobs. Wouldn't be required for Librems though since all blobs are available publicly via our purism-blobs repo

Then if it the case, I do not understand why they could not be automatically downloaded to have those boards built directly by CIs and be totally reproducible... And you are actually agreeing with #307!

MrChromebox, 2020-08-13:
> Then if it the case, I do not understand why they could not be automatically downloaded to have those boards built directly by CIs and be totally reproducible

once the boards move to coreboot 4.12 and the librem-blobs module is added, then sure. Until then, it would still be useful to have blob-free versions build tested, no?

> If the blobs can be downloaded (ME is not a problem for neither xx20 xx30 nor Librems, IFD is ok and GBE could be untouched/generated per bincfg) then Heads should provide them by downloading them/hosting them and produce full ROM images to be fwupd updateable directly from CIs.

I thought the whole point of this was to avoid hosting the ME for 20/30-series boards. If you're going to host it, then just host the cleaned blob and modified IFD and call it a day. Reuse the user's existing GBE so they don't lose their MAC address, easy to automate that

tlaurion, 2020-08-13:
> I thought the whole point of this was to avoid hosting the ME for 20/30-series boards. If you're going to host it, then just host the cleaned blob and modified IFD and call it a day. Reuse the user's existing GBE so they don't lose their MAC address, easy to automate that

This is https://github.com/osresearch/heads/issues/797 (downloading ME update capsule, extracting exe with innoextract, and then applying me_cleaner on top of it)

I have no idea how to automate calling external tools for archives downloaded by Make scripts which cannot be extracted per normal tools (not tar etc) which needs to be processed by cabextract/innoextract). Once Lenovo downloaded ME blobs are downloaded for xx20 and xx30 platform, if we host the ifd on heads repository (@osresearch has nothing against it) we can have maximal usable space with ME in ROM. Best would be to generate GBE with static MAC #796  (so bit by bit reproducible) but could be postponed and gbe also included with sacrificed mac address, or as you noted, not include GBE and expect user to have it still on (while rom not totally flashable: so BOARD config would need to specify which regions to flash: ME, BIOS..) Best would be the whole flash, once again, with the OS responsible to randomize mac address.

Note that those issues have bounties tag on them. If you have time are interested in those, feel free to propose a quote on those issues directly and you will receive funds upon PoW.

closed 2022-04-06 by tlaurion