linuxboot/heads #807: CircleCI not able to build Coreboot 4.8.1 after coreboot module change

tlaurion, 2020-08-19:
@MrChromebox 
Building [without cache](https://app.circleci.com/pipelines/github/tlaurion/heads/315/workflows/3a003bc2-676d-489f-8e0d-d3189e9223e0/jobs/343) since commit  c6a82a246411f091ccebdc683da83fafd989b08a results in:
```
gcc -I/root/project/build/coreboot-4.8.1/util/sconfig -Ix230-flash/util/sconfig -o x230-flash/util/sconfig/sconfig x230-flash/util/sconfig/lex.yy.o x230-flash/util/sconfig/sconfig.tab.o x230-flash/util/sconfig/main.o
/usr/bin/ld: x230-flash/util/sconfig/main.o:/root/project/build/coreboot-4.8.1/util/sconfig/sconfig.h:80: multiple definition of `head'; x230-flash/util/sconfig/sconfig.tab.o:/root/project/build/coreboot-4.8.1/util/sconfig/sconfig.h:80: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [util/sconfig/Makefile.inc:43: x230-flash/util/sconfig/sconfig] Error 1
rm x230-flash/util/cbfstool/fmd_scanner.c x230-flash/util/cbfstool/fmd_parser.c
make[2]: Leaving directory '/root/project/build/coreboot-4.8.1'
make[1]: *** [Makefile:382: /root/project/build/coreboot-4.8.1/x230-flash/.build] Error 1
make[1]: Leaving directory '/root/project'
make: *** [Makefile:587: all] Error 2
```

Any idea?

tlaurion, 2020-08-19:
Same happened back there at that commit [from CircleCI](https://app.circleci.com/pipelines/github/osresearch/heads/196/workflows/168e94c2-be36-47bd-8847-80a0be4e4b63/jobs/209)

MrChromebox, 2020-08-19:
what was the resolution when the issue first surfaced back then?

tlaurion, 2020-08-19:
The following successful tests were to test 4.12.

I'm trying this https://app.circleci.com/pipelines/github/tlaurion/heads/316/workflows/d636d5c4-d490-40c3-bfab-ac5e4aee95b5/jobs/344

tlaurion, 2020-08-19:
https://app.circleci.com/pipelines/github/tlaurion/heads/319/workflows/e04e0993-5745-4585-9891-b78cd8b9822f/jobs/347 implements https://github.com/osresearch/heads/issues/799#issuecomment-673059028 and pass the failing point of the current build.

@MrChromebox can you take a look at https://github.com/osresearch/heads/compare/master...tlaurion:make_421_fix-CircleCI_unhack?expand=1#diff-8d39c7575f026b40d86f213a4fcea2d6 and advise?

tlaurion, 2020-08-19:
@MrChromebox https://github.com/osresearch/heads/compare/master...tlaurion:make_421_fix-CircleCI_unhack?expand=1#diff-8d39c7575f026b40d86f213a4fcea2d6 

[Fails](https://app.circleci.com/pipelines/github/tlaurion/heads/318/workflows/f18d0d14-2fa8-4943-9ced-6745c3214b3b/jobs/346) at :
```
gcc  -g -O2 -Wl,--export-dynamic  -o make ar.o arscan.o commands.o default.o dir.o expand.o file.o function.o getopt.o getopt1.o guile.o implicit.o job.o load.o loadapi.o main.o misc.o posixos.o output.o read.o remake.o rule.o signame.o strcache.o variable.o version.o vpath.o hash.o remote-stub.o glob/libglob.a   -ldl 
make[3]: Leaving directory '/root/project/build/make-4.2.1'
make[2]: Leaving directory '/root/project/build/make-4.2.1'
make[1]: Leaving directory '/root/project/build/make-4.2.1'
LANG=C MAKE=/root/project/build/make-4.2.1/make /root/project/build/make-4.2.1/make  all
make[1]: Entering directory '/root/project'
modules/coreboot:71: *** recipe commences before first target.  Stop.
make[1]: Leaving directory '/root/project'
make: *** [Makefile:587: all] Error 2
```

[Where](https://app.circleci.com/pipelines/github/tlaurion/heads/319/workflows/e04e0993-5745-4585-9891-b78cd8b9822f/jobs/347 ) continues with: 
```
gcc  -g -O2 -Wl,--export-dynamic  -o make ar.o arscan.o commands.o default.o dir.o expand.o file.o function.o getopt.o getopt1.o guile.o implicit.o job.o load.o loadapi.o main.o misc.o posixos.o output.o read.o remake.o rule.o signame.o strcache.o variable.o version.o vpath.o hash.o remote-stub.o glob/libglob.a   -ldl 
make[3]: Leaving directory '/root/project/build/make-4.2.1'
make[2]: Leaving directory '/root/project/build/make-4.2.1'
make[1]: Leaving directory '/root/project/build/make-4.2.1'
LANG=C MAKE=/root/project/build/make-4.2.1/make /root/project/build/make-4.2.1/make  all
make[1]: Entering directory '/root/project'
2020-08-19 16:20:39+00:00 WGET https://www.coreboot.org/releases/coreboot-blobs-4.8.1.tar.xz
if ! wget -O "/root/project/packages/coreboot-blobs-4.8.1.tar.xz.tmp" https://www.coreboot.org/releases/coreboot-blobs-4.8.1.tar.xz ; then exit 1 ; fi ; mv "/root/project/packages/coreboot-blobs-4.8.1.tar.xz.tmp" "/root/project/packages/coreboot-blobs-4.8.1.tar.xz" \
[...]
```

tlaurion, 2020-08-19:
@MrChromebox https://github.com/tlaurion/heads/commit/51a0de3cc1127f212f746d0c27734c08baf852f6, being a revert of the condition actually failing has passed the point where past condition failed.

tlaurion, 2020-08-19:
@MrChromebox reverting here to test build.... I'm really confused.
https://app.circleci.com/pipelines/github/tlaurion/heads/323/workflows/968f5252-7bf4-4465-99cf-1767acae59ae/jobs/351

tlaurion, 2020-08-20:
@MrChromebox : culprit is bullseye docker image change... Might need more investigation. Reverted to docker:10 docker image in #808

tlaurion, 2020-08-20:
Build happening here https://app.circleci.com/pipelines/github/tlaurion/heads/331/workflows/90fa3dde-b35d-4e68-9fd7-d8194dbaac46/jobs/359