linuxboot/heads #815: Blank screen after internal T430 build flash

Trapacid, 2020-08-29:
I'm currently trying to get heads and coreboot up and running on my ThinkPad T430. I'm not sure what I'm doing wrong, but after `flash.sh -c /media/coreboot.rom` and a reboot, blank screen. Fan runs fine, power button works, but blank screen with no beeps or anything.

My steps are laid out here if someone wants to correct what I'm doing maybe?

1.  External dump 4+8 chips
2. ifdutil 8mb then me_cleaner 
3. External flash clean_8mb to 8mb chip.
4. `make BOARD=t430` and `make BOARD=t430-flash` on a clean Fedora 31 environment
5. External flash `t430-flash.rom` from build/t430-flash/t430-flash.rom to 4mb chip.
6. Boot up T430, plug in USB, mount using `mount-usb`
7. `flash.sh -c /media/coreboot.rom` from build/t430/coreboot.rom
8. `poweroff`, hit power button to turn on, blank screen as described above.

I've spent far too long on this and I'm out of ideas. There isn't a distinctive guide besides the weak documentation on this for the T430.

Any help?

Trapacid, 2020-08-29:
Reached out to @flawedworld and referred me to https://github.com/osresearch/heads/issues/667

Trapacid, 2020-08-29:
Reopened due to same issue. Tried to build coreboot with me_cleaner's shrinked blobs, but same issue persists: blank screen after reboot after internal flashing with flash.sh

tlaurion, 2020-08-31:
There are two things to test here and report given accordingly
- t430 coreboot config needs to be equal to this https://github.com/osresearch/heads/blob/master/config/coreboot-x230.config#L8 
- ME cleaning instructions should follow this https://github.com/osresearch/heads-wiki/blob/master/Clean-the-ME-firmware.md

After having followed the steps above, flash t430-flash externally, then coreboot.rom internally produce by make BOARD=t430 with `CONFIG_CBFS_SIZE=0x700000` modified under coreboot-t430.config.

Trapacid, 2020-09-02:
Setting `CONFIG_CBFS_SIZE=0x800000` to `CONFIG_CBFS_SIZE=0x710000` and recompile fixed internal flashing.

commit cde6e24e (2020-09-02): xx30 boards: fixing CONFIG_CBFS_SIZE=0x710000 for x230 and t430 boards to fix #815 and #667
closed 2020-09-02 by Trapacid