linuxboot/heads #826: Coreboot 4.12 doesn't build in debian:10 docker image

tlaurion, 2020-09-03:
Recent CircleCI build failed after merge of #806: https://app.circleci.com/pipelines/github/osresearch/heads/200/workflows/717b506a-a19f-4a39-a6d4-811710875a5a/jobs/218

#823 didn't help.
#824 is a no go.
#825 will be applied meanwhile, but the same problem will arise for building coreboot 4.12 on docker images.

Any help welcome, as usual.
Current CircleCi config in tree: https://github.com/osresearch/heads/blob/master/.circleci/config.yml

tlaurion, 2020-09-13:
@MrChromebox small advancement... 
https://app.circleci.com/pipelines/github/tlaurion/heads/376/workflows/97c6ef25-8356-4f96-936d-d607a665d4d4/jobs/405

Direct link to log download :
https://circleci.com/api/v1.1/project/github/tlaurion/heads/405/output/127/0?file=true&allocation-id=5f5e90906d7204599ad27df5-0-build%2F424FF711

Not yet sorted by age of log). That would be nice, but have not yet coded it :
https://github.com/tlaurion/heads/commit/2a1f301851b1fb0ce4a16f84ec0bc2dba0012983

tlaurion, 2020-09-13:
@MrChromebox : interesting note.

Remote docker path is:
`/root/project/build/coreboot-4.12/util/crossgcc/build-i386-elf-BINUTILS/`

local is:`
/root/project/build/coreboot-4.12/util/crossgcc/build-i386-elf-GCC/`

If it rings a bell, let me know.

MrChromebox, 2020-09-14:
@tlaurion it should be building both binutils and gcc, as well as quite a few other tools as part of the toolchain. What path is that for exactly?

tlaurion, 2020-09-16:
Weird. fixed under https://github.com/osresearch/heads/commit/53c74fa02ab5159816937898da32dc95e94d5abc

closed 2020-09-16 by tlaurion