linuxboot/heads #831: Have the GPG factory reset functions propose/default to RSA 4096 bits keys, not 2048

tlaurion, 2020-09-13:
@alex-nitrokey @MrChromebox : what approach you prefer?

MrChromebox, 2020-09-13:
@kylerankin's input would be good too

alex-nitrokey, 2020-09-14:
We are [currently using 3072 bit](https://github.com/osresearch/heads/commit/308df1e92a6bf6386d80e80ee2ac2422a624da28), it should be secure for quite some years and is faster than 4096.
I never created a PR for this as I wasn't sure if this is actually interesting for heads. I you like to include this approach, we are happy to create a PR.

alex-nitrokey, 2020-09-14:
3072bit is the gnupg default [since version 2.2.22](https://lists.gnupg.org/pipermail/gnupg-users/2020-August/064056.html) btw. That is to say, once we have updated gnupg, no workarround is needed anymore.

tlaurion, 2020-10-29:
@alex-nitrokey LTS is 2.21. So changing it will need to be manual.

alex-nitrokey, 2020-11-13:
> @alex-nitrokey LTS is 2.2.21. So changing it to 4096 will need to be done manually prior of generating keys.

Do you want me to create a PR for default 4096 based on our approach?

tlaurion, 2020-11-13:
@alex-nitrokey I think it would be a good idea. Putting my nose in the script to do the above in the next days

tlaurion, 2020-11-13:
Oups, wrong ticket. I meant #771

closed 2020-12-02 by tlaurion

MrChromebox, 2020-12-07:
I'm sure I'm missing something, but what's the driving force behind this "issue"? Is a 4096-bit key meaningfully better than 2048-bit now, or in the near future?

techge, 2020-12-08:
> I'm sure I'm missing something, but what's the driving force behind this "issue"? Is a 4096-bit key meaningfully better than 2048-bit now, or in the near future?

That's the point, yes. Many organization are considering 2048-bit not secure enough in the near future (that is, the next years) as processing power increases regularly. So using 3072-bit is recommended for new keys and this is the standard for newer gnupg versions as well.

PR #906 (2020-12-02): Default to 4096 bit for OEM factory reset (fixes #831)
Fixes #831
commit 014e5921 (2020-12-02): Merge pull request #906 from Nitrokey/gpg-default-keylength

Default to 4096 bit for OEM factory reset (fixes #831)