linuxboot/heads #833: exit in rescue shell tries to kill init and generates kernel panic.

Thrilleratplay, 2020-09-15:
If in the recovery shell and no other processes are running, `exit` will throw a kernel panic.

```
Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200

Kernel Offset: disabled
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000200
```

I typed it out of habit trying to reenter `gui-init`.  A system reboot is needed at this point.

MrChromebox, 2020-09-15:
can confirm

tlaurion, 2020-09-15:
@MrChromebox what would be the solution here, trying to exit a shell in single user mode?
@Thrilleratplay your expected behavior would be?

tlaurion, 2020-09-15:
On normal system with login manager, that would go back at login in with user if no authentication mechanisms are in place. 
I have no implementation insights hre without complexifying or adding more tool. Even less since normal UX is gui-init so when you are under shell experience, you normally know what you are doing and do not want to "exit". No?

MrChromebox, 2020-09-16:
> your expected behavior would be?

probably exit the current shell and start a new one

tlaurion, 2020-09-16:
A loop in init where ash is called might do it

Thrilleratplay, 2020-09-16:
@tlaurion  I have no preferred expected behavior, anything that allows the user to continue using heads without manually rebooting the machine.

MrChromebox, 2020-09-16:
> A loop in init where ash is called might do it

unfortunately not: since we exec the recovery shell, there's nothing to exit to. Need to drop the exec, and add a loop. I'll push a PR

tlaurion, 2020-09-16:
Fixed under https://github.com/osresearch/heads/commit/0eb1f692169f6b833315896f65bdc9587e52624d

closed 2020-09-16 by tlaurion