linuxboot/heads #837: Feature request: ASUS KCMA-D8 workstation/server board port to Heads

59iosl30, 2020-09-18:
The ASUS KGPE-D16 board is supported by Heads for nearly 2 years.
 [#134](https://github.com/osresearch/heads/issues/134).

Is it difficult to do the same with its little brother ASUS KCMA-D8 which has pretty much the same platform ?

This completely blob-free board works nice and very stable with Coreboot 4.11 and will be hopefully supported again with next coreboot release 4.13.

I think these great boards are important to Heads, Coreboot and Libreboot and on my opinion those boards are primarly responsible for the great popularity of these three projects in the past 5 years.

See also:
[https://www.reddit.com/r/coreboot/comments/dz0bkf/boards_like_asus_kgped16_the_most_powerful/](url)
[https://www.reddit.com/r/linux/comments/dz2hlt/boards_like_asus_kgped16_the_most_powerful/](url)

tlaurion, 2020-09-18:
@59iosl30 :
Can you point in the threads where you seen that it would be picked up and worked on to be reintegrated in 4.13?

I corrected your links in OP for others to simply click on your links.

Else, 

- there is community effort in putting KGPE-D16 (and would be not so hard to bring KCMA-D8 board config to build it once KGPE-D16 is in [here](https://github.com/osresearch/heads/pull/795) 

- and a funding proposition of required work [here](https://github.com/osresearch/heads/issues/719)

tlaurion, 2020-09-18:
@59iosl30 :

Some additional notes on of my head:

- Current KGPE-D16 board doesn't have TPM support. Should be included by default under coreboot 4.11 if [building was possible](https://github.com/osresearch/heads/pull/795#issuecomment-672002278)
- KGPE-D16 4.11 doesn't have C_ENVIRONMENT_BOOTBLOCK (which is one of the reason it got retired from coreboot 4.12 release) This has implication for VBOOT and Measured Boot since measurements are happening later on on boot path, impacting security and [current build process](https://github.com/osresearch/heads/pull/795#issuecomment-672001639)
- FBWhiptail [is damn slow (workstation board) because of linux support of the AST internal graphic card. That would be easily optimized by upgrading linux kernel.](https://github.com/osresearch/heads/pull/472#issuecomment-670179016)

Your participation is more then welcome on this.

59iosl30, 2020-09-18:
> @59iosl30 :
> Can you point in the threads where you seen that it would be picked up and worked on to be reintegrated in 4.13? I have took some distance following this subject lately, but from what I have got, nobody was actually working on it. I have a dormant working machine here that I could land to a developer, maybe. Pointing interested community developers to me/or where discussions are actually happening right now to have functional dev station might be of interest for the whole community. I could land my station; I do not have time to lead that project anymore.
> 
> I corrected your links in OP for others to simply click on your links.
> 
> Else,
> 
> * there is community effort in putting KGPE-D16 (and would be not so hard to bring KCMA-D8 board config to build it once KGPE-D16 is in [here](https://github.com/osresearch/heads/pull/795)
> * and a funding proposition of required work [here](https://github.com/osresearch/heads/issues/719)

> Can you point in the threads where you seen that it would be picked up and worked on to be reintegrated in 4.13?

59iosl30, 2020-09-18:
@tlaurion 
> Can you point in the threads where you seen that it would be picked up and worked on to be reintegrated in 4.13?

It was just a wish along with others. See URLs above.

> I corrected your links in OP for others to simply click on your links.

Thank you! But what was wrong with the URLs ?

> * there is community effort in putting KGPE-D16 (and would be not so hard to bring KCMA-D8 board config to build it once KGPE-D16 is in [here](https://github.com/osresearch/heads/pull/795)

This sounds good.

> * KGPE-D16 4.11 doesn't have C_ENVIRONMENT_BOOTBLOCK (which is one of the reason it got retired from coreboot 4.12 release) This has implication for VBOOT and Measured Boot since measurements are [...]

I did not tested this:
https://review.coreboot.org/c/coreboot/+/36908/

> * FBWhiptail [is damn slow (workstation board) because of linux support of the AST internal graphic card. That would be easily optimized by upgrading linux kernel.](https://github.com/osresearch/heads/pull/472#issuecomment-670179016)

The internal graphic card on the KCMA-D8 works fine without any glitches.

> Your participation is more then welcome on this.

I can do testing on a KCMA-D8 machine.

tlaurion, 2020-09-18:
> @tlaurion
> 
> > Can you point in the threads where you seen that it would be picked up and worked on to be reintegrated in 4.13?
> 
> It was just a wish along with others. See URLs above.
> 
> > I corrected your links in OP for others to simply click on your links.
> 
> Thank you! But what was wrong with the URLs ?

They were pointing to https://github.com/osresearch/heads/issues/url
> This sounds good.
>
>
> I did not tested this:
> https://review.coreboot.org/c/coreboot/+/36908/
>
>
> The internal graphic card on the KCMA-D8 works fine without any glitches.

This is unclear what you are referencing to. If you are talking about coreboot 4.11 with seabios, this means that Seabios is in charge of the graphic initializatio nand not Linux payload (Heads is coreboot + linux payload) so there is no comparison to do here.
> 
> > Your participation is more then welcome on this.
> 
> I can do testing on a KCMA-D8 machine.

We unfortunately need developers and donors at this point, not really testers.

tlaurion, 2020-09-18:
@59iosl30 added your name on https://github.com/osresearch/heads/issues/692#issue-577966678

tlaurion, 2020-09-18:
@59iosl30 I can put on my todo list the addition of the basic kcma-d8 and generalize https://github.com/osresearch/heads/pull/472 while not having the hardware myself. As you may wonder right now, it would be a coreboot 4.8.1 based board config without TPM support,[ running the server flavor of hte board config](https://github.com/osresearch/heads/pull/472/files#diff-a68a0f3d43cca51e99ef9f005d7a480a) ([pointing to the defconfig](https://github.com/osresearch/heads/pull/472/files#diff-a68a0f3d43cca51e99ef9f005d7a480aR17) generated by selecting kcma-d8 instead of kgpe-d16 and [creating the proper coreboot-config pointed by the board config](https://github.com/osresearch/heads/pull/472/files#diff-16cd2430346d99ae7c23f2ae5554e774), modifying the [paths in coreboot config accordingly](https://github.com/osresearch/heads/pull/472/files#diff-16cd2430346d99ae7c23f2ae5554e774R14-R16), hoping  [the linux config being the same](https://github.com/osresearch/heads/pull/472/files#diff-a68a0f3d43cca51e99ef9f005d7a480aR18), since I cannot test.)

If you want to play around and do a PR testing around that idea, since you own the hardware in question, would be awesome.

tlaurion, 2020-09-18:
@59iosl30 
> I did not tested this:
> https://review.coreboot.org/c/coreboot/+/36908/

See https://review.coreboot.org/c/coreboot/+/36908/2#message-e4c37abe14f11e261f92896b554e647efb9e3ebd

tlaurion, 2021-08-19:
@59iosl30 KGPE-D16 has tpm 1.2 support under coreboot 4.11. 
You have the board? Should not be hard to create a board config for KCMA-D8. Basically duplicating server/workstation/workstation+usb board, copying coreboot config and adjusting paths?

zamaudio, 2021-10-24:
Just wanted to mention that I worked on some raminit cleanups for AMD K8 and AMD FAM10 in 2017.  Looking at the coreboot history, it appears 3 or so relevant commits from librecore were never upstreamed into coreboot.  Feel free to reuse this code as it may save you a bit of work refactoring the ".c" includes unless someone has already fixed this again in coreboot: https://github.com/librecore-org/librecore/compare/41d5d16f65ddb2377016d2ba6ec0d89a6b67468c...67f0495709dbe3ce422256109f49e1e4375a1b71

jtf7, 2024-08-30:
What is our current status on a KCMA-D8 working configuration for use with Heads +Qubes? What level of work or resources would be required to get this working to the same level as the KGPE-d16?