linuxboot/heads #846: x230-nkstorecli reproducibility

daringer, 2020-09-27:
Follow-up for #817 

* the `libstdc++.a` built by the `musl-cross` module is not reproducible, leads to `nkstorecli` also being not reproducible
* first analysis shows that `libstdc++.a` is always of the same size and contents, but the contents (symbols) differ in their ordering
* naive approach using `ar` failed 
* further investigations needed, once `libstdc++` is reproducible, `nkstorecli` also should be

tlaurion, 2022-03-28:
@jans23 Ping

jans23, 2022-03-29:
I did some naive searching and found [this quote](https://blog.beuc.net/posts/Practical_basics_of_reproducible_builds_3/):

> A mere ar x && ar r *.o made the build deterministic.

Does this help?

daringer, 2022-03-30:
> I did some naive searching and found [this quote](https://blog.beuc.net/posts/Practical_basics_of_reproducible_builds_3/):
> 
> > A mere ar x && ar r *.o made the build deterministic.
> 
> Does this help?

As stated above: 
> naive approach using ar failed

so, nope.

But @tlaurion is `libstdc++.a` reproducible as of now on the `master`, this is the first issue to fix, or are you seeing this as part of this issue?