linuxboot/heads #848: Load original flash dumps

AdmerStroh, 2020-10-01:
So I want to revert to the original BIOS. Unfortunately I lost my backup dumps (.bin) of the both flash chips of my x230. Is there any way to extract them from the original bios update or can I load them somewhere else?

Thank you

tlaurion, 2020-10-02:
@AdmerStroh :
> So I want to revert to the original BIOS. Unfortunately I lost my backup dumps (.bin) of the both flash chips of my x230.

Well, is your machine still bootable? In the present ROM, outside of the BIOS region, you will find a backup of the GBE region, ME region and IFD.

>Is there any way to extract them from the original bios update or can I load them somewhere else?

You cannot download GBE from anywhere, where #703 includes a patch and instructions to generate the MAC address and the resulting blob. ME blob can be downloaded as instructed too.

May I ask what you were seeking to do with original ROM? please tag @tlaurion in your reply.

AdmerStroh, 2020-10-03:
@tlaurion I can still boot heads. My actual problem is that I once flashed the experimental x230_external_flash config (which in my case seems to not support hotp). Now I can only flash updates for this exact config internally. Trying to flash some other target rom like x230 or x230_hotp_verification internally renders the machine unbootable which I can only fix by flashing the external flash config externally. I think the problem is that there is some kind of region mismatch in the 8M chip caused by the incompatibilities between the x230 and x230_external_flash ROMs. Thus I wanted to completely start from zero by flashing the original ROM than flash skulls externally and over their feature flash standard heads 12M rom internally.

tlaurion, 2020-10-18:
OK I get your problem. X230-external-flash modifies the flashrom call for future upgrades to flash whole ROMs,  not just the BIOS region (check other board configs to see the different call specifying just BIOS region. Calling that would  permit you to switch back to x230 rom, having me neutered and deactivated with static mac address generated gbe region being untouched )

I will get back at x230-external-flash soon, sorry for the inconvenience. That board fully flashes the whole 12 MB rom, where all the others respect the CBSE region and do not overwrite the rest of the 12MB flash. Let me know if you need more help.

If you done have external backup, you have lost your original GBE region.

tlaurion, 2022-03-28:
Flashing internally with a flashrom command as explained under https://osresearch.net/Prerequisites#legacy-vs-maximized-boards should have fixed your issue @AdmerStroh . Please reopen otherwise.

closed 2022-03-28 by tlaurion