linuxboot/heads #870: x220, add GPG key to running Bios and reflash not working

userongihu, 2020-10-26:
After booting Heads I get the GPG Management Menu,but add GPG to running menu wont't work.I found this https://github.com/osresearch/heads/issues/675 .I can get flashrom working internally following this guide,but I am stuck at the and 6. step not knowing how to add checksums.Also not quite sure about what to do at step 4.After step 3 I get this output
https://postimg.cc/S2zKRgK3
At step 4 I ran bin/flash.sh -c chipname and the output was same like on the uploaded pic.Could some explain how I can get this thing working?

tlaurion, 2020-10-26:
@MrChromebox seems like last version of flashrom dropped support for x220? @SebastianMcMillan ?

tlaurion, 2020-10-26:
@userongihu @MrChromebox @SebastianMcMillan :
past statement for that board was : https://github.com/osresearch/heads/blame/83c22f3e4a154707b8cc00f1267d6591b6c24e41/boards/x220/x220.config#L36

MrChromebox, 2020-10-26:
@tlaurion flashrom 1.2 removed the need for `laptop=force_I_want_a_brick` and for most platforms `ich_spi_mode=hwseq` too, but I'm guessing the x220 still needs it (so `-p internal:ich_spi_mode=hwseq`).

MrChromebox, 2020-10-26:
note that with hardware sequencing enabled, there is no need to specify the chip via `-c xxx`

userongihu, 2020-10-26:
thank you for your answers.Does it means that wherever flashing is used "-p internal:ich_spi_mode=hwseq" must be added to the flashing command?

MrChromebox, 2020-10-26:
@userongihu it means that CONFIG_FLASHROM_OPTIONS needs to be updated for the board config to amend the parameter list.

for updating your device currently (after building with the updated params), would recommend simply flashing from the recovery shell:
`mount-usb`
`flashrom -p internal:ich_spi_mode=hwseq -f -n --ifd -i bios -w /media/<filename to flash>`
`reboot`

userongihu, 2020-10-26:
Thank you MrChromebox.Do I get it right that you updated the missing parameters?So all I need to do is make a new build running "make BOARD=x220" and then running the commands you mentioned above?Sorry for asking such a noob question...

MrChromebox, 2020-10-26:
@userongihu I submitted a patch (pull request) to add the missing param, but it has not been accepted (merged) yet, so until that time you would need to make the change manually on your local copy before rebuilding

tlaurion, 2020-10-26:
@userongihu please try 
`
flashrom -p internal:ich_spi_mode=hwseq -f -n --ifd -i bios -w /media/<filename to flash>`
If you report this as successfull, I will merge #871 to fix problem in newer builds.

tlaurion, 2020-10-26:
@SebastianMcMillan ?

userongihu, 2020-10-27:
@tlaurion ,here is the output of the flash https://postimg.cc/tYvk7pLS/2898bc5f
I rebooted,then tried to add GPG key to running Bios but it still won't flash.I am going to 
change the flash-params in the configs,rebuild and try again.

userongihu, 2020-10-27:
When rebooting after flashing there are still no GPG keys on the keyring.Setting up TOTP works.After rebooting TOTP also persists,saying because I could get GPG to the key ring ,but after a reboot the keys were somehow gone.Also when hitting the option "update checksums and sign all files in /boot" it requires a gpg-card,but it doesn't recognize my card when plugged in,although Heads recognizes the card insofar,that it is possible to generate PGP keys with card.Maybe I have to add
the device to the configparams before building.Also I do have now a persisting "thinkpad x220 boot menu" that wasn't there at the beginning.

Update:@tlaurion,I rebuilt and ran flashrom with the updated flashromparams that you suggested.Rebooted and tried "add PGP keys to running Bios and reflash" but failed and it returned to the usual screen.But after running "add PGP keys to standalone Bios and reflash" option it returned to a screen that says "TOTP secret cannot be unsealed".I never got that screen before.But I keep on getting the message "No GPG keys in the key ring".When running bin/gpg-gui.sh it imports keys https://postimg.cc/n9ryQZ1H.The error message "No Board....." below,I am getting only with the build that has the new flashparams.

userongihu, 2020-10-27:
To be more specific the screen said "ERROR:TOTP generation failed!" not "TOTP secret cannot be unsealed".

userongihu, 2020-10-27:
https://postimg.cc/n9ryQZ1H isn't working anymore,see https://postimg.cc/YGvJ9Gmq instead.

tlaurion, 2020-10-27:
> @userongihu please try
> ` flashrom -p internal:ich_spi_mode=hwseq -f -n --ifd -i bios -w /media/<filename to flash>`
> If you report this as successful, I will merge #871 to fix problem in newer builds.
> 
> Please also tag me with result so that I can see it right away.

Ok. I'm confused.
So following :
```
mount-usb
flashrom -p internal:ich_spi_mode=hwseq -f -n --ifd -i bios -w /media/<filename to flash>
reboot
```

You were able to flash new rom manually from recovery shell. Good.
What rom have you flashed?

On host buildsystem:
```
cd heads
git fetch origin
git reset --hard
make BOARD=x220 CPUS=2
(should work unless gpg2 toolstack now is too big. We do not have enough x220 boards partitipants, nor the board being under CI for checking this kind of regressions right now, so it might fail.)
sudo mount /dev/sdX /media
cp build/x220/heads-x220-.....gcd6ba01.rom /media
sudo umount /media
```

On your x220:
```
mount-usb
flashrom -p internal:ich_spi_mode=hwseq -f -n --ifd -i bios -w /media/<filename to flash>
reboot
```

Then on next reboot, all commands requiring to access the rom by doing read/write commands through flashrom will take the new :
https://github.com/osresearch/heads/blob/master/boards/x220/x220.config#L39
(Which is required when adding GPG public key, updating firmware internally...)

x220/t420/xx20 boards testers (@SebastianMcMillan @techge @eganonoa @shamen123 @Thrilleratplay)
https://github.com/osresearch/heads/commit/cd6ba01429fedfa2edd004f524e772fa5dd5fe06
and 
https://github.com/osresearch/heads/commit/e3519f2ecd8e396d92905317ac5151e590388b02
were merged without being tested per [board owners](https://github.com/osresearch/heads/issues/692).

SebastianMcMillan, 2020-10-27:
> e3519f2

I've been having tons of issues actually building heads, which is why I
haven't been able to do anything right now. Building in anything newer than
Debian 10 straight-up is broken and I've got no time right now to set up a
Debian 10 system. I'll get a Debian 10 system setup this weekend for me to
take a look though. However, here's my understanding:

GPG key is added and flashed, but is lost on reboot. This sounds like an
issue we had a while back where there wasn't enough space to add the key.

userongihu, 2020-10-27:
@tlaurion,what I did is going to board/x220/x220.config and changed the  CONFIG_FLASHROM_OPTIONS to the command you suggested.Then I ran "make BOARD=x220" and after quite a while the build was ready without any error notification.From shell I ran "mount-usb"(the newly build coreboot.rom was on the USBdrive of course) and then ran the flashing command,which gave me a positive
output,means no errors.Then rebooted straight into the "ERROR:TOTP generation failed" ,but I could set up the TOTP thing,means a 6digit number gets generated whenever I boot up,tried it already a couple of times.
Now whenever I boot up the first srceen says ERROR:GPG keyring empty,then when trying "add key to running Bios and reflash" flash is not working ,instead it changes
back to the ERROR:GPG keyring empty" screen.When running  "add key to standalone Bios and flash" it doesn't flash but changes into the"x220 boot menu"
screen instead.

MrChromebox, 2020-10-27:
@userongihu run gpg-gui.sh from the recovery shell, and choose the option to add the key and flash. Should give more error info when it fails and drops back to shell vs main menu

userongihu, 2020-10-27:
I flashed the corebootrom.rom that was in build/x220 after the buil was finished

userongihu, 2020-10-27:
@MrChromebox,I just ran that command and the output is the same like on here
https://postimg.cc/YGvJ9Gmq

MrChromebox, 2020-10-27:
@userongihu you're either using an old build, or building from old sources. The output filename should be `heads-X220-<build version>-<commit hash>.rom

userongihu, 2020-10-27:
From which source should I compile the right build?Should I git clone from this link https://github.com/osresearch/heads.git ?

userongihu, 2020-10-27:
I followed the instructions posted here http://osresearch.net/

userongihu, 2020-10-27:
Are these 2 commands necessary to get the latest build: "git fetch origin" and
"git reset --hard" ?

MrChromebox, 2020-10-27:
@userongihu `git fetch && git reset --hard origin/master`  

if you don't specify what to reset to, then it only resets any uncommitted changes to your current repo/branch, it doesn't reset to the latest remote sources

userongihu, 2020-10-27:
@MrChromebox,just rebuild following the guide mentioned above but the build was as usual
build/x220/coreboot.rom.Am I missing something?git cloned into git clone https://github.com/osresearch/heads and sticked to the steps shown above.

userongihu, 2020-10-28:
@tlaurion,before making the build,I followed the instructions in blobs/x220/read.me,which means 
I ran ifdtool -x Lenovo.rom and got four diffrent folders and than i ran ./extract.sh with -m and -i
option and got a lot of output.I believe that shrank ME.Saying because prior of installing heads
I made an coreboot install including neutering ME, and after that I checked with mei-amt-checker 
if ME was actually neutered and the output said it was.That makes me think that build/x220/coreboot.rom has a neutered ME.

userongihu, 2020-10-28:
What I mean is,that the output I got when running extract.sh -m -i in heads/blobs/x220 was producing the same output that I got when I neutered ME during the coreboot installation that
I made several weeks before installing Heads.

userongihu, 2020-10-28:
@tlaurion,the contents of my heads/blobs/x220 folder after running the mentioned commands is

extract.sh                         flashregion_3_gbe.bin  me.bin
flashregion_0_flashdescriptor.bin  gbe.bin                readme.md
flashregion_1_bios.bin             ifd.bin                romread1.rom
flashregion_2_intel_me.bin         layout.txt

userongihu, 2020-10-28:
here is the tool I used to check ME wether it was neutered or not
https://github.com/mjg59/mei-amt-check

userongihu, 2020-10-28:
@tlaurion I ran the command of this guide https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-coreboot#neutralize-and-shrink-intel-me and the new content of heads/blobs/x220
is 
00003000:00017fff  extract.sh        layout.txt  me_shrinked.bin  readme.md
00018000:007fffff  ifd_shrinked.bin  me_cleaner  out.bin          romread1.rom

userongihu, 2020-10-28:
@tlaurion never mind ,I am greatful for the info I am getting here.Right now I am trying to make another  build using the maximum possible shrinked ME.Let's see how it works.

userongihu, 2020-10-28:
@tlaurion ,the build is finished now,no errors.but it is called heads--v0.2.0-940-ge7faac2.rom .It is located in folder build/qemu-coreboot.There is no mention of x220 whatsoever within the /build folder.I ran "make BOARD=x220 CPUS=2" command,didn't ran nothing about qemu.Is that the right .rom?

MrChromebox, 2020-10-28:
@userongihu for some reason the board config value isn't being set properly, hence why it's missing from the filename, and why flashing internally (when adding a GPG key) fails.

userongihu, 2020-10-28:
@MrChromebox,would you suggest to delete the whole heads directory and git clone it and rstart from a new one?What would be the best way with dealing with it?

MrChromebox, 2020-10-28:
a fresh start certainly wouldn't hurt. git clone, copy blobs into correct location, `make BOARD=x220` -- that's it

tlaurion, 2020-10-28:
> @tlaurion,before making the build,I followed the instructions in blobs/x220/read.me,which means
> I ran ifdtool -x Lenovo.rom and got four diffrent folders and than i ran ./extract.sh with -m and -i
> option and got a lot of output.

For the sake of clarification.

Running` ./blobs/x220/extract.sh -f <romdump> -i ~/path/to/ifd -m ~/path/to/me_cleaner` will still result in running me_cleaner command [in the form](https://github.com/osresearch/heads/blob/master/blobs/x220/extract.sh#L58) of : 
`$MECLEAN -O "$BLOBDIR/me.bin" -r -t "$extractdir/flashregion_2_intel_me.bin"`

This instructs me_cleaner to extract neutered+deactivated into "$BLOBDIR/me.bin" , while relocating (r) and trimming (t) modules. 

That is nice. You won't have ME. 

But that freed space is not given back to the [BIOS region](https://github.com/osresearch/heads/blob/master/config/coreboot-x220.config#L7) if the provided [ifd](https://github.com/osresearch/heads/blob/master/config/coreboot-x220.config#L12) is not modified so that the BIOS region (that you flash from within Heads) does not benefit of that freed space, which would only happen if the me_cleaner call would change from:
`$MECLEAN -O "$BLOBDIR/me.bin" -r -t "$extractdir/flashregion_2_intel_me.bin"`

to (please [review this](https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-coreboot#neutralize-and-shrink-intel-me))
$MECLEAN -r -t -d -O unneeded.bin  -D "$BLOBDIR/ifd.bin" -M "$BLOBDIR/me.bin" "$extractdir/flashregion_2_intel_me.bin"
Which was adapted from[ me_cleaner documentation](http://osresearch.net/Clean-the-ME-firmware/#how-to-disabledeactive-most-of-it):
`python me_cleaner.py -r -t -d -O out.bin -D ifd_shrinked.bin -M me_shrinked.bin original_dump.bin`

Which is the only way to reuse liberated ME space for the sake of survival of this board in the future.

tlaurion, 2020-10-28:
@MrChromebox @userongihu : My point here is that the failure to add gpg public key is probably coming from inconsistencies between ifd region description of BIOS region (not expended) from non liberated ME space.

If ifd is not modified, then there is not enough space for actually defined BIOS region from CBFS defined coreboot region.
@SebastianMcMillan 
https://github.com/osresearch/heads/blob/master/config/coreboot-x220.config#L7 states:`
CONFIG_CBFS_SIZE=0x750000`

Which SHOULD depend of a extract.sh that is consistent of that region. This is not the case right now.
all extract.sh scripts should be modified. And specify an origin ROM version from Lenovo that should be the same. In BOARD config, or better, outputted in the extract.sh script with proper validation.

I tend to give too much explanations. Shot point is that 
t420: 
https://github.com/osresearch/heads/blob/master/blobs/t420/extract.sh#L58
and
https://github.com/osresearch/heads/blob/master/config/coreboot-t420.config#L7
and
http://osresearch.net/t420-flashing/

x220:
https://github.com/osresearch/heads/blob/master/blobs/x220/extract.sh#L58
and
https://github.com/osresearch/heads/blob/master/config/coreboot-x220.config#L7
and
http://osresearch.net/x220-flashing/

Should be congruent. Where http://osresearch.net/Porting/ has been updated to explain what I just did.

Can somebody (@SebastianMcMillan @techge @eganonoa @shamen123 @Thrilleratplay) update the doc with one successful steps?

MrChromebox, 2020-10-28:
@tlaurion I don't believe that to be true, given the error reported and the lack of board name in the output file. Though they may encounter that problem once they resolve the board issue.

tlaurion, 2020-10-28:
@MrChromebox : 

Extreme short version: 
Its impossible, even theoretically, to have `CONFIG_CBFS_SIZE=0x750000` on a board having 8mb of flash but if extract.sh me_cleaner call deactivates, neuter, relocate, export reduced me.bin, then extract consequent ifd.bin. ([x230 board](https://github.com/osresearch/heads/blob/master/config/coreboot-x230.config#L4) has `CONFIG_CBFS_SIZE=0x700000` following https://github.com/osresearch/heads/issues/608 and https://github.com/osresearch/heads/issues/667 learned errors.

userongihu, 2020-10-29:
Thanks a ton for all your support.When running ./extract.sh I left the  ifd-part in its original state.I just ran the me_cleaner.Now I understand that space needs to be created on the chip for making heads to be working.That must be the reason why the build for the x220 ins't working.For the sake of checking I made the builds for the x230 and it worked right away.
As far as the x220 is concerned, I will have to look into the me_cleaner stuff a bit more,in order to be sure that the right command is used to make space availabe on the chip.

userongihu, 2020-10-29:
@tlaurion,I made a build for the x220 having in mind what you said about liberating the ME space in order to make it usable for heads.I ran that command and had this output 
https://pastebin.com/d7VNjwQg Then ran make BOARD=X220 CPUS=2 and when the build was finished the output was https://pastebin.com/VLg7Z3Js
What does dirty-rom mean?Would this build be usable?

userongihu, 2020-10-29:
Maybe I should add that the content of blobs/x220 folder at the time of building heads was https://pastebin.com/udgErtJ2

userongihu, 2020-10-29:
It seems to work now.I flashed the -direty.rom,booted and chose "add GPGkeys to running Bios and reflash" and it did reflash.Then made a reset of TMP ,scanned the QR and it generates 6digit codes.Rebooted and saw that the GPGkey on the keyring persists,but when hitting "default boot" it says "no default boot options configured".How to proceed from here?
While resetting the TPM I had to enter passwords and I rememeber a notification said something like "params:srk not allowed" or "...not possible".

userongihu, 2020-10-29:
There is a loop when choosing "default boot",it always switches back to "no default boot option configured".

MrChromebox, 2020-10-29:
you need to select a default boot option (grub entry) and then sign that.

'dirty' is the status of your local git repo, it just means you have uncommitted changes in your local tree/branch

userongihu, 2020-10-29:
@MrChromebox ,I don't see any grub entry.Where do I have to look for that?By the way, the TOTP nr on the x220 is in sync with the nr, in the authenticator app on my smartphone.So,that works as well :)

userongihu, 2020-10-29:
Oh,I know what you mean.It's time to install the OS!Lets try.

userongihu, 2020-10-29:
Hey guys,again thanks a ton.Heads works now.I just booted Ubuntu-live from USB-drive and connecting to the internet worked great.I still don't undertsand how to configure "default boot".In order to boot ubuntu-live,I went to the "boot options" menu and chose the according USBdrive.But how do I sign a distro?I guess installing a distro on HDD would go the usual way.But signing the .iso before installing?

tlaurion, 2020-10-29:
> @tlaurion,I made a build for the x220 having in mind what you said about liberating the ME space in order to make it usable for heads.I ran that command and had this output
> https://pastebin.com/d7VNjwQg 

Good
Then ran make BOARD=X220 CPUS=2 and when the build was finished the output was https://pastebin.com/VLg7Z3Js

Good. So basically, what you say is that changing in extract.sh
`
$MECLEAN -O "$BLOBDIR/me.bin" -r -t "$extractdir/flashregion_2_intel_me.bin"`

to:

`$MECLEAN -r -t -d -O /tmp/unneeded.bin -D "$BLOBDIR/ifd.bin" -M "$BLOBDIR/me.bin" "$extractdir/flashregion_2_intel_me.bin"` works. 

Right?
> What does dirty-rom mean?Would this build be usable?

tlaurion, 2020-10-29:
> Hey guys,again thanks a ton.Heads works now.I just booted Ubuntu-live from USB-drive and connecting to the internet worked great.I still don't undertsand how to configure "default boot".In order to [...]

Is http://osresearch.net/InstallingOS/ answering your questions? if not can you open a new issue and close this one after answering previous questions?

If it works, i'll do a Pull Request (PR) to fix the extract.sh scripts themselves so that they correspond to `CONFIG_CBFS_SIZE=0x750000` as currently defined under both x220 and t420 coreboot config files.

tlaurion, 2020-10-29:
@SebastianMcMillan @techge @eganonoa @shamen123 @Thrilleratplay @userongihu :
Please test and report on #877

userongihu, 2020-10-31:
@tlaurion,I just booted a live-distro from usb and in order to make a regular install from installer.If that doesn't work I will try it the way it is shown on the link you posted.I will report on how things went.

tlaurion, 2020-10-31:
Know that luks should be in version 1, not luks2 since not supported by Heads for the moment until opened issue is fixed.

Qubesos latest works

userongihu, 2020-11-01:
@tlaurion ,I just finished the Qubes Install.I chose the latest version.It ended without error messages but booting it doesn't work somehow.The regular install that I tried yesterday doesn't work either.USB booting a non-live distro does work,no problems.Easy like booting a live-distro.What do you mean that I have to consider luks1 for luks2 is currently not supported
by heads?Also,after the install that I made yesterday,after booting and hitting "default boot"
I get a message that says "ERROR:missing hash files",then when choosing "uodate checksums and sign all files in /boot" it takes me to a screen that requires to plug in and confirm a GPGcard.After plugin in and confirming the GPGcard it takes ages to get a messages that says "gpg:no default secret key ,signing failed.I got that message already before,since I noticed that the card has only limited function on heads.Some tasks were possible some not.So,/boot is now not signed anymore ,which doesn't matter much because booting from HDD wasn't possible in first place.How to proceed from here?

userongihu, 2020-11-01:
@tlaurion,I put the Qubes***.iso ans Qubes***.iso.asc and qubes***.key.asc on a usbdrive
and imported the key and ran usb-scan.It showed the Qubes***.iso,from there chose the option to install,waited until finished and rebooted,but it is not working.Heads somehow does not recognize the HDD,although the installation doesn't give out any errors.
The TOTP on the x220 and the TOTP on the authenticator-app of the smartphone keep on generating the same numbers.Does that mean that the Bios is in a "good-signatured",non-tampered state?Asking just to be on the safe side.Don't want to make wrong assumptions.
.

tlaurion, 2020-11-01:
> @tlaurion ,I just finished the Qubes Install.I chose the latest version.It ended without error messages but booting it doesn't work somehow.

@userongihu Which QubesOS version?

- Supported : [LUKSv1 from default installation disk partitioning option ](https://mirrors.edge.kernel.org/qubes/iso/Qubes-R4.0.3-x86_64.iso)
- Unuspported: [LUKSv1 from default partitioning options ](https://ftp.qubes-os.org/iso/Qubes-R4.1.0-alpha20201014-x86_64.iso) because of [cryptsetup under Heads being 1.75 and not supporting LUKS2](https://github.com/osresearch/heads/pull/876)

> The regular install that I tried yesterday doesn't work either.USB booting a non-live distro does work,no problems.Easy like booting a live-distro.What do you mean that I have to consider luks1 for [...]

So the public key was not injected in ROM? I'm not sure I understand the path here. Current code is detecting:
IF no OS installed + no public key in ROM = prompt user to add key/generate key in ROM. So it seems that you went to install QubesOS without having first a public key injected in ROM? please detail.

>I got that message already before,since I noticed that the card has only limited function on heads.
Could you please detail what you mean by this? Required GPG functions are supported (signing/verified/encrypt).

>Some tasks were possible some not.So,/boot is now not signed anymore ,which doesn't matter much because booting from HDD wasn't possible in first place.How to proceed from here?

If I underatand well, you do not have a public key in ROM. You have to go in GPG menu, and generate key. You can also go in main menu as of now (until #874 ) and generate key, inject, totp/hotp seal (you do not have HOTP support in rom as of now. Which USB security donlge are you using), and sign /boot content.

When /boot config is signed, the next step is to define a boot default under boot options, which will give you the possibility to define a default boot option and should prompt for setting a Disk Unlock Key passphrase ,which would make the TPM release the Disk Unlock Key by the TPM if TPM measurements are valid.

userongihu, 2020-11-01:
@tlaurion,I installed Qubes 4.0.3
When booting heads,going into "GPG option" and looking in "list GPG on your keyring" there is
the GPGkey which was already there,since I got heads successfully working.It is still the same GPGkey.This might answer your question wether my GPGkey was injected in the ROM or not.

Since yesterday /boot is not signed anymore and trying to sign requires a gpg-card to inserted and confirmed.I plug in my nitrokey-card and after quite some time there is a message saying
gpg:no default secret key ,signing failed",although there keys on the card,but not the GPGkey
which is on the heads-keyring.
Before yesterdays try to make a regular install inside heads my /boot was signed.I was using an OS that I installed before having installed heads.And I did try a ot of times to set up default
boot but the HDD never showed up anywhere.That's why I think that heads does not recognize the HDD,if it did recognize HDD it would show up somewhere,since USBdrives are showing uo under "USB...".
Should I make a fresh OS-installation on another computer and reflash heads internally with the "add GPGkeys to running Bios and reflash" and then start from new in order to see how things work?Asking because till yesterday the /boot was definitly signed.

MrChromebox, 2020-11-01:
> Should I make a fresh OS-installation on another computer and reflash heads internally with the "add GPGkeys to running Bios and reflash" and then start from new in order to see how things work?Asking because till yesterday the /boot was definitly signed.

no. I would simply perform the OEM reset from the options menu, and let Heads generate a fresh key and sign everything, and make sure that works. That rules out any issues from your key or it being in the correct places.

userongihu, 2020-11-01:
@tlaurion,just ran "add GPGkey to running Bios and reflash" and the internal flash worked.Made a TMP reset,which worked as well.But hitting "default boot",I get a message
that says "ERROR:missing hashfile",then selecting "yes" it aks for the GPGcard.But why?My GPGkey is already on the heads-keyring.Should I import the heads- GPGkey to the GPGcard?

[11 more replies]

closed 2020-11-01 by tlaurion