linuxboot/heads #873: Adding xx20 boards into CI (x220, T420)

tlaurion, 2020-10-27:
Quick reflexion to launch debate for xx20 boards [owners/enthusiasts](https://github.com/osresearch/heads/issues/692):
@SebastianMcMillan @techge @eganonoa @shamen123 @Thrilleratplay

If the instructions of installation were specifying to
`flashrom -p ch341_a -c XXXXXXXX --ifd BIOS -w heads.rom`

- Wouldn't it be possible to produce valid externally/internally flashable roms which just doesn't touch ME/GBE and respect origin IFD board?
- Then, without releasing blobs bases repositories, have instructions showing how to unlock IFD, reduce ME, expend BIOS region on original ROM and then have another x220 board, which produces ROMs for that larger BIOS region, permitting other tools to merge in?

Asking the question to xx20 boards testers cause this board is not enough tested and consequently of the past merged commits that passed to all other boards, I believe that those boards might become less and less supported if at least regressions are not at least validated by CIs.

- Flashrom was invalid and blindly corrected without testing:
https://github.com/osresearch/heads/commit/cd6ba01429fedfa2edd004f524e772fa5dd5fe06
https://github.com/osresearch/heads/blob/master/boards/x220/x220.config#L39
- GPG toolstack got upgraded which fitted in other boards but probably not to xx20 boards:
https://github.com/osresearch/heads/commit/e3519f2ecd8e396d92905317ac5151e590388b02
- Flashrom was updated but not tested by any xx20 boards users
https://github.com/osresearch/heads/commit/3c24460f1a4976704d491bfd985ce0ee9fcc84ec

Time to change our approaches. Otherwise we will have to fixate boards to use a specific versions of modules in board configs, since upgrading them is becoming more and more difficult, each version of a module upgrade normally increasing in size. (coreboot, kernel, gpg2, etc)

The next approach, if noone joins in, will probably be to fixate modules version to last known working version prior of pushing changes upstream. That would mean, for example, that the xx20 would be stuck to old versions of everything.

SebastianMcMillan, 2020-10-27:
>    cd6ba01
>    e3519f2
>    3c24460

There simply isn't enough space to install heads on a XX20 system without
stripping the ME at all, not even the xx30 install-phase inits would be
small enough to fit on an untouched IFD/ME xx20 units.

Thrilleratplay, 2020-10-27:
@tlaurion 
>Wouldn't it be possible to produce valid externally/internally flashable roms which just doesn't touch ME/GBE and respect origin IFD board?

Externally, no.  There is only one 8mb rom chip and will need to contain the ME/IFD/GBE partitions to flash externally.  Internally, perhaps with the  `-p internal:ich_spi_mode=hwseq` flag merged in #871  but I have yet to test this.  The main issue with using the `-c XXXXXXXX` flag is that user needs to remember which chip it is each time they update.  I didn't know that had changed for the x220 and will try it out later with coreboot.

With additional scripts/modules, a solution similar to #703 could be implemented.  The GBE is the same for both models.

tlaurion, 2020-10-27:
@SebastianMcMillan : so only:
> x220-neutered (having CBFS for BIOS region maximized, taking into consideration that ME is cleaned and IFD corresponding to CBFS region defined in coreboot-xx20.config)

tlaurion, 2020-10-27:
@SebastianMcMillan : I edited OP.

SebastianMcMillan, 2020-10-27:
Also, I just noticed #871 being made. I know for a fact those changes were
made a while back and merged in, as my X220 has the flashrom options with
ich hwseq enabled.

Which commit reverted that?

tlaurion, 2020-10-28:
@SebastianMcMillan this is linked to https://github.com/osresearch/heads/issues/870#issuecomment-717590672

tlaurion, 2020-10-28:
Can you give the commit in your /etc/config?

tlaurion, 2020-10-29:
> Also, I just noticed #871 being made. I know for a fact those changes were made a while back and merged in, as my X220 has the flashrom options with ich hwseq enabled. Which commit reverted that? Edit: when flashrom 1.2 support was added, X220 flashrom options got changed and had that reverted. I added that specifically since it prevented the issue with different flash chips needing the -c flag.

Yeah. Seems like a regression was added https://github.com/osresearch/heads/blame/e3519f2ecd8e396d92905317ac5151e590388b02/boards/x220/x220.config#L39 thinking it was not required anymore.

tlaurion, 2020-10-29:
First step is #877 
Where gbe could then be generated, and where ME could be downloaded and extracted, so only a dead mainboard ifd should be provided by community to be dropped in github per xx20 blobs dir.

tlaurion, 2020-11-08:
We might have a problem
https://github.com/corna/me_cleaner/issues/343

tlaurion, 2020-11-09:
Won't happen until https://github.com/osresearch/heads/pull/877

tlaurion, 2020-11-25:
@Thrilleratplay Clarified xx30 board config with explanations and proper seperations of modules with explanations under https://github.com/osresearch/heads/commit/b981004710afa1bdf066adccb17675df1d5d862d

Thrilleratplay, 2020-11-27:
@tlaurion Sorry, I meant the direction the entire PR was going.  However, that was incredibly useful. The comments and missing functionality from the xx20 boards has been merged under #912,

tlaurion, 2022-03-28:
fixed under maximized xx20 boards

closed 2022-03-28 by tlaurion