linuxboot/heads #874: OEM factory reset option showed also under GPG menu

tlaurion, 2020-10-27:
> [...]OEM factory reset would fit better/be duplicated in the GPG menu, so that in case of no GPG key being present, the OEM factory reset option would be presented to the user as a possibility instead of just having the choice to go manual.

_Originally posted by @MrChromebox in https://github.com/osresearch/heads/pull/869#issuecomment-717508231_

MrChromebox, 2020-10-28:
so currently at boot, gui-init will check if there is a GPG key in the ROM, and if /boot files are signed. If no to both, then it prompts the user to perform a factory reset. So I'm not sure duplicating the option in the GPG menu is needed. So question is if it's better to have in the main options menu, or one level down under GPG options

tlaurion, 2020-10-28:
@MrChromebox the thing is that it's a meta function, doing USB Security dongle reset but also reset the TPM and sign boot config. That's why I would duplicate.

tlaurion, 2022-03-28:
Fixed

closed 2022-03-28 by tlaurion