linuxboot/heads #880: Configure default boot device

jans23, 2020-10-30:
If users use by default an OS installed on a USB drive, she would need to select to boot from USB drive manually at each boot. It would be more convenient to allow configuring boot drives, like it is possible in ordinary BIOS.

tlaurion, 2020-11-01:
@jans23 so /etc/config.user overlay specifying we depend on usb modules and usb storage to detect /dev/sda1 or alike?

tlaurion, 2020-11-01:
@jans23 : if we are talking about installed OS, we would reuse verified /boot with user public key? I think this is a good idea with the added protection of having that drive actually implementing RO mode if desired.

What are your implementation ideas here?

tlaurion, 2022-03-28:
@jans23  is this still relevant since factory rest fixes this?

jans23, 2022-03-29:
Please elaborate, how factory reset fixes this issue.

tlaurion, 2022-03-29:
Sorry, I read quick and in diagonal. The OEM factory defautl searches and sets a disk as DEFAULT_BOOT through clean boot detection. But booting from USB by default is still not supported @jans23