linuxboot/heads #883: Augment output to offer better boot options to users

tlaurion, 2020-11-01:
>> Then I powered up and hit default boot
and I got the message "The following files failed the verification process:/boot/grub2/boot.cfg This could indicate a compromise! Would you like to update your checksums now?"

>this is expected after the first boot of Qubes, since it updates the grub config

The user should have a better understanding the the first option is the default boot (dynamic) option

>> Finally I have now the "select boot option" message on the screen and there are 4 different options to choose from.Which one should I choose?

>Heads orders the boot options in the same order as they appear in the grub config. Generally speaking, the first / shortest entry is the correct one, but there are exceptions (like with some live USBs)

How can we clarify that menu?

_Originally posted by @MrChromebox in https://github.com/osresearch/heads/issues/870#issuecomment-720149774_

MrChromebox, 2020-11-01:
> The user should have a better understanding the the first option is the default boot (dynamic) option

sorry, first option where?

> How can we clarify that menu?

we could add an asterisk or something to indicate the default option from the grub config, but we'd need to make the parser a bit smarter. Currently it's enumerating options from the main menu and any submenus together, and the text doesn't match exactly due to some workarounds for whiptail. 

I think to really improve the boot menu selection, we'd need to more closely mimic the grub menu text and structure, vs the current approach of displaying everything at the same level, and showing the kexec params directly vs the config entry text

tlaurion, 2020-11-02:
> > The user should have a better understanding the the first option is the default boot (dynamic) option
> 
> sorry, first option where?

Sorry. In the selection of a new boot option, from Other boot options, show boot menu.
> we could add an asterisk or something to indicate the default option from the grub config, but we'd need to make the parser a bit smarter. Currently it's enumerating options from the main menu and [...]