linuxboot/heads #887: t420 flash.sh fails on chip detection

alexmaloteaux, 2020-11-03:
Hi 
on my t420, flashing with a ch341 extrernally or with -p internal always fails unless i specify the chip manually with
-c MX25L6406E/MX25L6408E.
Im then facing the same issue in heads from flash.sh script when dumping the spi for reflashing.
Only way for me to use heads was to edit the t420.config and add that parameter like this : 

export CONFIG_FLASHROM_OPTIONS="--force --noverify-all -p internal -c MX25L6406E/MX25L6408E --ifd --image bios"

Dunno if it is specific to my board or recent revision of flashrom so i created an issue instead of a pr.

BR

tlaurion, 2020-11-03:
@alexmaloteaux does [that](https://github.com/osresearch/heads/blame/master/boards/x220/x220.config#L37) help flashing internally? If yes, then x220 and t420 are to be dealt the same way and a PR should be made accordingly in this repo so that line is also present in t420 boards (which should be the same as x220).

For external flashing, a pull request would be helpful in [that page](http://osresearch.net/t420-flashing/)

Also, I see that your name is not under https://github.com/osresearch/heads/issues/692 for the t420. Do I have your consent to add you there?

alexmaloteaux, 2020-11-03:
> @alexmaloteaux does that help flashing internally? If yes, then x220 and t420 are to be dealt the same way and a PR should be made accordingly in this repo so that line is also present in t420 boards (which should be the same as x220).

Yes it does i have modified pr #889 accordingly

> For external flashing, a pull request would be helpful in that page

ok will do

> Also, I see that your name is not under #692 for the t420. Do I have your consent to add you there?

sure

closed 2020-11-06 by alexmaloteaux