linuxboot/heads #888: Add board schematics to the repository

Siproqu, 2020-11-03:
Wouldn't it be great to have the schematics of several Thinkpads in this repository?
It seems quite common that the EPROMs have to be flashed externally. So one doesn't have to search that long for the pin layout and EPROM location.
Some can be found [here](https://forum.thinkpads.com/viewtopic.php?t=121522).

Thrilleratplay, 2020-11-03:
@Siproqu While locating the SOIC NAND chips can be tricky on some Thinkpads, particularly the T430 and T430s, I am failing to see how board schematics help.  I am very familiar with the X230, I have been corebooting them for years and installed a Nitrocaster FHD on one, these schematics are if anything, more confusing as they do not easily indicate the relationship to each other let alone the location on the board itself.

If your only goal is to display the pin out to assist new users, a technical schematic is the wrong approach; it should be simple and straight forward.  I attempted to do with this [SOIC8 pinout svg](https://gist.github.com/Thrilleratplay/60dff7cf849de5bdb84d91cabbd4fcfd) which is being used in the [skulls documentation](https://github.com/merge/skulls/tree/master/x230).

Additionally, documentation is held in the [heads-wiki](https://github.com/osresearch/heads-wiki) and hosted on http://osresearch.net.  If you feel like the flashing guides could be improved, please submit a pull request there.

tlaurion, 2020-11-03:
For the sake of this discussion, here are the [schematics for the x230 ](http://plan9.stanleylieber.com/hardware/thinkpad/x230/x230.schematics.pdf)

@Siproqu please reopen this issue if you disagree with its closure. I tend to agree with @Thrilleratplay that is not really helpful and heads-wiki would be the place to have recommendations (issues) opened on that matter.

closed 2020-11-03 by tlaurion