linuxboot/heads #897: t430 no boot

alexmaloteaux, 2020-11-08:
Hi 

i compiled heads for t430 (normal and flash) and both just power on and off immediatly.
Coreboot 4.11 is working on it out of the box.

Any idea if this is a well known issue for some board version 

Here are the output of lspci and cpu

00:00.0 Host bridge: Intel Corporation 3rd Gen Core processor DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev 04)
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C216 Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C216 Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.2 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 3 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation QM77 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C216 Chipset Family SMBus Controller (rev 04)
02:00.0 System peripheral: Ricoh Co Ltd MMC/SD Host Controller (rev 07)

processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 58
model name	: Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
stepping	: 9
microcode	: 0x21
cpu MHz		: 1197.269
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips	: 5188.12
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 58
model name	: Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
stepping	: 9
microcode	: 0x21
cpu MHz		: 1197.417
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips	: 5188.12
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 58
model name	: Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
stepping	: 9
microcode	: 0x21
cpu MHz		: 1197.270
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips	: 5188.12
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 6
model		: 58
model name	: Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
stepping	: 9
microcode	: 0x21
cpu MHz		: 1197.293
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 3
initial apicid	: 3
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips	: 5188.12
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

tlaurion, 2020-11-09:
Hello there.
1- Is it commit https://github.com/osresearch/heads/commit/89c457706fdf5295ca5736aa99bdf41582d33972 ?
2- does https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-flash-v0.2.0-947-g89c4577.rom give the same result?

tlaurion, 2020-11-09:
Related to https://github.com/osresearch/heads/issues/667 ?
CBFS region was fixed with 7mb again here https://github.com/osresearch/heads/blame/master/config/coreboot-t430.config#L4 in  https://github.com/osresearch/heads/pull/838 following past hanges which were not fit and caused many problems.

the board config was :
- Orignally set to 8mb (unfit) https://github.com/osresearch/heads/blame/a0753473510ec598f9db86cfb610cf61156ca5cb/config/coreboot-t430.config#L8
- Then moved back to 7.1mb (unfit) https://github.com/osresearch/heads/blame/09ca500d3e9be3c741cc3a668b841d22677687bb/config/coreboot-t430.config#L8
- and fixed to IFD defined right size of 7mb in latest commit.

@alexmaloteaux i've added your github name https://github.com/osresearch/heads/issues/692

Consequently: @flawedworld @Thrilleratplay can you please validate that https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-flash-v0.2.0-947-g89c4577.rom works when flashed internally from https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-flash-v0.2.0-947-g89c4577.rom ?

I thought we were over with this bug from #815 #608 #838 fix.

alexmaloteaux, 2020-11-09:
hi i just flashed externally g89c4577 and still the same, board power off immediatly.

I did not follow since the beginning but why do we want to fit everything only on the bottom chip ? and not use the 12 MB + original ifd ? 

Also why do we use musl toolchain to compile coreboot ?
I tried to upgrade to 4.12 cb for t430 but after several patches with custom muslc toolchain i had to disable RELOCATABLE_RAMSTAGE as it was creating unknown relocation type and anyway endup with black screen too.

tlaurion, 2020-11-10:
> hi i just flashed externally g89c4577 and still the same, board power off immediatly.
> 
> I did not follow since the beginning but why do we want to fit everything only on the bottom chip ? and not use the 12 MB + original ifd ?

I personally want to get away of the two step flashing which takes into consideration that me is not neutered, see #703.
> 
> Also why do we use musl toolchain to compile coreboot and not rely on default one from coreboot itself?
Because otherwise we build against two different toolchains. https://github.com/osresearch/heads/issues/571#issuecomment-712948834 might be the place where you want to participate in debate.

> I tried to upgrade to 4.12 cb for t430 but after several patches with custom muslc toolchain i had to disable RELOCATABLE_RAMSTAGE as it was creating unknown relocation type and anyway endup with black screen too.

I'm personally lost into what is happening with coreboot 4.12 support #709 which is not functional @PatrickRudolph.
No board into Heads are currently depending on coreboot vboot+measured boot as of now. Only boards upported are without TPM support as of now and i'm truely getting bored into challenging technical people in making things go forward.

tlaurion, 2020-11-18:
> hi i just flashed externally g89c4577 and still the same, board power off immediatly.

I'm confused, since g89c4577 rom needs to be flashed internally from t430-flash.rom.
Isn't it what was done?
>> Consequently: @flawedworld @Thrilleratplay @alexmaloteaux @nickfrostatx @lsafd @bwachter @Trapacid can you please validate that https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-v0.2.0-947-g89c4577.rom works when flashed internally from https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-flash-v0.2.0-947-g89c4577.rom ?

So was https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-flash-v0.2.0-947-g89c4577.rom  flashed externally then https://262-64810881-gh.circle-artifacts.com/0/build/t430-flash/heads-t430-v0.2.0-947-g89c4577.rom flashed internally? I'm confused here.

> 
> I did not follow since the beginning but why do we want to fit everything only on the bottom chip ? and not use the 12 MB + original ifd ?

Maybe this shema will make it clearer: https://doc.coreboot.org/mainboard/lenovo/Ivy_Bridge_series.html#flash-layout
the x230/t430 base x230 image contains a fake ifd, and takes into consideration that x230-flash was initially flashed, so that only the [BIOS region of the IFD is flashed from x230-flash](https://github.com/osresearch/heads/blob/master/boards/t430-flash/t430-flash.config#L17), with consistent BIOS region size from original Lenovo rom image. The same applies when flashing upgrades internally.

Going out of this manual path is the goal of #703 and #797 

> 
> Also why do we use musl toolchain to compile coreboot and not rely on default one from coreboot itself?

That is a good question, actually. I'm not sure about the musl-cross-make path we chose [while building alternatives](https://github.com/osresearch/heads/issues/571#issuecomment-712948834).

> I tried to upgrade to 4.12 cb for t430 but after several patches with custom muslc toolchain i had to disable RELOCATABLE_RAMSTAGE as it was creating unknown relocation type and anyway endup with black screen too.

The black screen itself is the problem to resolve here and should be linked with inconsistencies from ifd defined size of BIOS region and ME regions.

alexmaloteaux, 2020-11-19:
i had misunderstood the steps , 

flashed me cleaned  original bottom
flashed t430-flash on top
then internally reflashed t430 and all is working well now : 

thanks for clarification

closed 2020-11-19 by alexmaloteaux