linuxboot/heads #899: Possible to use one GPGcard to sign /boot on different Heads-laptops?

userongihu, 2020-11-14:
@tlaurion,in order to check out if my ch341a works,I reflashed the x220 with the Heads.rom that I still had.It all works,but one thing is still the same:the OS on the HDD could only start after making an "OEM Factory reset".Right now I am building a Heads.rom for the x230.Will I be able to get the x230 running without making the "OEM Factory reset"?What's your experience?I am asking because making the "OEM Factory reset" would implicate that the GPGkeys on the GPGcard,that were generated for the x220,would get deleted since the "OEM Factory reset" wipes everything.Could I use the GPGkeys that are on the GPGcard for the x220,also for signing the /boot of the x230?Do I need a different GPGcard for each laptop in order to use Heads on them?

Tonux599, 2020-11-15:
As long as you can get the public key from your PGP Card in the form of an `.asc` file and place it onto a flash drive. Heads can import that key so you can use your PGP card to sign files in /boot on as many machines as you like.

tlaurion, 2020-11-15:
@userongihu https://github.com/osresearch/heads/commit/5dc65f63b6107bf3628433b5d5f49e2f3dd0f47b permits you to export public key to USB drive. Inject that key into another Heads ROM from the GPG menu and you can use the same USB Security dongle on multiple laptops. Not to be confused with HOTP, whcih can only remote attest one laptop at a time.

So you can use one USB Security dongle with multiple x230, x220, t430 boards produced roms.
While you can only bound that USB Security dongle per laptop enforcing [HOTP](https://github.com/osresearch/heads/blob/master/boards/x230-hotp-verification/x230-hotp-verification.config#L34) for remote attestation.

You do not want to run factory reset. You cannot export a public key from a USB Security dongle. But you can export a public key from Heads, which was facilitated with referenced commit, directly from GUI.

If that answers your question, please close this issue.

userongihu, 2020-11-15:
@tlaurion @Tonux599,thanks for your answers.Closing issue here.

closed 2020-11-15 by userongihu