linuxboot/heads #901: x230 flashed top-chip.How to proceed from here?

userongihu, 2020-11-15:
@tlaurion @MrChromebox  ,I just flashed Heads onto the top_chip of the x230.For that I went into builds/x230-flash and flashed x230-flash.rom.Now when I start the x230 I get into Heads-recovery-shell.How do I proceed from here?When plugging in the USBdrive that has the second Heads.rom on it,which needs to be installed internally,typing "cd /media" and then "ls" it shows no content,although there are some files on the USBdrive.

Edit:Before flashing Heads on the x230,I cleared the keys of the security-chip in the Lenovo-Bios.But I didn't take ownership.On the x220 I cleared the keys and took ownership with tpm-tools before flashing Heads.

userongihu, 2020-11-15:
I got the USBdrive mounted and now I see all the files on it.But I don't know if I do have to take ownership of TPM first before Heads could work.

userongihu, 2020-11-15:
@tlaurion @MrChromebox ,I installed the 12MB Heads-rom and it worked,got the running Bios signed with GPGkeys that were on an USBdrive.GPGkeys showing up on the keyring.But it turned out that setting up the TOTP  wasn't possible because I didn't set up a password for the TPM in the Lenovo Bios before installing Heads.So I have to reflash Lenovo Bios again to set up the password for TPM.Before installing Heads I made backups of the vendor firmware of both chips.What would be the flashrom command in order reflash the Lenovo Bios?By the way,before installing Heads I  cleaned ME following the tutorial on osresearch.net.

What would be the flashrom command for that

tlaurion, 2020-11-16:
@userongihu you have to reset TPM from the TPM menu under options menu.

userongihu, 2020-11-16:
@tlaurion,thank you.I could set up TPM inside Heads.Everything works now.

closed 2020-11-16 by userongihu