linuxboot/heads #905: Building toolchain fails

LukasWerthmanns, 2020-11-21:
Hi,
I'm trying to build heads for the x230 but it fails building the toolchain.
I could not find anything to help me. Please tell me if you need further information.
Running Archlinux.

here is the full output of:
make BOARD=x230
```
 Wrong make detected: 4.3
tar xf "/home/lukas/computer_stuff/heads/heads/packages/make-4.2.1.tar.bz2" -C "/home/lukas/computer_stuff/heads/heads/build"
touch "/home/lukas/computer_stuff/heads/heads/build/make-4.2.1/.extract"
( cd "/home/lukas/computer_stuff/heads/heads/build/make-4.2.1/" ; patch -p1 ) < "patches/make-4.2.1.patch"
patching file glob/glob.c
touch "/home/lukas/computer_stuff/heads/heads/build/make-4.2.1/.patch"
cd "/home/lukas/computer_stuff/heads/heads/build/make-4.2.1/" ; \
./configure 2>&1 \
| tee "/home/lukas/computer_stuff/heads/heads/build/log/make.configure.log" \

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for ranlib... ranlib
checking how to run the C preprocessor... gcc -E
checking for ar... ar
checking for perl... perl
checking the archiver (ar) interface... ar
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for library containing strerror... none required
checking for a sed that does not truncate output... /usr/bin/sed
checking whether NLS is requested... yes
checking for msgfmt... /usr/bin/msgfmt
[... 624 lines cut ...]
obj/iasl ../../../source/components/utilities/utmath.c
obj/iasl ../../../source/components/utilities/utmisc.c
obj/iasl ../../../source/components/utilities/utmutex.c
obj/iasl ../../../source/components/utilities/utnonansi.c
obj/iasl ../../../source/components/utilities/utobject.c
obj/iasl ../../../source/components/utilities/utownerid.c
obj/iasl ../../../source/components/utilities/utpredef.c
obj/iasl ../../../source/components/utilities/utresdecode.c
obj/iasl ../../../source/components/utilities/utresrc.c
obj/iasl ../../../source/components/utilities/utstate.c
obj/iasl ../../../source/components/utilities/utstrtoul64.c
obj/iasl ../../../source/components/utilities/utstrsuppt.c
obj/iasl ../../../source/components/utilities/utstring.c
obj/iasl ../../../source/components/utilities/utuuid.c
obj/iasl ../../../source/components/utilities/utxface.c
obj/iasl ../../../source/components/utilities/utxferror.c
- Link obj/iasl
/usr/bin/ld: obj/aslcompilerparse.o:(.bss+0x8): multiple definition of `AslCompilerlval'; obj/aslcompilerlex.o:(.bss+0x0): first defined here
/usr/bin/ld: obj/prparserlex.o:(.bss+0x0): multiple definition of `LexBuffer'; obj/dtparserlex.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [../Makefile.rules:15: obj/iasl] Error 1
make[1]: Leaving directory '/home/lukas/computer_stuff/heads/heads/build/coreboot-4.8.1/util/crossgcc/acpica-unix2-20180531/generate/unix/iasl'
make: *** [Makefile.common:7: iasl] Error 2
cp: cannot stat 'bin/iasl': No such file or directory
```

userongihu, 2020-11-23:
I remember having that output before.Solution was installing all the dependencies that are stated osresearch.net.Before that I installed the dependencies that were mentioned in a different Heads-install tutorial that I found online.I installed Heads on debian 10 and works as supposed.

userongihu, 2020-11-23:
@LukasWerthmanns ,as far as flashrom is concerned I used the one that comes with debian 10
and just in case you want to remove IntelME just stick to the tuto that is mentioned on osresearch.net.It worked for me copy&paste all the way.

LukasWerthmanns, 2020-11-27:
@userongihu Thanks. Installing all the dependencies worked on ubuntu.

closed 2020-11-27 by LukasWerthmanns

AldenIsMe, 2021-06-21:
Hello!
Does anyone have those dependencies? I've been going at this 6 hours straight.. Tried the wayback machine on the heads site, installed depends, downgraded make, after having to modify the glob.c file in the make source (that one took 3 hours after realizing you cant apt make=4.2)

Any help would be appreciated. Trying to "make BOARD=x230-hotp-maximized" on Kali Linux WSL. Getting same error as above:

Building IASL v20180531 for host ... failed. Check 'build-IASL/build.log'.

in the build.log:

- Link obj/iasl
/usr/bin/ld: obj/aslcompilerparse.o:(.bss+0x8): multiple definition of `AslCompilerlval'; obj/aslcompilerlex.o:(.bss+0x0): first defined here
/usr/bin/ld: obj/prparserlex.o:(.bss+0x0): multiple definition of `LexBuffer'; obj/dtparserlex.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [../Makefile.rules:15: obj/iasl] Error 1

LukasWerthmanns, 2021-06-21:
@AldenIsMe 
Ok, the way I remember it I found the dependencies inside a file in the github repository not the wiki.
I'm not 100% sure but it could have been this one.
https://github.com/osresearch/heads/blob/master/.circleci/config.yml
After installing everything featured in the 'apt install' lines on ubuntu 18.4 building finally worked for me.

I hope my foggy memory is of any help to you.

AldenIsMe, 2021-06-22:
> @AldenIsMe
> Ok, the way I remember it I found the dependencies inside a file in the github repository not the wiki.
> I'm not 100% sure but it could have been this one. [...]

Thanks for your quick reply! In the end I gave up trying to install on Kali WSL, so I just rented a quick 2 hour vps from bitlaunch so its semi secure haha! And it seems to be working, maybe its an issue with the windows subsystem? Not sure.. But I did try on 1gb of ram, and it ran out of memory, so I'm currently trying on 8GB, hopefully it builds fine :) Taking its time as usually.. I'll update this if it works! Thanks for the depends, as it might not of even worked on the vps without it.