linuxboot/heads #910: How to investigate a TOTP mismatch?

Siproqu, 2020-11-26:
I am looking for some help how I can investigate the cause of TOTP mismatches.

The mismatch occurred after I booted my laptop this morning. This was the first cold boot for 1-2 weeks, before that it has only been in S3.
The time was clock was one hour off (which it hasn't been 1-2 weeks before), but even after changing it to the correct time the TOTP still doesn't match.

Any hints how I can start to investigate this before I reflash?

tlaurion, 2020-11-26:
Reflashing won't fix. Should probably put this somewhere else but here you go for [my draft](https://zeronet.now.im/1DMb3CV66qZPwJqkgm4z12nu8BrAwDoD4g/?Post:44:Setting+time+into+Heads+when+TOTP+code+is+bad)

Siproqu, 2020-11-26:
I have already set the time with date -s and hwclock -w. hwclock -r returns the correct time after reboot (I have also tried to remove the main battery for 5 minutes to check whether the RPC battery is good - it is).
But the TOTP values are still incorrect.

> That should probably be under https://github.com/osresearch/heads-wiki (and http://osresearch.net and a result).

Shall I close this one and reopen it on https://github.com/osresearch/heads-wiki?

tlaurion, 2020-11-26:
So your firmware was modified?

Siproqu, 2020-11-26:
Not that I am aware of. Any chance a Qubes update modified the firmware?

tlaurion, 2020-11-26:
> Not that I am aware of. Any chance a Qubes update modified the firmware?

@Siproqu : No. QubesOS only modifies /boot components, which triggers a detached signed (kexec_hashes.txt signed under kexec.sig) discrepency and invalidates default boot validation.

You are sure you didn't inserted a new public key or saved user config changes under Heads? Those are the only moments the firmware should be modified. Of course, if you updated your BIOS, that would also invalidate TPM measurements in PCRs.
http://osresearch.net/Keys/#tpm-pcrs

Siproqu, 2020-11-26:
First of all, thank you very much for you detailed answer!

Now comes the embarrassing part. The initial mismatch of the TOTP was likely because of the time shift. Unfortunately, when I corrected the time I didn't set it to GMT-0. Set it to GMT-0 worked obviously.

@tlaurion Thank you for helping out and sorry that I wasted your time by not reading your post more thoroughly.
I'll take some time tomorrow and create a pull request on heads wiki for this kind of issue.

For all the other scatty people who ran into this problem:
The [TOTP algorithm](https://tools.ietf.org/html/rfc6238#section-4) uses the current _[Unix time](https://en.wikipedia.org/wiki/Unix_time)_ to calculate the next password. And the current Unix time is in 00:00:00 **GMT-0**/UTC on 1 January 1970.
So if you set Heads time to GMT-1 or any other time zone other than GMT-0, it generates passwords for the future or the past.
The ppp on your phone will also use GMT-0 to calculate the passwords. Regardless what time zone you are currently in.

closed 2020-11-26 by Siproqu