linuxboot/heads #911: Error loop if Heads is configured and hard drive is missing or replaced

Thrilleratplay, 2020-11-26:
This is an outlier issue but wanted to document it.  My daily driver x230 coreboot laptop died and to be able to continue working, I swapped the hard drive with my x230T Heads test machine.  The x230 hard drive is fully encrypted and assumed Heads would show a configuration error letting me know that something is wrong but could still boot the hard drive using kexec.  Instead, it was an infinite loop saying could not mount /dev/sda1 (I should have written down the exact verbiage but it is easy enough to replicate).  

While I do not expect anyone else to do what I did, this may still occur if a hard drive fails and is replaced with a blank drive.  The Heads device is useless at this point and cannot be configured without reflashing.  I am not sure what should be displayed in this event but the option to reconfigure the machine or at least access to the shell would be useful.

tlaurion, 2020-11-26:
@Thrilleratplay I do not understand the situation here.
So you swapped drives from a dead x230 to X230t working test machine. Drive is expected to be good with QubesOS sda1 being boot and sda2 being encrypted. Booting Heads with https://github.com/osresearch/heads/pull/869 merged and not being offered a usb boot option?

It seems here that your drive may be the problem here?

Thrilleratplay, 2020-11-26:
@tlaurion #869 improves how Heads behaves if it is not configured like when a user is first setting up the device.  You are correct, it is my drive that is the problem because it is fully encrypted and does not have a `/boot` partition.  I described my scenario as an example and but that may have been confusing.  Heart of the issue is that if a Heads user has a hard drive failure or tries to upgrade the drive, then replaces it with a new blank drive, there is a n error loop that prevent the machine from getting to the menus or shell.  The only way to resolve this would be to flash Heads externally and reconfigure as if it were a new device.

tlaurion, 2020-11-26:
Well, if that helps,[ typing 'r' frenetically prior of "Heads" ascii art being showed sends the user in recovery.](https://github.com/osresearch/heads/blame/master/initrd/init#L72-L90)

Where `mount-usb` and `usb-init` permits to boot from USB device.
I would have swore that a default boot to usb was implemented to boot to USB when no /boot detected....

Thrilleratplay, 2020-11-26:
@tlaurion I should have written down the error before opening the issue.  

`mount: mounting  /dev/sda1 on /boot failed: invalid argument`

This is not an issue when a hard drive is removes or fails, only when the first drive partition cannot be mounted.  And you are correct, hitting "r" will skip this and go directly to the shell prompt.  This is far less worrying and more of a test case when cleaning up the initrd scripts.  I'll close this and add it to that ticket.

closed 2020-11-26 by Thrilleratplay