linuxboot/heads #919: 4096-bit key length breaks OEM reset with Librem Keys

MrChromebox, 2020-12-04:
ugh, not sure how I missed testing #906, but 4096-bit key length breaks the OEM reset with Librem Keys. Ends up hanging on GPG key generation, with the dongle returning 'card busy' forever. Setting the key length back to 2048 resolves the issue. What's the best way to guard or override this?

tlaurion, 2020-12-04:
Should work but takes 15 minutes to generate keys?

tlaurion, 2020-12-05:
@MrChromebox : This is text is misleading though https://github.com/osresearch/heads/blob/master/initrd/bin/oem-factory-reset#L423

MrChromebox, 2020-12-07:
@tlaurion how does something like that not get mentioned in the PR? going from 2 mins to 15 mins is a massive change, one which would cause most users (and devs) to think something had broken.

As a 15-min keygen time isn't ideal for Purism boards, I'm thinking that a patch to default to 4096, but to allow boards to override via board config would be ideal -- same way we do now for BOARD_NAME / CONFIG_BOARD_NAME

Any objection to that approach?

tlaurion, 2020-12-07:
@MrChromebox : No objection, other then not being sure what is the point of having that added security inside of the key if not taken advantage of at the moment of generating the keys (which you do only once). I think users would have nothing against it [if they were warned about it](https://github.com/osresearch/heads/pull/551/files#diff-8da0a88ec342eb21e063e3dc455ef516aedf7a4e8e9d2f387e9c993995f001bfR90-R91). (I received 0 complain as of now for the past year).

The idea here is to secure data at rest for longest possible time against traditional computing attacks.
While:
`For RSA-2048, current computers would take around 300 trillion years to discover RSA keys using a brute force attack.`

Nothing happened, really, in the news sphere, but:
`When protecting against attacks by traditional computers, systems can achieve great security by extending the key-length of the algorithm. An organization could switch to RSA 4096, for example. With quantum computers, however, that approach will not work. Even though RSA 4096 uses a key that is twice as long, the difference in time for a quantum computer to break RSA 4096 vs. RSA 2048 is negligible.`

[SRC](https://sectigo.com/resource-library/quantum-safe-cryptography-surviving-the-upcoming-quantum-cryptographic-apocalypse)

kylerankin, 2020-12-07:
I support the idea of this being optional/configurable if the default is to be changed both to account for customers *and* companies who have different threat models. In our case the key is aimed only to protect secrets during shipping. We advise people to change keys upon receipt and I believe many of our customers would interpret a 15 minute key generation like @MrChromebox did: as the system freezing/failing, and they would contact our support team for help.

Our average customer does not have NSA-with-a-quantum-computer in their threat model, or if they do, the reasonable among them would still not expect RSA 2048 to be cracked within a day or two during shipment interdiction (and if they did, we could accommodate them during our anti-interdiction process to generate a larger key). We suggest change they keys upon receipt (which any customer with NSA-with-a-quantum-computer in their threat model would do as good practice anyway) and at that point the customer with that threat model could opt to use a stronger key. For everyone else who doesn't have that extreme threat model a 15 min wait might risk discouraging them from doing a key change.

tlaurion, 2020-12-07:
https://github.com/osresearch/heads/issues/831#issuecomment-691897411

So yes, it seems that the default should set to something, and that something should be configurable.
My insight would be to prompt the user ("Use default of $KeySize bits? Y/n") and then validate for valid input prior of going forward.

techge, 2020-12-08:
At Nitrokey we are using 3072-bit keys and this is reasonable fast imho.

tlaurion, 2020-12-08:
@kylerankin @MrChromebox @techge : I would agree that following upstream decision should be the default.
As reported https://github.com/osresearch/heads/issues/831#issuecomment-691897411, the latest versions default to 3072, where the LTS version that was pinned into gpg2 module doesn't. I would suggest simply making the user/oem aware of the waiting time and have the default option output waiting time correctly.

@techge @MrChromebox @kylerankin :
- 4096 bits RSA subkey generation takes 5 minutes each for a total wait time of ~15 minutes (maximum)
- 2048 bits RSA keypair generation is taking a while 2 minutes ( so 45 seconds per subkey)
- @techge what are the waiting time for 3072 keypair and subkey generation time?

MrChromebox, 2020-12-08:
> Once again, keygen is not happening everyday... and should not happen everyday.

but it does, both in the production workflow of preparing devices from shipping, and in testing.

> My point is that if user/OEM fulfillment guy pairing USB Security dongle to platform (to protect it in transit and then for data at rest, right?), he a priori knows that 15/7/2 waiting minutes are needed prior of doing next steps of fulfillment) which is actually a false problem and another choice of lesser security for more convenience.

it's not a false problem at all -- all security decisions are tradeoffs. it's a question of what threat model is attempting to be mitigated by a given measure/set of measures, and are the measures sufficient.

So the real question is, what is the threat model, does is a 3072 or 4096-bit key provide a quantifiable security benefit against that model that justifies the 'inconvenience'

tlaurion, 2020-12-08:
@MrChromebox There won't be consensus. So the path needs to be to define default under boards and offer GUI to be able to choose from preconfigured options in code to switch between those, limiting input error and not complexifying different use cases/different threat models.

MrChromebox, 2020-12-08:
@tlaurion I'll implement and push a PR, and then can adjust as needed

jans23, 2020-12-09:
> @techge what are the waiting time for 3072 keypair and subkey generation?

Roughly, 1-3 minutes.

tlaurion, 2020-12-09:
@jans23 so it is 9 minutes total (3 minutes each subkey?) or 3 minutes total keygen session (1 minute per subkey?) 
Sorry, I just want numbers given by each defender of the default value to be changed.

jans23, 2020-12-09:
With the current way of generating RSA-3072 keys, it's not more than 3 minutes in total for all keys.

tlaurion, 2020-12-10:
@MrChromebox : considering this, I think middle ground for all parties would be to default for 3072 which stands in the middle at all points. I can push a PR to notify of the proper delay to expect and 3072 to address the current issue.

MrChromebox, 2020-12-10:
@tlaurion I think that's a perfectly acceptable compromise :)

fhvyhjriur, 2020-12-10:
A PR, where it can be adjusted more simple to 4096 would be nice like @MrChromebox mentioned. There are for sure some users/journalists that cant code but have such a treat model or simply would like to wait 15min instead of 3min for the additional keysize.

MrChromebox, 2020-12-10:
@fhvyhjriur so you mean a selection prompt when performing the OEM reset? As opposed to a board-level config override?

fhvyhjriur, 2020-12-10:
I dont know what is the best from the user-experience perspective. It should be something that at some point the user see this option and can decide to choose it. The user should see in a short sentence that when the user choose this option, some waiting-time could be trade for higher security with no real issues except this 15min waiting time.

A example user-experience implementation that got into my mind is the question some linux-os ask when full disk encryption is been chosen. When disk encryption is chosen the user also gets in many cases the ability to overwrite everything with random data. This takes some time and adds some benefits with no real issues in later usage.
The user should simply see this 4096-bit key possibility at some time to be able to choose it.

tlaurion, 2021-01-03:
@fhvyhjriur @MrChromebox @jans23 @kylerankin @techge #948 fixes the user expectation until we come with a more brilliant solution.

PR #948 (2021-01-03): oem-factory-reset: set default KEY_LENGTH to 3072 and change expectation management message to console
(Fixes #919)
commit 64b1712e (2021-01-03): oem-factory-reset: set default KEY_LENGTH to 3072 and change expectation management message to console (Fixes #919)
closed 2021-01-03 by tlaurion