linuxboot/heads #920: [help wanted] hdd i/o error.

bencejuhaasz, 2020-12-07:
I followed the instructions on the wiki, used the master branch on ubuntu 18, got a .rom that starts&can boot things from usb, but can't read /dev/sda. How should I try to build this ? Are the releases outdated ?

Thrilleratplay, 2020-12-07:
@bencejuhaasz It would be helpful to know what device you are trying to set heads up on.  Is there a hard drive installed in the device?

bencejuhaasz, 2020-12-07:
Sorry, I forgot it, it's an x230, and it has a hdd installed.

Thrilleratplay, 2020-12-07:
@bencejuhaasz Unfortunately, "followed the instructions on the wiki" does not help explain what you have done thus far.  Describe the steps you took to install Heads and be as descriptive as you can be.  Did you build it?  Which board did you build for? Externally flash? What does Ubuntu 18 has to do with your Heads installation?  Yes [these releases](https://github.com/osresearch/heads/releases) are out of date, did you use a release build?  What is on the hard drive?  What else does it say other than `/dev/sda`?

Is the error saying `/dev/sda` or `/dev/sda`?

bencejuhaasz, 2020-12-07:
I followed this guide: http://osresearch.net/Building/ and used x230 as the BOARD variable, then I flashed the 12MB image with this (https://github.com/gch1p/thinkpad-bios-software-flashing-guide) exploit. I used an Ubuntu 18.04 install to build the .rom, and an other one as a live cd to flash. There's a Qubes install on /dev/sda. The problem is, that Heads can't read a single bit out of /dev/sda, doesn't even see the partition table. If I do `cat /dev/sda`, no random data fills the screen, I just get more kernel log errors about some kind of i/o error. I'll try to paste these errors here.

bencejuhaasz, 2020-12-07:
What would be an ideal, known to work build environment ? Are there succesfull builds for the x230 with the current version?

Thrilleratplay, 2020-12-07:
You executed `cat /dev/sda` in the Heads rescue shell?  If so, try `dmesg | less` and scroll through to see if there are any errors.  The hard drive is internally and connected via sata?

bencejuhaasz, 2020-12-08:
Yes, internally and with sata.

bencejuhaasz, 2020-12-08:
https://pasteboard.co/JDWNk5c.jpg

bencejuhaasz, 2020-12-08:
Looks like my hard drive was (or got?) physically damaged at flashing. Replaced it, Heads works fine. Thanks for your great work! (And fast response).

closed 2020-12-08 by bencejuhaasz

tlaurion, 2020-12-08:
> What would be an ideal, known to work build environment ? Are there succesfull builds for the x230 with the current version?

@bencejuhaasz ROMs are built for most of the most used platforms where blobs are not present (faked IFD, no ME redistribution without permission, or ME downloaded directly from manufacturer, extracted and put in place under blobs directory) here for each pushed in master, where contributors are invited to register themselves over CircleCI and make their branch followed there so that their PR are already built and testable by the community prior of merging. You can click on the green checkmark under Heads.

For example, at the time of writing this:
- https://github.com/osresearch/heads/commit/671522eff43bbf87e3112ed0075f29fbf1a391fe is the latest commit.
- Green mark of that commit points to [CircleCI builds](https://circleci.com/gh/osresearch/heads/275?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link), which shows trace of builds, where it failed if that is the case, outputs the hashes of each built module and rom, and link you to the artifact section where built roms are downloadable.
- [Artifacts section](https://app.circleci.com/pipelines/github/osresearch/heads/260/workflows/1912855b-2150-4986-82fe-498be1700e80/jobs/275/artifacts) lists all the boards build artifacts.
- Where the [direct link for that latest commit is directly downloadable](https://275-64810881-gh.circle-artifacts.com/0/build/x230/heads-x230-v0.2.0-972-g671522e.rom).

Note that ROMs differentiate x230 board definitions per boards modules inclusion and size restrictions (internally flashed per 1vyrain or other exploits not neutering ME nor taking advantage of the total available space, which requires external flashing): x230, x230-hotp-verification, x230-hotp-maximized, x230-maximized)

**To take advantage and use the maximized boards, initial external reprogramming is required.**

tlaurion, 2020-12-08:
@bencejuhaasz would you consent to be added to board owners https://github.com/osresearch/heads/issues/692 ?