linuxboot/heads #921: Signing /boot without physical gpg key device

bencejuhaasz, 2020-12-08:
Can I sign /boot with a software gpg key from a usb stick, without specific hardware for storing keys ? I can import my public key but the UI won't sign anything or import my private key, only from a special smartcard device.

tlaurion, 2020-12-26:
Not in current Heads codebase. A USB security dongle is required, for which signing operations depends on gpg --card-status and where private key is on external dongle.

Any implementation ideas on workflow? The private key could be injected in rom and only require the user to type associated passphrase... But that would lower security.

How to detect and propose to user to connect USB disk to load private key seemlessly? 

@bencejuhaasz The private key would be in encrypted LUKS container on USB disk?

fhvyhjriur, 2021-01-11:
I know at least one person that would like to test out heads without the need of a security dongle. This person know that its insecure to have private and public key on a unencrypted usb drive just as files that everyone can read.
Recently there is no insecure "find out if you like heads before spending money for a dongle"-mode. Such a mode would be nice for testing heads.

tlaurion, 2021-01-21:
Iterative thought process happening.

Corrently, all heads codebase but the following takes into consideration the use of a smartcard, calling gpg without a homedir, where .gnupg is created locally at heads runtime and solely in ram.

The following are the exception, specifying homedir to not interfere with the rest of codebase which implied locally created .gnupg dir for smartcard operations. The following is called to support all iso related disto key validation, so that booting Qubes/Tails actually trusts their distro keys implicitly. key-init is called really early in init, and then kexec-iso-init is called on demand when booting from USB:

```
user@x230-master:~/heads$ grep -Rn homedir initrd/
initrd/bin/key-init:11:gpg --homedir=/etc/distro/ --import /etc/distro/keys/* 2>/dev/null || true
initrd/bin/key-init:13:gpg --homedir=/etc/distro/ --list-keys --fingerprint --with-colons|sed -E -n -e 's/^fpr:::::::::([0-9A-F]+):$/\1:6:/p' |gpg --homedir=/etc/distro/ --import-ownertrust 2>/dev/null || true
initrd/bin/key-init:14:gpg --homedir=/etc/distro/ --update-trust 2>/dev/null || true
initrd/bin/key-init:17:gpg --export | gpg --homedir=/etc/distro/ --import 2>/dev/null || true
initrd/bin/kexec-iso-init:20:gpgv --homedir=/etc/distro/ "$ISOSIG" "$MOUNTED_ISO_PATH" \
```

We could "replace" a smartcard by a LUKS encrypted container's partition, having local, non homedir bounded gpg calls, taking into consideration that `/.gnupg` is:
- existing over crytpsetup luksOpen call being embedded into mount-usb logic, so that if partition is tested positive to `cryptsetup isLuks`, decrypted, mounted and then `/media/.gnupg` softlinked into `/.gnupg`
- Possibliy protected by PIN on private key.

Code changes that would be required under Heads
- mount-usb logic changed to support cryptsetup partition and mounts (where booting ISO from that will just not work on kexec calls)
- key-init changes applied if GPG_CARD_IS_LUKS in board config (and consequently under /etc/config) with above detailed logic
- gpg-gui, oem-factory-reset oand other scripts calling `gpg --card-edit` should do only if not `GPG_CARD_IS_LUKS `i nboard config

My thought experiment is that all other gpg calls are plainly depending on bindings between public key and private key identity being offloaded to card ID. So gpg is simply requesting that smartcard presence to offload its operations after the mapping is done through gpg --card-status and gpg knows if private key is local or on smartcard. 

Thoughts, better idea?

tlaurion, 2021-01-21:
The outcome of this should be used to document https://github.com/osresearch/heads-wiki/issues/62

bencejuhaasz, 2021-01-30:
I've found a soultion since. I just had to remove the line checking for a physical device, and import my encrypted key with gpg, saying "N" for changing things. But sadly now I can't use heads, as I've switched to arch linux, so I'm locked to secure boot.

closed 2021-01-30 by bencejuhaasz

tlaurion, 2023-12-23:
Implemented under https://github.com/linuxboot/heads/pull/1515