linuxboot/heads #922: Please enable TLS/HTTPS on osresearch.net

fhvyhjriur, 2020-12-08:
The webseite is missing even the most basic TLS encryption. And thus it does not load any more with modern browsers like Firefox where in the security settings (no addons required) following is enabled and would be enabled by default in the future:

HTTPS provides a secure, encrypted connection between Firefox and the websites you visit. Most websites support HTTPS, and if HTTPS-Only Mode is enabled, then Firefox will upgrade all connections to HTTPS.

Thrilleratplay, 2020-12-08:
Duplicate of https://github.com/osresearch/heads-wiki/issues/14

tlaurion, 2020-12-08:
@osresearch or anyone knowing how to fix this, please point us in the good direction under osresearch/heads-wiki#14

tlaurion, 2020-12-08:
@fhvyhjriur : meanwhile, one possible mitigation is to use [archive.org](https://help.archive.org/hc/en-us/articles/360001513491-Save-Pages-in-the-Wayback-Machine)

tlaurion, 2021-01-29:
"Looks like it is protected now. The issue was the A records pointed to an old version of the GitHub servers, which were not supported by their ssl cert system"

closed 2021-01-29 by tlaurion

fhvyhjriur, 2021-01-29:
Yes, TLS/HTTPS is enabled now. Thanks for closing the issue.