linuxboot/heads #926: xx20-maximized boards SPI already full when adding cryptsetup2

tlaurion, 2020-12-11:


Thrilleratplay, 2020-12-12:
@tlaurion This is going to be an on going issue.  I have no suggestions on what to keep or what to take out.  What ever is decided, there will be someone asking for another configuration.  Whatever you decide to add or remove for the CI build is fine with me but if possible would prefer, at some, we create a chart of features and the required space to allow the user to build the version that best suits them.

fhvyhjriur, 2020-12-12:
From the OpenWrt project for devices with 4mb SPI chips the space also got low with growing linux kernel and maybe it helps to know how its done there:
The default image that can be installed without any compilation and that is kept updated for security and other updates is the one that probably most user would be happy with. There is no real discussion with the users about what most user would probably like to have. The developer decide the default usability on their own.
There should be one default and ready image available for download and all user with different wishes have to build their own image.
Example in OpenWrt for custom user-wishes: Users that want to have VPN functionality instead of the Webbrowser-GUI. Other users swap out their SPI chips for more space or solder USB wires to some legs of a chip to have a USB-port and then also need custom images they build with small changes to the already existing default image-configuration.

tlaurion, 2020-12-13:
@fhvyhjriur agreed, while board additions are welcome from the community users, where board names should be clear enough to state what the use is. For example, if a SPI change is required, the board name should explicit that with `-16mb_SPI`, and where users pushing those boards should also be the maintainers of such boards. The direction that will be pushed forward is that if a board cannot be built anymore for space limitation, that board will need to be removed from CI for other boards to continue to be built and an issue will be opened, stating the last commit ID with which it built with its current subset of functionalities. 

This issue is fixed with #928, which removes NITROCLI, while being ready for CRYPTSETUP2 change which will be needed for #237 enforcing LUKS2 at installation.

closed 2020-12-13 by tlaurion